PiSupply/PiJuice #606: Battery 12000mAh not charging, showing wrong temperature

S-lac, 2020-12-09:
Hey !

Thanks for your great work with PiJuice ! 

I recently bought a PiJuice for a remote application with cameras and a Pi4. Everything ran smoothly with the native 1820mAh battery. Charging/Discharging correctly, showing temperature correctly and PiJuice waking up when needed for a week or so. After running some tests, I decided to go for a bigger capacity battery and bought the 12000 mAh from PiSupply website. 

I tested it yesterday for the whole day, I plugged correctly the 3 wires in the connectors like in the wonderful PiSupply video :) I correctly switched settings in the PiJuice GUI to the _PJLIPO_12000 (charge enabled ticked). I have the green LED properly showing on the board too.

[image]

But the battery temperature has been stuck to 246°C and battery % stucked to 52% for a few hours. Even tho it wasn't hot at all 246°C would have burned my hand I guess ? 

[image]
[image]

Changing the NTC parameter to 'not used' in the GUI for PJLIPO_12000 and I went back to 25°C and % battery load is now showing charging correctly 👍 

[image]
[image]

Even thought my problem seems to be resolved, I wanted to let you know about the behavior. I have one question now, how do I correctly get the battery temperature without using the NTC ? During my testing before sending the system to the field, I'd like to make sure everything goes smoothly, even the battery temperature behavior, and it seems that NOT_USED makes it stay at 25°C forever. 

Final small question, what do you guys do with the camera cable when using with PiJuice, Is it ok to leave it out like I did ? 

Thanks for your help and the great work !

S-lac, 2020-12-13:
[UPDATE]

I did a test of discharge for both batteries, from 95 to 10%. 12000mAh is set on 'NOT_USED' for the temperature part. 1820mAh is standard. I have firmware 1.4 (last), only a Pi4 connected to wifi, exactly the same setup for each battery (didn't even move the Pi4. 1820 gives a solid 90 mins of charge, while 12000 gives a solid... 50 mins :)

tvoverbeek, 2020-12-13:
The GPIO side loks OK. The strange thing is the battery current for the 12000 mAh battery is more than twice the current for the 1820 mAh battery.
I start to suspect a faulty 1200 mAh battery. Where did you buy it?

S-lac, 2020-12-13:
Hi @tvoverbeek, thanks for the reply. 
I bought it straight from PiSupply a week ago, I'll double check my tests and wire but seeing that the 1820mAh works like a charm I'm starting to think the same thing :)

S-lac, 2020-12-16:
[final update]

Did the same tests 3 times and got pretty much the same errors. I'm now sure I have a faulty battery there. I bought a new one from PiSupply and tested it, had exactly the same problems with the NTC (is it a firmware problem ?), I had to use 'NOT_USED' for temperature sensor to make it work correctly. Now the new battery correctly charges and discharges correctly and I have a solid 511 mins of update with a Pi4, LEDs on and Wifi on + HDMI. I highly recommend to have a look at the NTC of the 12000mAh, seems to be buggy or unusable with the PiJuice for now.

kaulex99, 2020-12-31:
#577 I have the same problem

shawaj, 2020-12-31:
@tvoverbeek I just got the following email from @mmilann after sending him a battery to test:

```
12000 battery you sent seems to have different specifications, NTC value is around 40K at room temp. Do you have any specs from the manufacturer, and NTC specs like nominal resistance and B constant?
```

So looks like an issue with the NTC. Will email the manufacturer and report back. Looks like people may need to adjust the settings of the NTC in the battery profile.

shawaj, 2021-03-15:
Closing in favour of #659

closed 2021-03-15 by shawaj

S-lac, 2021-03-16:
Sorry to disappoint there but neither the new firmware nor the new battery profile did repare the probleme for my 12.000 battery. With the new profile and the new firmware, my battery reading is still 65210°C :) Battery has been correctly charging and discharging for 2 month now without any problems. 

I installed firmware as stated in the 'install from command line' ==> [v1.5]
Went to chek the battery profile to input as custom : LIPO_12000_PIS-1129.JSON

Am i missing something ?

tvoverbeek, 2021-03-16:
The config in LIPO_12000_PIS-1129.JSON is for a batch of 12000 mAh batteries which have a deviating NTC resistance causing wrong temperature readings. The original ones should use the built-in profile.
As a work-around change the temperature sense setting to NOT_USED. This will force a temp of 25 deg C for the firmware.

S-lac, 2021-03-16:
Hi ! 

Thanks for the answer, I had the problem on my 12000 mAh since 3 months: wrong temperatures readings, and have been using the 'NOT_USED' ever since (see issue #606). I thought that with the new profile and firmware it was going to get corrected but in fact the change is from a -102 °C value to a 65210°C. Neither the initial 12000 or the new one does the trick.  

As my setup needs to go in the field in a month, I might try something else to correct this.

tvoverbeek, 2021-03-16:
@S-lac The 1.7 deb packages are now in the raspbian repos. Please update.
The updates to pijuice_cli and pijuice_gui are needed to get the correct temperature readings from the 1.5 firmware.
From the 1.7 release notes:
- Fixed Bad signed conversion in function GetBatteryTemperature.

Please update and see if you get more reasonable battery temperatures.