PiSupply/PiJuice #607: Set I2C address over I2C

Nicehavetoo, 2020-12-12:
Goodday ya'll,

I'd like to be able to set the I2C address over de I2C bus.
How do I do this?
Which register do I need to write for this?

Nice_to_have guy

Nicehavetoo, 2020-12-12:
[image]

Which one do I need to write with the  i2cset -y 1 command

tvoverbeek, 2020-12-12:
Ir is not clear to me what you want to do with i2cset.

If you want to change the I2C addresses of the PiJuice (defaults: main = 0x14, RTC = 0x68) look at the pijuice.py source code: https://github.com/PiSupply/PiJuice/blob/a050850c1295d2c34198f1217103c94a7a44dce9/Software/Source/pijuice.py#L1488
slave=1 is the main PiJuice address, slave=2 is the RTC address.
The message has to include a checksum. See the WriteData routine and its low level routines.