PiSupply/PiJuice #608: Pi Juice HAT + 12 AH battery temperature + charging issues

kubiaj, 2020-12-15:
Hi all,
I have a RPIv4 + PiJuice HAT + 12Ah battery  (preset via onboard switch) and I have a following issues:

1) battery constantly report wrong temperature cca 46°C  but the real temperature is about 22°C  (measured independently on the PCM board for a 30minutes)
2) never stops charging - in the GUI it keep changing the battery status between CHARGING_FROM_IN and NORMAL. It is powered directly from RPI power supply for a +5 days. In Friday it has about 95% and now it have about 99%.

Any idea what is wrong?

kubiaj, 2020-12-15:
One additional note - I have measured that the current it toggling between 3mA and 0.2mA but the Termination current is set to 350mA  (PJLIPO_12000 profile) so it should be considered as Charger a long time ago.

S-lac, 2020-12-16:
- I assume you potentially have the same problem as me. Can you try setting 'temperature sensor' to 'NOT_USED' in the battery profile ? 
- Have you tried charging it from the PiJuice input to see if it changes anything ?

kubiaj, 2020-12-16:
Hi,

- Can you try setting 'temperature sensor' to 'NOT_USED' in the battery profile ?
I have tried it and it keeps changing the charging status. Just the temperature is 25C. But I have notified that there is no difference between NTC and ON_BOARD setting.  But wiring is correct. The board could have 45C.

- Have you tried charging it from the PiJuice input to see if it changes anything ?
Yes, no difference (the same for USB or external input).

- Do you have the same problems after a reboot of the pi4 ?
Yes for many times.

snecklifter, 2020-12-16:
I'm also having this issue. Will test the above and report back my findings.

snecklifter, 2020-12-16:
The 1000mAh battery I had ordered arrived today. I swapped them out and it started charged without issue. I'm therefore assuming the 1200mAh model I received is part of a bad batch and have raised a return.

snecklifter, 2020-12-16:
I had the same issue with the 1000mAh not charging however changing "Temperature sense" to NOT_USED on the Battery tab of the HAT configuration has switched the state to CHARGING_FROM_IN and the % is increasing.

shawaj, 2020-12-16:
@snecklifter which battery is it that you had the issues with? You say 1200mAh but the one mentioned in original issue is the 12000mAh

Could you link to the one you're talking about?

snecklifter, 2020-12-16:
@shawaj definitely the PiJuice Zero 1200mAh, not the 12000.

shawaj, 2020-12-16:
@snecklifter Do you have the sku from the packaging box of the battery? Or a link

snecklifter, 2020-12-16:
> @snecklifter Do you have the sku from the packaging box of the battery? Or a link

PIS-1189 <- is that it?
There is also a number 123035

shawaj, 2020-12-17:
Pis-1189 is the one, thanks.

So your issue was the temperature reporting incorrectly @snecklifter ?

kubiaj, 2020-12-17:
Hi, 
my bat is LITER 9068125.
I have measure 10,237 kOhm on the NTC  (Yel + black cable) and from online calculator this should be cca 24,4C but it shows cca 46C (it is the same value what it shows for onboard NTC).

Is something wrong with my HAT?

snecklifter, 2020-12-17:
> Pis-1189 is the one, thanks.
> 
> So your issue was the temperature reporting incorrectly @snecklifter ?

Yes. Is this a battery issue or a HAT issue?

snecklifter, 2020-12-17:
@shawaj the 1200mAh is now charging after reconnecting it. I'll cancel the return but would be good to get temperature sensor working before this goes into the field.

shawaj, 2021-01-01:
@snecklifter we have found the issue with the NTC being incorrect spec. Hopefully we should be able to get out reporting correctly by changing some parameters in the battery profiles pages. Will report back once we have more

kubiaj, 2021-01-03:
@shawaj  it there any chance that it will be also solution for my issue - it wont stop charging the battery. The battery life time is OK, charging  up to 90~95% is also ok, but the last few percent takes ages (many many days). I would believe that it will not stop at given Termination current and it keep it charging.  The independent measurement of charging current show about 3~6mA charging current but at this Termination current is set to about 350mAh.... Any idea what is wrong?

shawaj, 2021-01-03:
@kubiaj I feel like I've seen this issue before and that there was a resolution but can't seem to find the issue.

@tvoverbeek do you remember this at all? I think someone had a similar issue before with it but getting to 100%.

shawaj, 2021-01-08:
@kubiaj I guess you have got the device on when you are taking this reading? I guess it can't reach 100% when it is on

kubiaj, 2021-01-08:
@shawaj  no the device is turned off during the charging. 

I dont mind if it is charged to 98% or 100% I need to turn off the blue led when it is close to the charged state.  Our device have no display - I see only LED1 - the status led.

snecklifter, 2021-01-08:
> @snecklifter we have found the issue with the NTC being incorrect spec. Hopefully we should be able to get out reporting correctly by changing some parameters in the battery profiles pages. Will report back once we have more

Thanks for the update. Happy to test whenever.

kubiaj, 2021-01-25:
Hi @shawaj  @tvoverbeek 
I need to move on and pass our device to customer. I still have non working battery temperature sensing (less important issue) but the major issue is that the charging never stops. I didnt find a way how to force the blue LED stop blinking.

The use case: Battery fully discharged -> start charging while the RPI is  turned off - BLUE led is blinking and it never stop blinking even after days.  I need to have a way how the customer could determine that it is fully charged (actually I don't care what the fully charged means, lets say anything above 95% would be OK). This issue is show stopper for us. 

So far I only once saw that it reach 100% but it keeps charging  and  toggling between CHARGING_FROM_IN and NORMAL. I can remeasure the current, but in the last time it was charging with 3mA~6mA current - so it is already charged.

kubiaj, 2021-01-25:
FYI: I tried to set CHARGE STATUS for a Blue to 95 but nothing happens (https://github.com/PiSupply/PiJuice/tree/master/Software#pijuice-hat-config-leds-menu).

tvoverbeek, 2021-01-25:
If you mean the blue setting on the LED menu, that is only the blue intensity when blue is on.
What you could do is to run a job polling the charge level regularly (e.g. once every few minutes) and when the charge level passes 95%, disable the charging.
Of course, you need to enable charging again when the charge level drops below a certain threshold (50%??)

kubiaj, 2021-01-28:
Hi @tvoverbeek 
I have missed github notification...   The rpi is turned off during the charing period so I cannot pool the status. In other hand if I will let it powered on then the RPI4 consumes incresdible big ammount of energy (between 400mA ~1.5A, depends on running taks, and we are also powering external device via RPI USB) so charging will take a super long time. 

Main question what is wrong with the board? I have ordered a second pair of HAT + 12Ah battery I will retest it.

tvoverbeek, 2021-01-28:
OK. How do you switch off the Pi before charging? Is the 5V from the GPIO to the Pi switched off? If you do a 'normal' shutdown from the command line or a script the 5V from the PiJuice is not switched off (unless you usw the Software Halt Power Off option) and your USB peripherals might still use power.
If you to charge to only 90-95% you could use the wakeup-on-charge option and then disable charging.

kubiaj, 2021-01-28:
@tvoverbeek  I use SW1 for powering off - so the connected USB peripheral are also powered off.   The whole device have only one button for powering on and off. 

"If you to charge to only 90-95% you could use the wakeup-on-charge option and then disable charging." this looks like a solution (workaround). However what is the problem behind? It should work out of box as the HAT have profile for 12Ah battery and I bought boot from your distributor.

shawaj, 2021-03-15:
closing in favour of #659

closed 2021-03-15 by shawaj