PiSupply/PiJuice #609: PIZero Weather station

bbakerzzz, 2020-12-16:
Looking for some help.

My current setup is 
	PIZreo
	PIJuice Zero
	BC Robitices Weather hat.

I have done all software and kernel updates which address the shutdown issue after a few minutes of being booted up while using the PIJuice Zero or BC Robitices Weather hat.  The issue I am having now is it appears there is a conflict between the PIJuice and the BC Robotics Weather Hat.  When the Weather Hat and the PIJuice are stacked together I get communication error for all the PIJuice software.

Is there a way I can fix this?

Thanks
Bryon

bbakerzzz, 2020-12-16:
I get the same issue when I use the PIJuice non Zero Version.

shawaj, 2020-12-16:
@bbakerzzz is this the board you are talking about? https://bc-robotics.com/shop/raspberry-pi-weather-board/