PiSupply/PiJuice #610: Communication Error

mikehooper, 2020-12-16:
I've got 2x Pi4 2GB each with a PiJuice HAT. Running the latest Raspberry Pi OS (fully updated). Both PiJuice have latest firmware.

Both Pi's experience the 'communication error' in the Pi Juice, Settings, HAT tab.

When the error occurs 'i2cdetect -y 1' pauses before returning its data.

```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.4.79-v7l+ #1373 SMP Mon Nov 23 13:27:40 GMT 2020 armv7l GNU/Linux
pi@raspberrypi:~ $ 

```

[image]

[image]

shawaj, 2020-12-16:
@mikehooper do you have any other boards attached to the Pis?

mikehooper, 2020-12-16:
One Pi4 has a custom board attached, the other only has the PiJuice.

mikehooper, 2020-12-17:
I've setup a Pi3 with PiJuice and latest OS, not seen the communication error so far.

shawaj, 2020-12-17:
@mikehooper interesting so it seems to be something to do with the pi 4. 

@tvoverbeek do you think this could be to do with the clock speed issue you had mentioned in another thread?

mikehooper, 2020-12-17:
I believe the ZymBit/ZymKey suppliers have had issues with the Pi4 clock speed changes

shawaj, 2020-12-17:
Ah that's interesting. Is there a github issue (or similar) anywhere for that?

mikehooper, 2020-12-17:
https://community.zymbit.com/t/dev-team-known-issues/996

tvoverbeek, 2020-12-17:
@shawaj @mikehooper Yes, the issue mentioned in https://community.zymbit.com/t/dev-team-known-issues/996 is the same one (scaling governor).
The solution in https://community.zymbit.com/t/dev-team-known-issues/996 is equivalent to fixing the core frequency in /boot/config.txt.

mikehooper, 2020-12-18:
Changing the cpu governor from on demand to performance does appear to solve the issue.

Can this be raised with the Pi foundation?

I have created a post: https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=295232

shawaj, 2020-12-18:
It might be worth posting an issue on RasPi Linux repo and linking back to hear any zymbit.

https://github.com/raspberrypi/linux/issues

piPixel29, 2021-02-08:
Hi,
I have the same error message with my brand new PiJuice and my Pi4, but changing cpufreq governor has no effect.
Plus I think the (new) battery is bad because as soon as I unplug the power cord, capacity immediatly drops from 94% to 60%.

Do you think a bad battery could cause the communication_error message ?
Thanks

CometSL9, 2021-03-09:
Hi,

I've got two PiJuices and a Pi 4B with the same issue.

Setting the governor did not change anything.

What did help somehow was to put 
`dtoverlay=i2c-bcm2708`
into the /boot/config.txt. That uses the older i2c-bcm2708 module instead of the default i2c-bcm2835.

Seems that the default I2C communication of the RPi is written a little bit "unclean".

Best regards,
   Carsten

LGM29, 2021-03-09:
Hi,
dtoverlay did not solve the issue here.
Neither did changing CPU governor.

Adding these lines in config.txt did make the PiJuice work:
`core_freq=500
core_freq_min=500`

HTH