PiSupply/PiJuice #611: RTC detection not stable on Pi4

lotte67890, 2020-12-17:
Hello again, I´ve got more or less the same problem as #563, although I use the Pijuice on a Pi4B (4GB) with latest Buster OS. The chances that the RTC is recognized properly are 50%. `i2detect -y 1` then shows `68` instead of `UU` and it does not matter whether `dtoverlay=i2c-rtc,ds1339` exists in `/boot/config.txt` or not. It does not look like the issue mentioned in #610 cause the Gui always shows proper Information. Let me mention that the Pi boots from an SSD, so the OS boot process takes approximately 10 sec.
As @tvoverbeek mentioned https://github.com/PiSupply/PiJuice/issues/563#issuecomment-712768692 , would it help to first detect that the Firmware is loaded before the first OS to RTC request occurs (in the mean of once one request to the RTC goes wrong all the next go wrong as well)?
And is there a way to (re)load the RTC after the OS has started (rmmod, insmod/modprobe)?

Thanks a lot

L

tvoverbeek, 2020-12-17:
@lotte67890 How do you power your setup? If you power via the RPi and start from completely off (no 5V to the RPi) then the PiJuice  initialization might be too slow for your fast booting system (Pi4 booting from SSD). Hence the 50/50 chance detecting the RTC.

lotte67890, 2020-12-17:
You name it, that´s exactly what I do. Would a power supply directly to the pijuice make a difference? The system is installed in my car, so that´s why I can´t leave it powered on whilst not driving..