PiSupply/PiJuice #612: PiJuice v1.5 firmware release

shawaj, 2020-12-24:
@mmilann has just released updated firmware V1.5...

- Wake-up on charge restore.
- Watchdog restore.
- Extended maximum possible value range for battery capacity configuration to  [0-4194175mAh].
  - Added new profile for PiJuice 2500mAh battery.
  - Introduced experimental FreeRTOS

There are full descriptions of changes in Change log. 

It will be good that more people test it. 

There are also software changes and @mmilann  will see to commit them directly to github. It will be V1.7 with firmware V1.5 integrated, so people  can install as package to test and evaluate.

[PiJuice.zip](https://github.com/PiSupply/PiJuice/files/5740286/PiJuice.zip)
[Change log.txt](https://github.com/PiSupply/PiJuice/files/5740287/Change.log.txt)
[PiJuiceFirmwarev1_5.zip](https://github.com/PiSupply/PiJuice/files/5740290/PiJuiceFirmwarev1_5.zip)

A little early Christmas present for you @ryanteck @tvoverbeek

mmilann, 2020-12-24:
@tvoverbeek any potential issues to commit software updates to master branch? Maybe to create tag?

Most updates are to pijuice_sys.py:
    - Fixed undesired behavior of system task missing to halt system if input power 
    is lost during system boot. noPowCnt initialized to 0 instead of 100 so 
    _EvalPowerInputs function triggers function execution if there is "no power input"
    at startup.
    - Fixed undesired behavior of system task triggering low charge or low battery 
    events when battery is missing or is bad immediately after startup. This false 
    triggers were noticeable when "no battery turn on" config is enabled and system 
    halt on "low battery voltage" or "low charge" events are configured, so it halts
    immediately after startup. _EvalBatVoltage, _EvalCharge and _EvalPowerInputs are
    updated to detect this case and avoid undesired triggers.
    - Introduced two additional system events: "sys_start" triggers at system task 
    startup and "sys_stop" triggers at system task stop. These events can be used
    to run custom user scripts like automated PiJuice board configuration.
    - Update to restore PiJuice I2C Address from configuration file.
    - Fixed double watchdog activation at startup.

mmilann, 2020-12-24:
Base package for test.
[pijuice-base_1.7_all.deb.zip](https://github.com/PiSupply/PiJuice/files/5740676/pijuice-base_1.7_all.deb.zip)

shawaj, 2020-12-25:
@mmilann can you commit to the v1.7 branch I just created? https://github.com/PiSupply/PiJuice/tree/v1.7

tvoverbeek, 2020-12-25:
@mmilann Agreeing with @shawaj. let us work on the V1.7 branch. When we are satisfied with testing we can merge back to the main branch and make an official release.

mmilann, 2020-12-25:
Current issue is building gui package error: source package has two conflicting values - pijuice-gui and pijuice-base

shawaj, 2020-12-26:
@mmilann you are having an issue building the debs?

mmilann, 2020-12-26:
@shawaj yes, just someone to clone v1.7, run pckg-pijuice.sh and confirm, maybe something missed to update.

tvoverbeek, 2020-12-26:
@mmilann Found the problem. The changelog for the gui (https://github.com/PiSupply/PiJuice/blob/v1.7/Software/Source/debian-gui/changelog) has in line 1:
`pijuice-base (1.7) unstable; urgency=low`
should be
'pijuice-gui (1.7) unstable; urgency=low
Already fixed on the V1.7 branch.
Then the pijuice-gui package builds OK.

mmilann, 2020-12-26:
@tvoverbeek thanks, probably formatting copy-paste mistake.

shawaj, 2021-01-01:
Added a fix for #614 to the v1.7 branch https://github.com/PiSupply/PiJuice/commit/e2dca1f8dcfa12e009952a882c0674a545d193d6

shawaj, 2021-01-01:
Closed by mistake

shawaj, 2021-01-01:
@tvoverbeek you know this change https://github.com/PiSupply/PiJuice/commit/2ead4d82aebd73cdfc6a48fc443a0084e5daa786

Will people still need that warning if they are below v1.4 firmware? 

I guess most of them will have either upgraded or added the config.txt like by now but just thinking there may be a few who haven't for whatever reason.

ApuliaMonitoringNetwork, 2021-01-04:
Many thanks for this cool upgrade and your developments.  Can you please you guys suggest the best and simple way/approach to download and test V1.7 with firmware V1.5 integrated, without creating any conflicts with old packages (for dummies, let's say). It's just a matter of time of installation, can't wait for official release.
Thanks again.

tvoverbeek, 2021-01-04:
@ApuliaMonitoringNetwork Still testing the changes and correcting the code for the RPi side of things.
Right now there are no deb packages available on the v1.7 branch on Github.
If you are really adventurous you can build the packages yourself from the v1.7 branch using the pckg-pijuice.sh script (https://github.com/PiSupply/PiJuice/blob/v1.7/Software/Source/pckg-pijuice.sh).

ApuliaMonitoringNetwork, 2021-01-05:
@tvoverbeek Thanks.

shawaj, 2021-03-13:
@mmilann @tvoverbeek what are your thoughts on moving v1.7 software to master now?

tvoverbeek, 2021-03-13:
@shawaj As far as I am concerned the 1.7 debs in their current state can be released.
Note: documentation is not updated yet (e.g. the store option for wakeup-on-charge and watchdog).

mmilann, 2021-03-13:
@shawaj Package can be merged and released. Documentation about new features is missing, but it can be updated to master branch.

closed 2021-03-15 by shawaj