PiSupply/PiJuice #614: Python Version >3.9: AttributeError: 'Thread' object has no attribute 'isAlive'

3urex, 2021-01-01:
Hi & happy new year,
I just updated my kali linux and its python installation to 3.9.1. After the update, pijuice stopped working returning the error "Thread object has no attribute 'isAlive'".
I figured out that the method has been renamed to is_alive with python version 3.9.  Maybe that is causing the problems?

shawaj, 2021-01-01:
I would imagine that is the issue yes. Do you need python 3.9.1 for any reason in particular?

shawaj, 2021-01-01:
Ok so further investigation shows that this was deprecated in 3.8 and removed in 3.9

```
Thread.isAlive() is deprecated in 3.8 and removed in 3.9:

    $ python3.8 -c 'import threading; t = threading.Thread(); t.isAlive()'
    <string>:1: DeprecationWarning: isAlive() is deprecated, use is_alive() instead

    $ python3.9 -c 'import threading; t = threading.Thread(); t.isAlive()'
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
    AttributeError: 'Thread' object has no attribute 'isAlive'
```

shawaj, 2021-01-01:
Have added a fix for this to the v1.7 branch. Will leave this open for now but @3urex if you modify the pijuice.py file with the changes here https://github.com/PiSupply/PiJuice/commit/e2dca1f8dcfa12e009952a882c0674a545d193d6

Then it should work fine for you.

Would you be able to test this? 

And let us know if you see any other errors too!

3urex, 2021-01-01:
Thanks for your quick reply. Changing the three occurrences to is_alive fixed the problem for me.

shawaj, 2021-01-01:
@3urex awesome thanks very much for testing 👌

shawaj, 2021-03-17:
Closing based on updates in #678 that fixes this

closed 2021-03-17 by shawaj