PiSupply/PiJuice #616: Diode D1 after three days stay always on blue color

IW2FAG-11, 2021-01-02:
This PIJuice HAT arrived five days ago and it looks like something has already broken.
This morning I found the led diode D1 fixed at the blue color. It usually flashes between green and blue while the battery is charging and stays green when the battery is charged. To verify that it had no configuration problems, I configured D1 as NOT_USED in PIJuice and configured LED D2 as CHERGE_STATUS. With this configuration I see the LED D1 always fixed at the blue color and the diode D2 which remains green when the battery is charged and flashes between blue and blue during charging. I suspect that something has broken in the management of led D1.
Firmware is at review 1.4 V1.4_2020_01_17.elf.binary

Below I attach photos of the problem
[image]
[image]
[image]

I have another PIJuice which has been running for about two years and still works perfectly.

tvoverbeek, 2021-01-02:
For the chargelevel led the colors you see depend on the values you have defined for R, G and B: Rval, Gval and Bval
```
For chargelevel < 15%:                             LedR = Rval, LedG = 0
For chargelevel between  15% and 50%: LedR = Rval, LedG = Gval
For chargelevel > 50%:                             LedR = 0,    LedG = Gval
When charging:                                     LedB alternates between 0 and Bval
When charging is complete:                         LedB = Bval
```
The default values for Rval, Gval and Bval are 60, 60, 100.

When you specify `NOT_USED` for a led it should be off.

So it definitely seems something is broken for led D1

shawaj, 2021-01-02:
Maybe a firmware bug @tvoverbeek @mmilann ?

I can't seem to reproduce the issue though.

IW2FAG-11, 2021-01-02:
I think the problem may be hardware.
For a couple of days everything worked perfectly with the D1 led.
While charging, the green and blue lights will flash.
But from this morning solid blue whatever you do, even if I detach the HAT from the rasp (3B +)

mmilann, 2021-01-02:
In my case it is off with same settings. May be accidentally broken when SW is pushed. Visually inspect solder points around LED.

IW2FAG-11, 2021-01-02:
I took a look around the led and around the Q2 mosfet. Everything looks fine, but not having a microscope I have a hard time seeing every micro detail. In my case, the LED is always on, so unless the Mosfet has broken (drain source short circuit), there must be a continuous drive at the gate.
[image]

IW2FAG-11, 2021-01-03:
**Where i have to send back mine to get repaired ?**

IW2FAG-11, 2021-01-03:
pi@raspberrypi:~ $ python3
Python 3.7.3 (default, Jul 25 2020, 13:03:44) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import time
>>> import datetime
>>> import calendar
>>> import sys, getopt
>>> from pijuice import PiJuice
>>> pijuice = PiJuice(1,0x14)
>>> 
>>> def checkValue(val):
...     val = val['data'] if val['error'] == 'NO_ERROR' else val['error']
...     return val
... 
>>> status = checkValue(pijuice.status.GetStatus())
>>> print('Status = ', status) 
Status =  {'isFault': False, 'isButton': False, 'battery': 'NORMAL', 'powerInput': 'NOT_PRESENT', 'powerInput5vIo': 'PRESENT'}
>>> fault =  checkValue(pijuice.status.GetFaultStatus())
>>> print('Fault = ', fault)
Fault =  {}
>>> 
>>> charge = checkValue(pijuice.status.GetChargeLevel())
>>> fault =  checkValue(pijuice.status.GetFaultStatus())
>>> temp = checkValue( pijuice.status.GetBatteryTemperature())
>>> vbat = checkValue(pijuice.status.GetBatteryVoltage())
>>> ibat = checkValue(pijuice.status.GetBatteryCurrent())
>>> vio =  checkValue(pijuice.status.GetIoVoltage())
>>> iio = checkValue(pijuice.status.GetIoCurrent())
>>> print('Charge =',charge,'%,', 'T =', temp, ', Vbat =',vbat, ', Ibat =',ibat, ', Vio =',vio, ', Iio =',iio)
Charge = 98 %, T = 24 , Vbat = 4167 , Ibat = -48 , Vio = 5023 , Iio = -476
>>> regmode = checkValue(pijuice.config.GetPowerRegulatorMode())
>>> print('RegulatorMode =', regmode)
RegulatorMode = POWER_SOURCE_DETECTION
>>> runpin = checkValue((pijuice.config.GetRunPinConfig()))
>>> print('RunPin =', runpin)
RunPin = NOT_INSTALLED
>>> sysswitch = checkValue(pijuice.power.GetSystemPowerSwitch())
>>> print('SystemPowerSwitch =', sysswitch)
SystemPowerSwitch = 0
>>> ledconfig = checkValue(pijuice.config.GetLedConfiguration('D1'))
>>> **print('LedConfigD1 =', ledconfig)
LedConfigD1 = {'function': 'NOT_USED', 'parameter': {'r': 0, 'g': 0, 'b': 0}}**
>>> ledblink = checkValue(pijuice.status.GetLedBlink('D2'))
>>> print('LedBlinkD2 =', ledblink)
LedBlinkD2 = {'count': 0, 'rgb1': [150, 0, 0], 'period1': 200, 'rgb2': [0, 100, 0], 'period2': 200}
>>> but3 = checkValue(pijuice.config.GetButtonConfiguration('SW3'))
>>> print('ButtonSW3 =', but3)
ButtonSW3 = {'PRESS': {'function': 'USER_FUNC3', 'parameter': 0}, 'RELEASE': {'function': 'USER_FUNC4', 'parameter': 0}, 'SINGLE_PRESS': {'function': 'NO_FUNC', 'parameter': 0}, 'DOUBLE_PRESS': {'function': 'NO_FUNC', 'parameter': 0}, 'LONG_PRESS1': {'function': 'NO_FUNC', 'parameter': 0}, 'LONG_PRESS2': {'function': 'NO_FUNC', 'parameter': 0}}
>>> wakeuponch = checkValue(pijuice.power.GetWakeUpOnCharge())
>>> print('WakeUpOnCharge =', wakeuponch, '(127 = disabled)')
WakeUpOnCharge = 127 (127 = disabled)
>>> watchdog = checkValue(pijuice.power.GetWatchdog())
>>> print('Watchdog =', watchdog)
Watchdog = 0
>>> buttonevents = checkValue(pijuice.status.GetButtonEvents())
>>> print('ButtonEvents =', buttonevents)
ButtonEvents = {'SW1': 'NO_EVENT', 'SW2': 'NO_EVENT', 'SW3': 'NO_EVENT'}
>>> batprofilestat = checkValue(pijuice.config.GetBatteryProfileStatus())
>>> print('BatteryProfileStatus =', batprofilestat)
BatteryProfileStatus = {'validity': 'VALID', 'source': 'HOST', 'origin': 'PREDEFINED', 'profile': 'BP7X_1820'}
>>> batprofile = checkValue(pijuice.config.GetBatteryProfile())
>>> print('BatteryProfile =', batprofile)
BatteryProfile = {'capacity': 1820, 'chargeCurrent': 925, 'terminationCurrent': 50, 'regulationVoltage': 4180, 'cutoffVoltage': 3000, 'tempCold': 1, 'tempCool': 10, 'tempWarm': 45, 'tempHot': 59, 'ntcB': 3380, 'ntcResistance': 10000}
>>> batextprofile = checkValue(pijuice.config.GetBatteryExtProfile())
>>> print('Extended BatteryProfile =', batextprofile)
Extended BatteryProfile = {'chemistry': 'LIPO', 'ocv10': 3649, 'ocv50': 3800, 'ocv90': 4077, 'r10': 209.0, 'r50': 205.0, 'r90': 202.0}
>>> tempsense = checkValue(pijuice.config.GetBatteryTempSenseConfig())
>>> print('Battery temperature sense =', tempsense)
Battery temperature sense = AUTO_DETECT
>>> 
>>> pjaddr = checkValue(pijuice.config.GetAddress(1))
>>> print('PiJuice I2C address =', pjaddr, '(hex)')
PiJuice I2C address = 14 (hex)
>>> eepromwrprot = checkValue(pijuice.config.GetIdEepromWriteProtect())
>>> print('HAT eeprom write protect =', eepromwrprot)
HAT eeprom write protect = True
>>> eepromaddr = checkValue(pijuice.config.GetIdEepromAddress())
>>> print('HAT eeprom address =', eepromaddr, '(hex)')
HAT eeprom address = 50 (hex)
>>> fwver = checkValue(pijuice.config.GetFirmwareVersion())
>>> print('Firmware Version =', fwver)
Firmware Version = {'version': '1.4', 'variant': '0'}

**With this hw state, mine D1 led still on and seems to be at his maximum power. I think thak Q2 is broken or there is a cc short circuit on some net.**

For a good job on mosfets, is advisable to put a resistor in series on gate from source to avoid latch up and an high 1M or so connected to gnd or vss to keep off.

IW2FAG-11, 2021-01-20:
After changing MosFet Q2 that control D1, the problem was solved.
Thanks for data of Q2.

IW2FAG-11, 2021-01-20:
Solved, Thanks

closed 2021-01-20 by IW2FAG-11

shawaj, 2021-01-20:
@IW2FAG-11 can you email sales@pi-supply.com with your order details as I would like to provide some reimbursement for your issue.