PiSupply/PiJuice #617: 12000mah battery not charging

xxSPEEDFREAKxx, 2021-01-02:
I have a pijuice zero with a 12000mah battery and no matter what I do I can not get it to charge. I don't know if there's a something I'm missing or if anyone can provide an instructional to help but I really need assistance asap please. I've tried connecting the power to the pi and to the pijuice board and it will not charge please help.

xxSPEEDFREAKxx, 2021-01-02:
[image]
[image]
[image]
This is currently connected to a pi 3b+ (project was originally supposed to run off of zero but couldn't) and this is also connected to the pi through a pimaroni hacker hat that has 2 extra gpio extensions. Battery is cold, not receiving any charge.

shawaj, 2021-01-02:
This looks like the same issue as #608 and #606 I think. It's because the temperature sensor is reading wrong.

We are working on a fix

shawaj, 2021-01-02:
@xxSPEEDFREAKxx by the way you should be using the 12000mAh battery profile not the 1000mAh one you have currently.

And you can override the temperature sensor in the little drop-down there as well.

xxSPEEDFREAKxx, 2021-01-02:
I had tried changing to the 12000mah but it didn't make a difference unless
I need to reboot after. Also in what way am I adjusting the temperature?
And when you say an issue you're currently working on is there going to be
like a system update going out/patch to download or software adjustments to
make on our/user end? (Adjustments we make on our end)

shawaj, 2021-01-02:
In the screenshot you posted above there is an option `temperature sense` you can change that to the other setting.

And yes we will have some new settings for you to apply soon that will fix the temperature detection

tvoverbeek, 2021-01-02:
If you use 'NOT_USED' for temperature sense the firmware will use a fixed temperature of 25 degC and the battery will charge. Now the firmware wrongly believes the high temperature which is above the hot temperature and the battery will not charge.

S-lac, 2021-01-03:
@xxSPEEDFREAKxx As stated above, I have the same problem and dealt with it by using the "NOT_USED", option in the drodown menu "TEMPERATURE SENSE". After this the battery charges / discharges correctly. I believe every time I do the switch between battery profile / parameters it should be better to reboot the system. @shawaj Is the battery temperature a linear expression of NTC B and resistance parameters ? Because in that case I can probably measure battery temperature manually and try to fit the parameters.

shawaj, 2021-01-03:
@S-lac that's probably a better question for @mmilann

mmilann, 2021-01-03:
NOT_USED can be selected so it does not disable charging on false
temperature measurement, also or ON_BOARD option can be selected, if
battery is sticked to pijuice it's temperature is nearly the same as board
temperature. Temperature is calculated based on two parameters NTC B and
R0(resistance at 25C), that are for this new battery different then ones in
the current 12000mAh profile.

xxSPEEDFREAKxx, 2021-01-03:
That fixed the issue thanks everyone.

closed 2021-01-03 by xxSPEEDFREAKxx