PiSupply/PiJuice #618: Pijuice not waking on alarm and not detected via i2cdetect -y 1

budlee85, 2021-01-03:
Hi all,

New here but recently set up a pijuice on a pi zero w that was successfully working initially and set up some alarms using python via I2C Command API as per the software document.  It is a 4G connected timelapse camera which uploads after every photo so has timesyncd to keep track of time.  Ideally I want it to wake every 10 minutes, run the code (~1 min) and shutdown before running again on the next 10-min cycle.  Will be solar powered and running a 20000mah LIPO but right now it's powered with the battery and recharged every so often via pijuice usb in

Just recently I noticed it has been misbehaving and not waking despite setting wake and a wake using minute period. On further digging through suggestions on other issue threads I noticed that i2cdetect -y 1 is not giving me UU on 68.  I've tried various solutions like
1. Changing ID EEPROM to 52 and dtoverlay=i2c-rtc,ds1339 (worked the first time then broke)
2. Changing ID EEPROM to 52 and adding dtoverlay=i2c-rtc,ds1307 (no luck)
2. Subsequently adding dtoverlay=i2c-bcm2708 to use an older module (had success but on subsequent boot reverts back to 68)
3. Running echo ds1307 0x68 | sudo tee /sys/class/i2c-adapter/i2c-1/new_device (worked once but on subsequent boot reverts back to 68 and then running this again says device is busy (rather than the Invalid Argument in the previous issue thread)
4. Using a fresh updated install (as of 3 Jan 2021) and installing pijuice-base (pijuice hardware kept its original settings from previous install)

Currently /boot/config.txt has the following additions (no other change)
dtoverlay=i2c-rtc,ds1307
dtoverlay=i2c-bcm2708

Result of 
print(pijuice.rtcAlarm.GetControlStatus())
{'data': {'alarm_wakeup_enabled': True, 'alarm_flag': True}, 'error': 'NO_ERROR'}
print(pijuice.rtcAlarm.GetAlarm())
{'data': {'second': 0, 'minute_period': 10, 'hour': 0, 'day': 'EVERY_DAY'}, 'error': 'NO_ERROR'}

My power off script always ensures the alarm wakeup is enabled
pijuice.rtcAlarm.SetWakeupEnabled(True)
print(pijuice.rtcAlarm.GetAlarm())
print(pijuice.power.SetPowerOff(15))

even with a simple sudo shutdown now after the alarm is set doesn't bring it up when the alarm should

Other things i've found are
dmesg | grep rtc
rtc-ds1307: probe of 1-0068 failed with error -5

I'd like to clean start the pijuice but not sure if there is a 'factory reset', there was mention in a previous issue that the EEPROM might get damaged but I'm not sure how to check if this has happened or reset it.

Kinda stumped and I was hoping a fresh raspberrypi os would get me fixed but maybe the problem is in the parameters stored in the pijuice?

shawaj, 2021-01-03:
Have you tried reloading the firmware using the GUI?

Also some other info here https://github.com/PiSupply/PiJuice/tree/master/Firmware

Lastly, you say you are using a 20,000mAh battery. Is this a single cell lipo?

tvoverbeek, 2021-01-03:
When you are on the current firmware (1.4, and it seems you are since you (re)installed pijuice-base) then you do not need the `dtoverlay=i2c-bcm2708` line in /boot/config.txt.
The problem seems to be the simulated RTC in the firmware is initialized too late in the boot sequence for the OS to recognize the RTC.
Anyway, for the repeated wakeup you do not need OS support.
Only problem remains time synchronization. With your relative short run time timesyncd might nor run in time (4G needs to be up). For time sync of the RTC by the OS you do need the ds1307 support.
Have you checked if the RTC has the correct UTC time?

budlee85, 2021-01-04:
Thanks for the quick responses guys.  Did some more playing around since I posted and had some good success.

I'm using headless so I tried resetting settings using pijuice_cli again.  I think the first time i tried it didn't work properly as i noticed this time that the pijuice zero both leds turn blue during the reset.  I am back on 0x50 with only dtoverlay=i2c-rtc,ds1339 in /boot/config.txt.  The documentation says I need to do that only if i change to 0x52 but for me I need it even using 0x50.  Also, it may take a few resets before it becomes 'UU' so I am wondering perhaps it is a boot timing issue?

I am ideally trying to keep powered time short but timesyncd seems to keep the pi time correct and i've not noticed any errors with RTC in pijuice_cli either.  Do I need to have the RTC loaded with 'UU' as I'm updating the time via the internet and only need the RTC for wakeup alarm (i.e. not to bring an offline PI time back in sync)?

Regarding the wakeup alarm I figured out I struggled with the documentation and found out how to set repeated wakes correctly.  In order for minutes_period to run consecutively I also needed to change hour to EVERY_HOUR which wasn't as intuitive as I thought.  Perhaps a crontab format standard might be easier?
{'data': {'second': 0, 'minute_period': 10, 'hour': 'EVERY_HOUR', 'day': 'EVERY_DAY'}, 'error': 'NO_ERROR'}

I also struggled to get a regular set alarm time working though however whilst typing this reply I had an epiphany that the alarm had to be on UTC-time not what the locale the Pi is at which requires a bit more mental conversion and may run into daylight savings issues?  Perhaps a note in the documentation might help more newbies realise the pijuice operates on UTC whilst the pi OS operates on a separate local time.

@shawaj , FYI the battery I'm using is https://core-electronics.com.au/3-pin-lipo-battery-for-pijuice-20000mah.html1.  Which doesn't have a battery profile but I copied the 10000mah profile and changed the capacity to 20000mah.  Hopefully that should be ok?

tvoverbeek, 2021-01-04:
Good  to hear you found out how to make it work.

You wrote: `The documentation says I need to do that only if i change to 0x52 but for me I need it even using 0x50.`
The 0x50 is the address of the ID-EEPROM. This only applies to the PiJuice HAT, not the PiJuice Zero.
The Pijuice  Zero does not have an ID-EEPROM. Hence you have to load the RTC driver via /boot/config.txt.

You need the RTC driver loaded ('UU' in i2cdetect output) if you want timesyncd to also synchronise the RTC with internet time. The RTC in Linux is always kept in UTC. Hence you need to set alarm times in UTC.
After the RTC is synchronised it will remain running as long as the PiJuice Zero is powered. So you do not need to resynchronise every time you start the RPi. The simulated RTC has a lower accuracy than a real battery backed RTC.

shawaj, 2021-01-04:
Hmm not tested with that battery. But the folks at @CoreElectronics are great so I'm sure it's a high quality unit. I'd quite like to get some proper settings sorted for that battery and their other ones though if possible

budlee85, 2021-01-05:
Thanks @tvoverbeek for the explanation, it's good to see that there's active contribution on this great piece of kit. Did I miss what you wrote in the documentation or is it still catching up with the development (I notice there's not as much covering the nuances of the Pijuice Zero).

I'd ideally like to understand why I don't always have the RTC driver loaded but given the Pijuice Zero should be almost permanently powered with a sufficient solar panel and big battery (I bought it originally to handle more frequent wakes and therefore power consumption) the RTC should always be running and I shouldn't run into any issues from it not always loading.

Thanks for the explanations, I learnt a fair bit through this process