PiSupply/PiJuice #619: Issue running USER FUNC

Steve-I, 2021-01-06:
I have installed a PiJuice on a Raspberry Pi 3B+, and I am running Openhabian (openhabian-pi-raspios32-v1.6.2), an image file based on Debian with OpenHAB pre-installed (https://github.com/openhab/openhabian). I then installed the pijuice-base (openhabian does not have a GUI).

Using SSH to access the raspberry pi, I can run pijuice_cli and have access to the configuration software (status option shows it seems to be working). The issue I have is that the scripts I have set up for User Func do not run. So far:
- The User Func is able to activate a shutdown when power is removed (when configured to do so), so it recognises No Power. 
- The script I have written runs from the command line (as user Openhabian).
- However, when No Power is set to User Func1, the script (python3 /etc/openhab/test.py) does not run when power is removed.

I used chmod 777 to set the permissions of the python script (test.py) to rwxrwxrwx. It writes to a log file (/etc/openhab/pijuice_log.log) which has permissions rw-rw-rw-.

My hunch is that it might be a user or permissions issue. Openhabian does not have the default user of 'pi', if that makes a difference.
Any ideas on what to do next?
Thanks
Steve

tvoverbeek, 2021-01-06:
User scripts have to be owned by a non-privileged user and the owner has to belong to the pijuice group.
The pijuice-base package install creates the pijuice group and adds the default user (user id 1000) to the pijuice group.

Check if user Openhabian belongs to the pijuice group and has numeric userid 1000..
Also if I read the install script for Openhabian, the sudo command without password is removed for user Openhabian.

Steve-I, 2021-01-06:
My linux skills are not great, hopefully I have done this correctly! 
Using `getent group pijuice` gives `pijuice:x:1002:openhabian`, so it looks like user openhabian is a member of pijuice. And `id` gives `uid=1000(openhabian) gid=115(openhab) groups=115(openhab),4(adm),5(tty),20(dialout),24(cdrom),27(sudo),29(audio),44(video),46(plugdev),60(games),100(users),105(input),109(netdev),112(bluetooth),997(gpio),998(i2c),999(spi),1000(openhabian),1002(pijuice)`, so openhabian is UID 1000.

I don't know how to test  "_the sudo command without password is removed for user Openhabian_" , so let me know what you need me to do. Thanks

tvoverbeek, 2021-01-06:
@Steve-I this all looks the way it should be.
You could try to run the service manually in a ssh session and see if you get any error messages when the no power event is supposed to trigger your user function:
- Stop the pijuice service: `sudo systemctl stop pijuice`
- Start it manually in a terminal (ssh session): `sudo -u pijuice python3 /usr/bin/pijuice_sys.py`.
- Trigger the no power event (or any other event which starts a user script.
- Check for any error message in the terminal session

Steve-I, 2021-01-07:
I've stopped the service and tried to start it manually in terminal as user pijuice. However, it looks like it is unable to run `pijuice_sys.py`. It doesn't return to the prompt and when I interrupt (`^C`), I get the following message:

`$ sudo -u pijuice python3 /usr/bin/pijuice_sys.py`
`^CTraceback (most recent call last):`
`  File "/usr/bin/pijuice_sys.py", line 347, in <module>`
`    main()`
`  File "/usr/bin/pijuice_sys.py", line 343, in main`
`    time.sleep(1)`
`KeyboardInterrupt`

And checking the service (`systemctl status pijuice`) confirms it hasn't started:

`● pijuice.service - PiJuice status service`
`   Loaded: loaded (/lib/systemd/system/pijuice.service; enabled; vendor preset: enabled)`
`   Active: inactive (dead) since Thu 2021-01-07 09:25:22 GMT; 3min 12s ago`
`  Process: 503 ExecStart=/usr/bin/pijuice_sys.py (code=killed, signal=TERM)`
`  Process: 6299 ExecStopPost=/usr/bin/pijuice_sys.py stop (code=exited, status=0/SUCCESS)`
` Main PID: 503 (code=killed, signal=TERM)`

tvoverbeek, 2021-01-07:
This is all as expected.
When you start manually (`sudo -u pijuice python3 /usr/bin/pijuice_sys.py`) there will be no output to the terminal
until an error occurs. From the output after you interrupted with Control-C you can see it was running.
Did you try to trigger the no power event (assuming your user function was connected to the no power event)?

The user function needs to be executable. What is the output from `ls -l <path-to>myuserscript.py`?
Also the first line of the script (assuming it is a python script) should be `#!/usr/bin/python3`.

Steve-I, 2021-01-07:
Sorry, I made a mistake on my last post (now deleted) - still figuring out linux! I'll try again:
I ran `pijuice_sys.py`, and `pijuice_cli` shows that the pijuice power is "charging" / "no power" as applicable (under status).
However, in the shell where I run `sudo -u pijuice python3 /usr/bin/pijuice_sys.py`, there are no messages when power is removed.

The script does start with `#!/usr/bin/python3`.
I think it is executable, `ls -l /etc/openhab/test.py` gives:
`-rwxrwxrwx 1 openhabian openhab 401 Jan  5 22:31 /etc/openhab/test.py`
Should user pijuice be the owner of this file?

tvoverbeek, 2021-01-07:
Permissions and executable script are fine.
Does /etc/openhab/test.py produce any output?
If not add a few print statements for debugging.
First run the script directly: `/etc/openhab/test.py` and see if you get the expected output.
Then run pijuice_sys.py manually again (with the pijuice service stopped) and see if the 'no power' event also produces this output.

I suppose you have specified e.g. USER_FUNCn for the 'no power' event on the System Events tab and enabled it.
On the System Task tab the system task should be enabled.
On the User Scripts tab USER_FUNCn should  be specified as `/etc/openhab/test.py`

Steve-I, 2021-01-07:
I think I've solved it - the power needs to be off for ~ 9-10 sec before the No Power System Event is triggered. I was only removing for about 5 seconds - presumably there is some debounce in there?

Out of interest, is it possible to pass an argument from the User Function to the script. For example, can the User Function be set to (for example) `/etc/openhab/test.py arg1`? I've tried it and it is not behaving as expected (but this could be due to my limited coding!).

Thanks for all your help!

tvoverbeek, 2021-01-08:
You cannot pass an argument to a user function (as you noticed).
However the user function is called with 1 or 2 arguments.
First argument is the event which caused the call (in your case 'no power').
The second parameter is more info about the event (e.g. which button was pressed, low charge level, etc. depending on the event).
As an example look at https://github.com/PiSupply/PiJuice/blob/master/Software/Test/user_func1.py which just logs the arguments.

Steve-I, 2021-01-08:
That's great, thanks for the info.

I still have one issue, which I think is with permissions. I have a script which is very similar to `shutdown.py` found [here at Pi-Supply](https://learn.pi-supply.com/make/how-to-run-a-user-script-with-the-pijuice-power-events/). Based on print statements for  debugging, I think the issue is this line: `os.system("sudo halt")`. It runs every part of the script apart from this last line.

When I run it from the command line (as user openhabian, there is no user pi), it asks for a password. If I comment out the `os.system("sudo halt")` line, it doesn't. Is it possible that when the script is run as a User Func, it doesn't have the required permission? The permissions for the python script are:
`-rwxrwxrwx 1 openhabian openhabian  1550 Jan  8 20:15 pijuice_log.py`

tvoverbeek, 2021-01-09:
Yes. this is the thing I mentioned before:
`Also if I read the install script for Openhabian, the sudo command without password is removed for user Openhabian.`
If you want to execute sudo commands without a password as user openhabian:
- as root create a file named e.g. 010_openhabian-nopasswd. Should be read-only for root, no access for others
- its contents the line `openhabian ALL=(ALL) NOPASSWD: ALL`.

Then you should be able to execute a `sudo halt` from a python script.

Steve-I, 2021-01-11:
Thanks for the response. A few more questions as I don't fully understand this!

The shut down script doesn't work from PiJuice. When the User Func runs the script, it runs every part of the script except the final Shut Down command (`os.system("sudo halt")`) - I've added a few various logging steps to test this.

I also ran the script as user Openhanbain to test the script and discovered that it asked for the password at this final shutdown step (not surprising). This confirmed that sudo access is required for this command.

Ultimately, I don't need the script to run as user Openhabian (I was just doing this to test the script), I want the PiJuice HAT to be able to execute the script. There is already the file `020_pijuice-nopasswd`, so presumably user pijuice should be able to sutdown the pi? 

Or, does the user Openhabian **_also_** need to have the ability to run sudo without password for the PiJuice to shutdown the pi?

tvoverbeek, 2021-01-11:
USER_FUNC user scripts are run as the owner of the script.
If openhabian is the owner, any sudo command needs a password unless you add the file as in https://github.com/PiSupply/PiJuice/issues/619#issuecomment-757275174 above
If you change the owner to pijuice it should be able to run specific sudo commands without password as specified in 020_pijuice-nopasswd (incl. /sbin/halt).

Steve-I, 2021-01-13:
`If you change the owner to pijuice it should be able to run specific sudo commands without password as specified in 020_pijuice-nopasswd (incl. /sbin/halt).`

That has solved it - thanks! 

BTW, is there a typo on the Software page (https://github.com/PiSupply/PiJuice/tree/master/Software) - should `SetWakeupOnCharge` be `SetWakeUpOnCharge` (with a capital 'U')?

tvoverbeek, 2021-01-13:
Thanks for catching the typo. It indeed has to be `SetWakeUpOnCharge`.
Will be corrected.
We are not very consistent since `SetWakeupEnabled` is with small 'u' (although it is a different wakeup: by the RTC).