PiSupply/PiJuice #620: PiJuice & NiMh

p-e-s, 2021-01-08:
I would like to use a very large battery pack with my PiJuice (up to 100Wh) and without any of the Lithium batteries transportation constraints.

Is it possible to power the PiJuice with a NiMh pack and what kind of NiMH battery pack configuration can i use?
What is the minimum voltage i should use ? Would a 3x1.2V pack be ok?
Is the PiJuice able to manage charging of NiMh battery packs ?

Thanks

tvoverbeek, 2021-01-08:
I would not even start thinking about it.
NiMh and LiPo have completely different characteristics.
Google on NiMh versus LiPo.
The PiJuice is optimized for LiPo characteristics.

shawaj, 2021-01-09:
As @tvoverbeek says PiJuice is for charging and discharging single cell lipos.

If you want to use NiMH for a particular reason, I guess you could use them with a 5v regulator to output to the PiJuice micro USB input. However you would still need a lipo attached and this wouldn't be able to charge the batteries.

closed 2021-01-09 by shawaj