PiSupply/PiJuice #621: GPIO power input showing BAD and battery not charging

martenjacobs, 2021-01-09:
I'm using a balenaFin 1.0.0 as my main home automation controller and would like to protect it against power outages using a PiJuice. I'm using a PiJuice Zero with a PJLIPO_5000 battery, and my balenaFin is powered from a 24V/0.5A DIN-rail power supply. I expected this to work quite well because I thought there would be more than enough power to keep the battery topped up for the occasional power drop. However, it seems the PiJuice has decided otherwise. It keeps reporting my GPIO power supply as being "bad" and it's not charging the battery:
[image]

What can I do? I don't mind the PiJuice charging slowly, because power outages are rare and usually short, but it should nonetheless keep the battery at 100%, which isn't happening at the moment.

shawaj, 2021-01-09:
The reason it is reporting bad is because it's under 5volts.

I can't remember the exact threshold for it reporting "bad" but maybe @tvoverbeek or @mmilann will remember.

Also @mmilann I wonder if we could update it in the new firmware to have it so that in some "bad" cases it will still charge if it can - such as maybe from 4.5v or something. Maybe we could have "bad" and "bad not charging" or something like this.

martenjacobs, 2021-01-09:
I agree that the voltage is quite low. I'd actually expect it to be higher on the Fin, and I'm not sure why it's not. I tried disconnecting all USB devices but the voltage remained the same, so I believe the DC/DC converter on the Fin may just be calibrated to the low side or it could be damaged. However, since the voltage is still significantly higher than the battery voltage I think it should be able to deliver enough current to at least charge the battery slowly. It would be great if there were more advanced control over this in the firmware.

shawaj, 2021-01-09:
@martenjacobs yeah I was thinking that myself. There's not any reason I can see why it shouldn't charge when it has ~4.9v... Maybe give a warning that the voltage is low but it can still charge. 

I'm not sure having it configurable in the software/firmware is a good idea to be honest. Unless it was just a tick box to "allow charging when voltage low" or something. 

@mmilann might have a different opinion I'm not sure.

martenjacobs, 2021-01-11:
> Unless it was just a tick box to "allow charging when voltage low" or something.

That's pretty much what I was thinking of, although I don't really like the fact that it classifies my power supply as being "bad" if that's not necessarily the case. The problem, as I understand it, is that with different hardware the voltages could be slightly different, but not necessarily bad or wrong. 
As I said, I'm using a balenaFin, so the power supply is quite different from a standard RPi. The Fin uses a LT8650SEV which should be able to provide at least 4A on both 5V and 3.3V (both rails on the Fin are rated at 2.5A though). You can see the [schematics for the balenaFin 1.0.0 here](https://github.com/balena-io/balena-fin/blob/master/documentation/PDF/balenaFin/v1.1/schematics/20173009_balena-fin_v10.pdf). I think mine is calibrated to the low side, but if it can provide 2.5A @ 4.9V without a significant voltage drop then that should be more than enough to power peripherals and have enough left over to charge the battery. The compute module doesn't use the 5V rail at all because it's all 3.3V.
I think the best solution would be to let the user select the exact hardware they are using, from which the PiJuice would then know what voltages to expect and what currents to pull. However, I can see that this could be rather an edge case and it may not be worth the effort. So a checkbox or (even better) an advanced menu where you can control the voltages and currents for charging and classification (within safe limits) would be a nice solution I think. 
I can understand your apprehension of giving end users too much control over these variables (I guess it's because they could easily damage their Pi/PiJuice/battery), but for edge cases such as mine it's frustrating to have a device that should technically be able to work fine, being rendered pretty much unusable by a couple of variables in its firmware.

mmilann, 2021-01-11:
It is more hardware settings. Rpi nominal voltage range is 4.75V - 5.25V. PiJuice detect threshold is fixed to 4.79V slightly having somewhat noise margin. It is bad if it cannot supply continuously at least 100mA for charging and maintain voltage above 4.79V. you may see 4.9V un-loaded, but if you draw any current in addition to Rpi  current it drops below. Try connecting to PiJuice USB IN it has lower threshold of 4.2V.

martenjacobs, 2021-01-17:
@mmilann I'm not sure that that's what's happening here. I've checked the voltage of the 5V rail with and without USB devices present, and it doesn't seem to impact the voltage. I just connected the PiJuice USB in to one of the Fins USB ports, and it's now charging.

Before plugging in the USB port:
[image]

After plugging it in:
[image]

Note that in this case the voltage on the 5V rail is still around 4.8V. Connecting the PiJuice to the USB port pulls the voltage down by only about 0.03V. It's also weird that after plugging in it doesn't say that my GPIO power is bad anymore. 

The problem with this is that when the power goes out, the USB voltage will be supplied from the PiJuice, so it will not properly be able to detect that the power is off. I'm not sure what would happen then...

What's the reasoning behind having different characteristics for the GPIO and USB inputs? And would it be reasonable to make them user selectable?

mmilann, 2021-01-18:
@martenjacobs is it charging when you connect your power supply to only PiJuice (over gpio 5V) no balena attached?

martenjacobs, 2021-01-18:
> @martenjacobs is it charging when you connect your power supply to only PiJuice (over gpio 5V) no balena attached?

I think you could be misunderstanding my set-up. The [balenaFin](https://www.balena.io/fin/) is an industrial baseboard for the RPi Compute Module; the Fin is no HAT, but it does have a HAT-compatible header, just like a Pi.

mmilann, 2021-01-18:
@martenjacobs ok, I suspect that power supply is not capable to provide stable voltage over 4.8V when loaded with charging current (minimum 100ma). Anyway you could do tests by adding some test load like 47R resistor between GPIO 5V and GPIO GND and use voltmeter to check voltage drop.

martenjacobs, 2021-01-18:
@mmilann I tried with a 39R resistor (the closest I could find). It doesn't really seem to impact the voltage measured by the PiJuice:
Before connecting the resistor:
[image]

After connecting it:
[image]

Double-checking it with my volt meter yields a similar result:
[image]

mmilann, 2021-01-18:
Voltmeter voltage definitely shows that it is very close to the threshold
where pijuice cannot reliably distinguish. I can suspect on ps noise that
can be order of ~50mA, one more task if it is available to check it with an
oscilloscope and see it on time scale 10 - 1000us.

martenjacobs, 2021-01-18:
@mmilann Sure, I'll try to check it with my scope tonight. 

Assuming it's dipping just below the threshold, I'd like to restate my question: 
> What's the reasoning behind having different characteristics for the GPIO and USB inputs? And would it be reasonable (and possible) to make them user selectable?

mmilann, 2021-01-18:
@martenjacobs For GPIO power input it needs to be 5V +-5% and it is intended for UPS applications where power supply is supposed to be good. USB input is more for non-reliable power sources like solar panels and minimum can be set in range 4.2V-4.76V.

Going to setup this case with adjustable power supply in order to get better insight..

mmilann, 2021-01-18:
I can now repeat this situation, when setting gpio below 4.9V. It like that charger ic bq24160 has some margins around 100mV when qualifying input as bad. Potentially this can be solved with firmware update just setting DPM below 4.76V.

mmilann, 2021-01-18:
@martenjacobs If you are able to load firmware and test it, attached is quick update with lower DPM 4.68 for GPIO input.
[PiJuice_V1.5_2021_01_18.bin.zip](https://github.com/PiSupply/PiJuice/files/5830446/PiJuice_V1.5_2021_01_18.bin.zip)

martenjacobs, 2021-01-18:
@mmilann I just updated to that version and it's working now!
[image]

mmilann, 2021-01-18:
@martenjacobs Great to hear about progress and Thanks for testing!