PiSupply/PiJuice #622: Device tree overlays not loading at boot.

cbashhmore, 2021-01-09:
Hello, my pijuice is working well generally but I have a question about DToverlays for rtc loading at boot time.If I include the  line dtoverlay=i2c-rtc,ds1339=1 in /boot/config.txt then eveything works as expected, if not then i2cdetect -y 1 shows 68 instead of the expected UU. The id-eeprom address is set to 50. I have inspected the id-eeprom contents using eepflash tools and it appears to be truncated. Is there a copy off this file available to reflash it please?

shawaj, 2021-01-09:
Just so you know you don't need to put =1 at the end of ds1339.

The dtbo I'll dig out and upload. At the moment it's not on this repo.

However, for the RTC this isn't enabled by default anyway and should be enabled as you have done in the config.txt

tvoverbeek, 2021-01-09:
@shawaj Correction, the RTC is enabled in the eeprom overlay, at least in the PiJuices I have.

shawaj, 2021-01-13:
@tvoverbeek you are correct, I don't know why I forgot that.

fdtdump here:
[pijuice.dts.txt](https://github.com/PiSupply/PiJuice/files/5811107/pijuice.dts.txt)

Will upload the settings.txt and .dts file on this repo for reference. 

You should be able to reprogram it with the raspberry pi eepmake tools - however I guess you will need to short the write protect pin to ground. Let me check that

shawaj, 2021-01-13:
@cbashmore what version of Pi are you using? I wonder if this relates to #611

shawaj, 2021-01-13:
@cbashmore the files are now here - https://github.com/PiSupply/PiJuice/tree/master/Firmware/EEPROM

I will provide some instructions to reflash this soon. Note that you should make sure to edit the settings.txt file with your actual device UUID which you can find by running the command `cat /proc/device-tree/hat/uuid`

cbashmore, 2021-01-13:
Many thanks, I will try with this. I'm now wondering whether this might be related to #611 as I am running a 4B.

shawaj, 2021-01-13:
@cbashmore 

From the EEPROM datasheet you will se that the EEPROM has a write protect pin:
[image]

Basically, you need to remove the write protect on the EEPROM, so you can then flash it with the raspberry pi tools. For that you can go to the [HAT config menu](https://github.com/PiSupply/PiJuice/blob/master/Software/README.md#pijuice-hat-general-config-menu), and tick the "ID EEPROM Write unprotect" box.

Then you can use the [raspberry pi eeprom utils](https://github.com/raspberrypi/hats/tree/master/eepromutils) to program the eeprom. You can also find a guide [here from the raspberry pi forum](https://www.raspberrypi.org/forums/viewtopic.php?t=108134).

You can use these commands:
```
# Compile the DT
sudo dtc -@ -I dts -O dtb -o pijuice.dtb pijuice.dts ; sudo chown pi:pi pijuice.dtb

# Make the .eep file
./eepmake settings.txt pijuice.eep pijuice.dtb

# Blank the EEPROM and then flash it
sudo ./eepflash -w -f=blank.eep -t=24c32
sudo ./eepflash -w -f=pijuice.eep -t=24c32
```

cbashmore, 2021-01-13:
So, I have flashed eeprom with the new code and commented out dtoverlay=12c-rtc,ds1339 in /boot/config.txt. After reboot, i2c 1 address 68 now shows UU, but I still can't access the rtc from the command line.

cbashmore, 2021-01-13:
I have flashed new eeprom code. All now works without dtoverlay=i2c-rtc,ds1339 in /boot/config.txt. Many thanks for all of your help.

closed 2021-01-13 by cbashmore

shawaj, 2021-01-14:
@cbashmore great, glad to know it is working.

Have created a readme here for the eeprom programming https://github.com/PiSupply/PiJuice/tree/master/Firmware/EEPROM/README.md