PiSupply/PiJuice #624: Configuration with config file vs python lib

p-doyle, 2021-01-17:
I was hoping to be able to configure the shutdown/wakeup settings via the pijuice_config.json file once and not have to mess with it again but it seems that is not the case, at least for the wakeup_on_charge setting, #176.  Are all of the System Task settings the same?  

For example, can I set the ext_halt_power_off setting in the config file, or do I have to do `pijuice.power.SetPowerOff()` in Python?  It is a little confusing to have a config file that gets ignored for some things but not others...

shawaj, 2021-01-19:
@p-doyle what exactly are you trying to achieve?

We have this also https://github.com/PiSupply/PiJuice/tree/master/Software/Source/Utilities

closed 2021-01-19 by p-doyle

shawaj, 2021-01-19:
@p-doyle did you manage to solve this?

p-doyle, 2021-01-19:
Yes, I just gave up using the config file and do all the config on my end with python.