PiSupply/PiJuice #625: PiJuice battery / charging issues

twixster, 2021-01-17:
I bought a PiJuice a while ago to power a RPI 4 custom portable solution.  However, because the standard BP7X battery doesn't really keep everything going that long, I bought the 12000 mah battery.  But, I seem to be having nothing but issues with the bigger battery since connecting it to the PiJuice.  Also, near constant "COMMUNICATION_ERROR" messages in the software now.  

Has anyone else has issues with the 12000 mah battery and the PiJuice?  

I also just noticed that it thinks the battery is several hundreds of degrees celsius.  If I disconnect the yellow lead from the NTC port on the PiJuice the temp drops down to 48 degrees celsius and it appears to start charging.  However, this is temporary and it eventually shoots back up to 248 degrees (it's cold to the touch...)

I've got the PI4 connected to an 18w PD charger which I feel should be plenty to run the PI4, 7" LCD Touch Screen, and the PiJuice.  However, I have also now connected a beefier MicroUSB charger directly to the PiJuice.  

Ideas?  I'm trying to build a portable laptop like solution, but this is definitely putting a stop to those plans.

twixster, 2021-01-17:
Also want to mention, if I remove the new large battery and set it up for the BP7X battery, everything goes back to normal.  

Is there something wrong with the new battery perhaps?  I'm fairly sure i can't return it as the installation calls for you to clip the connector off to hardwire the battery to the terminals on the PiJuice.  

Please help!

tvoverbeek, 2021-01-17:
It is known that the current barch of 12000mAH batteries has problems with the temperature measurement. It seems the NTC parameters are different from the ones in the 12000mAH battery profile.
For now set the battery measurement to 'not used'. This forces a temperature of 25 deg C and then the battery will charge.
When the measured temperature is too high the firmware stops charging the battery.
You must have a recent Pi4. The first version had an error in the PD recognition, causing the PD power supply to believe it is a headphone and thus insufficient power was delivered.
Connecting a PD charger to the Pi and an USB charger to the PiJuice at the same time is not recommended

shawaj, 2021-01-17:
Related to #608 and #617 and #606

twixster, 2021-01-17:
I have a RPI 4 model B Rev 1.2, this version was "supposed" to not have the USB C PD issue from what I've read online.  The first version did have issues with PD chargers.  Is there a way to tell if it is getting enough power?  I don't get a lightning bolt in the Rasbian GUI like I would expect with chargers that were not up to the task.

When powered over USB-C with the RPI 4, Display, and PiJuice all connected.  
    The GPIO Power Input in PiJuice shows ~4.8*V, ~-.7A, Weak

When powered over USB-C with the RPI 4 and PiJuice connected and the Display connected to External Power 
    The GPIO Power Input in PiJuice shows ~4.9*V, ~1.3A, Present

Can you track this?  Is this data stored somwhere that it can be extracted for use?  

Good to know it's not recommended to run both USB C and MicroUSB for power.  I wish there was a second version of the PiJuice that supported USBC PD Charging, so we didn't have to rely on the Raspberry PI's somewhat limited implementation of USB-C for power.  I've looked at the PiSugar2 Pro which "may" have better power support, but I'm not interested in switching at this point.  

Also, I've got what I believe is a fairly inefficient 7inch LCD (1024x600) touchscreen running from GPIO and connected to USB port on the PI for touch input.  I still need to power a Teensy for a keyboard controller.

For now, it looks like the large battery is charging.  I am still getting many "COMMUNICATION_ERROR" messages in the app.  Maybe because I'm watching it now and I wasn't before?  lol

Thanks guys, I hope there will be a fix soon.  I've waited almost a year for this battery, somewhat anti-climatic to have issues right out of the gate.  I would feel better if the temp sensor was registering correctly as a safety precaution.

twixster, 2021-01-17:
@Shawag, thanks.  I thought i had looked thoroughly enough, apparently not...  Although i was more looking for "Communication_Error" when i was searching at midnight last night.

shawaj, 2021-01-18:
No problem at all @twixster - I just wanted to link up the issues so we know to check if we can close other ones when one is solved.

Re USB-C we are working on this already see #580

shawaj, 2021-01-18:
Also @twixster what version of the firmware and software are you running on the Pi and PiJuice? 

Are both fully updated?

Zeromusta, 2021-02-21:
I won't raise a new issue but will let you know I'm in the same boat. New 12000mah battery, SUSPEND charging, reporting 245 degrees but cool to touch. Using the disable NTC workaround.

[image]

shawaj, 2021-03-15:
closing in favour of #659

closed 2021-03-15 by shawaj