PiSupply/PiJuice #629: Repeated Wake-up enable using Pure Data with shell-object

deankneefel, 2021-01-19:
I'm currently working on a small installation and I ran into a few problems with enabling "repeated wakeup".

So the idea of the installation is a small set-up that records audio for 5 minutes every hour. After that, it will run some log tasks and will shut down. After 55 min of shut down, the Pi will boot up and starts recording again for 5 min, etc. 

The recording will eventually be done with a Cirrus Logic Audio Card HAT, a small microphone, and a small Pure Data patch. The Pd-patch is ready to go. I've installed a few externals within the patch that allows me to control the terminal, whit the shell object from the 'ggee external', and it will also log the current battery data, time, and date when the patch was running. After 5 minutes the shell object will receive a 'sudo shutdown' message to shut the Pi down.

The Pd-patch includes  "python /home/pi/pijuice_util.py --enable-wakeup" since it is mandatory to do so after every boot.

Every time I open the patch, while the Pi is running, it all seems to work. Wakeup enables, logs, everything. Also when I run this with a very simple shell script and via the terminal straight away with "sudo /etc/rc.local/" it also works. 

But ONLY the logs work after rebooting the Pi. Even though the Pd-patch is running after boot without GUI. 

I've also tried the repeated wakeup_enable.py tutorial in the Software read me. This also works running straight from the terminal (like described above). But doesn't when rebooting.

[image]
[image]
[image]

Any idea?

tvoverbeek, 2021-01-19:
I do not know PD, but /etc/rc.local runs very early in the boot sequence before the whole system is up (e.g. I2C might not be initialized yet).
Putting sleep() statements in /etc/rc.local is a bad idea since it holds up the remainder of the boot.
Put the scripts you call from /etc/rc.local in the background and add the sleep(10) inside the script-pd.sh

deankneefel, 2021-01-19:
I have delayed the execution of the shell script and now it works!! I think you were right that the I2C communication wasn't up yet. 

Best regards!

deankneefel, 2021-01-20:
We are testing repeated wake up every 5 minutes. This works for a number of wakeups but then stops working. 

The PiJuice is connected to an external power source. The Pi connects to WLAN. What could be causing the stopping?

tvoverbeek, 2021-01-20:
If you use the default settings (EEPROM address at 0x50) the OS will load the DS1339 driver.
When the Pi has an internet connection the timesyncd daemon will synchronise internet time with the system and the RTC.
**Timesyncd will also clear the RTC wakeup flag**
So if the first tiem sync happens during your 5 minutes run, the following runs will not work.

Workaround 1:
Change EEPROM address to 0x52 and in /boot/config.txt add `dtoverlay=i2c-rtc,ds1307`.
DS1307 has the same interface as DS1339 but it does not have an alarm function, hence the OS will not clear the RTC wakeup flag. But you still get the benefit of time synchronization with the internet.

Workaround 2:
At the end of your 5 minute job before you shut down set the RTC wakeup flag:
`pj.rtcAlarm.SetWakeupEnabled(True)`

deankneefel, 2021-01-22:
Thanks a lot, it solved our problem!

closed 2021-01-22 by deankneefel