PiSupply/PiJuice #630: Battery slowly discharge while the power is on

thekyashan, 2021-01-19:
Hello guys,
I have installed the latest Raspbian Lite (v10 Buster) on a Raspberry Pi 2B and installed all the packages for the PiJuice board. The board is obviously installed and I powered everything with an official Raspberry Pi power supply 2.5A. The power supply is connected to the PI.

The PiJuice battery is the BP7X 1820, the software profile loaded is correct and also the dipswitches are set as per documentation (I must say that when I removed the little yellow tape, the dip switches were both on the OFF position, which is wrong for the BP7X).

The first thing I noticed was that the battery didn't charge fully but just reached the 98%. I thought it might required more time to get to 100%, so I let it be. After one week (now) the battery didn't reach the 100% but it's draining slowly slowly: we are at 93%.

Below a screenshot of the PiJuice_cli

░░░░░░░░│  HAT status                                               
░░░░░░░░│                                                             
░░░░░░░░│  Battery: 93%, 4.103V,  24°C, NORMAL 
░░░░░░░░│  GPIO power input: 5.071V, 0.540A, PRESENT            
░░░░░░░░│  USB Micro power input: NOT_PRESENT                       
░░░░░░░░│  Fault: None                                                 
░░░░░░░░│  System switch: Off 

What do you think is the issue here?
A faulty battery? Is there anything I can do?

Thanks in advance
Regards

tvoverbeek, 2021-01-19:
Battery status NORMAL means it is not charging.
Is charging enabled on the General tab?

thekyashan, 2021-01-19:
I have shut down the PI and restarted again and now the status says CHARGING FROM 5V IO.
The battery was charging back again, led was alternating green and blue, but at 98% became again solid blue and the status changed back to NORMAL.
So, I have been to the GENERAL tab to see if charging was enabled and it was. I have done a quick disable/enable and now the status says CHARGING FROM 5V IO.
Last-minute edit: after 2 minutes the led is again solid blue and the battery is not charging.

Is this behaviour normal? The PiJuice is supposed to stop charging the battery at 98%? or it has to do charge it continuously?
I haven't configured anything like that, no user scripts are running; I have just installed the PiJuice software for Raspbian lite.
Do you have any recommendations?

░│  HAT status                                           
░│                                                            
░│  Battery: 98%, 4.172V,  24°C, NORMAL                     
░│  GPIO power input: 5.076V, 0.280A, PRESENT

tvoverbeek, 2021-01-20:
Yes, this is normal.
The charge level (State Of Charge) is estimated.
The solid blue led means 'Charging is done' and occurs when the charging current drops below the termination current specified in the battery profile. It does not consider the estimated charge level.

thekyashan, 2021-01-21:
Thanks tvoverbeek.
So at some stage, the battery charge should drop but the PiJuice should kick in and charge it to bring it back to the 98% or so.
Is my assumption correct?

tvoverbeek, 2021-01-21:
yes

thekyashan, 2021-01-21:
Thanks tvoverbeek.
I’ll keep monitoring the PiJuice, just to understand when it decides it’s time to charge the battery.
But to be honest with you, the PiJuice logic/behaviour in this case doesn’t seem right to me.

I believe that while the power supply is still available, the PiJuice should keep the battery fully charged (or at least allows a small fluctuation maybe between 97 to 100%) in case the power is not available (for whatever reason i.e. blackout, solar power not available, etc). In this way the Raspberry Pi will have the max juice to keep itself alive for more minutes/hours.

Don’t know if you agree with me, but for instance my APC UPS keeps its battery fully charged so that in case of power outage, the battery will be at its top to keep my computer alive and running for the max time possible.
If there’s a power outage and the PiJuice has left the battery to discharge until the 60%, the Raspberry will last significantly less rather than if it had the full battery available.

What do you think? Am I expecting too much for a board that claims to be an uninterruptible power supply for Raspberry Pi.

shawaj, 2021-01-22:
@thekyashan I think you have misunderstood. It does keep the battery fully charged, that is what Ton is saying.

thekyashan, 2021-01-22:
Apologies, so this means that my PiJuice is not behaving as expected.
I've checked right not and the battery is discharging: it's at 97%

░░░░░░░░│  Battery: 97%, 4.141V,  24°C, NORMAL             
░░░░░░░░│  GPIO power input: 5.073V, -0.376A, PRESENT       
░░░░░░░░│  USB Micro power input: NOT_PRESENT                         
░░░░░░░░│  Fault: None                                                
░░░░░░░░│  System switch: Off

shawaj, 2021-01-22:
As Ton said the percentage is an estimation. It is not accurate for various reasons.

If it is not charging and says normal that means it is full.

thekyashan, 2021-01-22:
Thanks, Shawaj, so the battery is fully charged that's why the PiJuice is not charging it. And for the same reason, the led is solid blue.

I have another question: last week, I simulated a power outage to see how the PiJuice reacts to that. The RPi was still on and when I switched the power back on (after few secs) the PiJuice led was alternating green and blue, so it was charging the battery.

Why the PiJuice thought it was necessary to charge the battery only after I switched off the power supply if a few seconds before it thought the battery was fully charged?

shawaj, 2021-01-22:
Because that's how batteries work. When you unplug it, it will drain a bit. Then the power comes back and it needs charging again.

Also we have two protection circuits in the design for extra safety which makes the battery percentage detection not so accurate.

thekyashan, 2021-01-24:
All right, so everything normal.
The percentage is not accurate. What about the battery’s voltage? Not accurate as well?
Is there a value I can take in consideration and, more important, read using Python?

I bought the PiJuice to build a power outage monitor. I’d like to read the battery status to send me regular alert when the power is off, just to understand when the Rpi is about to shutdown. But at this stage I don’t know which parameter/value to rely on.
What would you recommend?

tvoverbeek, 2021-01-25:
If you want to detect external power loss look at the power input status for USB IN and/or 5VIO.
For an example see http://frederickvandenbosch.be/?p=2876

thekyashan, 2021-01-26:
Thanks Ton, I'll have a look.

thekyashan, 2021-02-04:
Hello all,
during the past days I kept monitoring the voltage of the battery instead of the percentage; it was dropping slowly, it was very close to 4 Volts instead of the 4.141 Volts of the full charge; the day after all of sudden the voltage and percentage came back to the original values.
I guess the PiJuice charged the battery when the voltage dropped below the 4 volts.

alpapak, 2021-03-01:
> Yes, this is normal.
> The charge level (State Of Charge) is estimated.
> The solid blue led means 'Charging is done' and occurs when the charging current drops below the termination current specified in the battery profile. It does not consider the estimated charge level.

Hello, 

in Battery Profile, my charge current is 925 mA and termination current is 50 mA. If i understand correct and my battery is BP7X-1820, charging will start when it will be 50%. Can you please explain with numbers this set up settings.