PiSupply/PiJuice #631: battery discharging too fast

sevdh, 2021-01-23:
I use a combination of the Pi 3B+ and the PiJuice. The Pi is powering up every hour and shuts down 2 minutes later. In these two minutes the battery voltage and charge is monitored and logged.
With a fully charged battery (the standard one coming with the PiJuice) the battery charge goes down about 15% every hour (although the Pi only runs for two minutes). I would expect the charge to go down much much slower.

What could be causing this fast discharge?

Just in case: there are no peripherals connected to the Pi during the tests.

Best regards! Edwin

tvoverbeek, 2021-01-23:
How do you shut down the Pi at the end of your 2 minutes run?
If you do not explicitly remove the 5V power from the PiJuice to the Pi, the P 3B+ remains powered (red power led on Pi stays on).
In that case the battery keeps discharging during the whole hour.

sevdh, 2021-01-23:
You are right, I’m using a sudo shutdown in the shell and the red led stays on.

Am I right to assume I should use the SYS_FUNC_HALT_POW_OFF script instead?

tvoverbeek, 2021-01-23:
You cannot call SYS_FUNC_HALT_POW_OFF yourself, it is an event sent by the PiJuice and handled by the pijuice service (system task).
Instead on the System Task tab in the GUI or CLI enable 'Software Halt Power Off' and set a long enough delay to give the Pi enough time to complete the shutdown before the power disappears.
I usually specify 20 seconds. The service will then command the PiJuice to remove the 5V after e.g. 20 seconds when the shutdown (commanded externally) is already in progress.

Zou moeten werken in dit geval.

sevdh, 2021-01-23:
I’m sorry not to have understood this function.

It works like a charm.

Thanks a lot.

Het werkt perfect!

closed 2021-01-26 by sevdh

JolleJolles, 2021-02-13:
Happy to have found this threat as I had exactly the same issue. This is NOT clear from the documentation so should be clarified ideally. 

I was also wondering if this is possible to do via a USER_FUNC python script rather than using the PiJuice services. For example, I have a script that automatically turns off the Raspberry Pi. Could I then not add the following to do the same as the service as you describe above:

```
time.sleep(30)
pijuice.power.SetSystemPowerSwitch(0)
```

shawaj, 2021-02-13:
@JolleJolles do you have any suggestions for how we could document this better? Open to ideas :-)