PiSupply/PiJuice #632: Early power loss

Rocknrolldoggie, 2021-01-23:
Good day

second day of owning a pijuice hat and wanted to test the battery run time but the pijuice shutdown / totally lost power after 30 min. when rebooting checked battery level which was at 52%. Any ideas on when it went off at while there should have been power ?
appreciate the help..

tvoverbeek, 2021-01-25:
What is your load when running on battery? E.g. using a Pi 4B with an attached 7 inch touchscreen will certainly drain the default BP7X in a few hours.
Also if the power to supply is larger than what the PiJuice can deliver, it will shut down the Pi. It seems that is what happened since your battery wa still at 52%.

Rocknrolldoggie, 2021-01-25:
Hi Ton

Thank you for the reply I really appreciate it. unfortunately I do not have
the skill to answer your question.

All that is running on the pi is an external hdd but that is usb powered by
the pi. I use it as a media server (plex) and have used it for more than a
year so far pre pi juice.

I am running a 3b+ and of course expect running the pi powered hdd will
cause a quicker drain it’s the 52% part confusing me.

Craig.

Rocknrolldoggie, 2021-01-25:
Perhaps I should have mentioned too that the hdd was running from 98% to
the 52% as I wanted to see primarily how long it would last with using the
hdd. Was watching 720p video.

tvoverbeek, 2021-01-25:
Your hdd will use something like 500mA. The Pi when showing video around 1A (guestimates).
This is already quite a bit for the PiJuice with the standard battery.
See https://github.com/PiSupply/PiJuice/tree/master/Hardware#max-current-that-the-pijuice-will-be-able-to-supply

Rocknrolldoggie, 2021-01-25:
Hi.

I had ordered a 12 000 battery which will arrive shortly as knew it would
need more power. Ideally I am looking for 4 hours life without power, this
is due to the occasional rolling power cuts we can have (yay living  in
South Africa). Again I always knew that the unit would a bigger battery
hence the order I was just concerned the brand new unit had an issue or I
needed to change settings because of the  half power “apparently” still in
the battery.