PiSupply/PiJuice #633: @ChristopherRush Where/How did you find this?

DannyQuantum, 2021-01-28:
@ChristopherRush Where/How did you find this?
The no power event is generated in the service (pijuice_sys.py) when both the USB and GPIO inputs have status 'NOT PRESENT' or 'BAD'.
See function _EvalPowerInputs() in pijuice_sys.py.
The 'Input precedence' is not used.
Tested it with Input Precedence both ways and it does not make any difference.
The code does already what you propose.

_Originally posted by @tvoverbeek in https://github.com/PiSupply/PiJuice/issues/206#issuecomment-420262108_

ChristopherRush, 2021-01-28:
@DannyQuantum Anything I can help you with?

DannyQuantum, 2021-01-28:
The flagging of 'No power' is causing issues in my case, I'm using this event to start a user function that:

1. Checks that power is genuinely disconnected (waits 10 secs then checks for 'NOT PRESENT')
2. Shuts down the pi if battery is 10% less than at time of connection

Because I don't treat 'Bad' connection as a case for shutting down (I'm using the PiJuice to protect SD cards on a project that ideally **cannot** have any down time where avoidable. As I understand it it, because they system is running my function when the no power event is created by the GPIO 5v 'BAD' status and not calling it again, my script is effectively running, halting and then not running again when required - though this in itself is not consistent behaviour - as such this is more of a prognosis on my part than a diagnosis.

I'm just wondering if there is anyway that the settings can be changed (without having to modify source code and using my own version if possible) that this can be changed so that the 'BAD' status does not trigger the event?

DannyQuantum, 2021-01-28:
> @DannyQuantum Anything I can help you with?

Sorry, was typing - need to update the title too.

shawaj, 2021-01-28:
@DannyQuantum oops didn't see your updated title.

DannyQuantum, 2021-01-28:
@shawaj no worries!

DannyQuantum, 2021-01-28:
With further testing the nature of the issue seems to have evolved and I'm less suspecting of the 'BAD' power being the issue with my user function.

I have set the `No power` event to `sys_func_halt_pow_off` and still when disconnecting power the system is not shutting off.
System tasks are enabled and the only other setting I have set up is `WakeUpOnCharge` which is set to 0. I'm able to count 20 seconds before plugging power back in (testing on very low battery now). My latest observation is that actually the GPIO power input actually doesn't stop reading 'BAD'. Voltage is 4.94 and 0.0Amps?! On that.

tvoverbeek, 2021-01-28:
@DannyQuantum Note that SYS_HALT_POW_OFF waits **60 seconds** before removing the 5V. If you reapply power within 60 seconds the 60 second wait is aborted and the power remains on.
See lines 59-62 in pijuice_sys.py (https://github.com/PiSupply/PiJuice/blob/master/Software/Source/src/pijuice_sys.py)

DannyQuantum, 2021-01-28:
@tvoverbeek duly noted. we've not had any issues with getting the pi to power on after shutting down however (the wake on 0% charge is working perfectly), it's more that the system now doesn't appear to be detecting a change in connected when removing the 5v input USB-C from the Pi 4. We've been testing with the Pi Juice for the last few days and this is the first we've seen of this issue.

DannyQuantum, 2021-01-28:
We're currently changing our wiring to see if that solves our issue, previously we were running power into the pi and then feeding from there to the PiJuice over about 1.5 meters of extending wire from GPIO on the Pi to the GPIO pins on the PiJuice. The wiring we're changing to (which is how we have it on the unit that isn't having these same issues) is having the PiJuice and the Pi powered in parallel and joining the two via GPIO, will update if that gets us any closer.

DannyQuantum, 2021-01-28:
Changing wiring didn't solve it, nor did increasing voltage to the loom, swapped for a new PiJuice (both board and battery) and the problem seems to have been solved.

closed 2021-01-28 by DannyQuantum

shawaj, 2021-01-28:
Maybe you need to reload the firmware on the PiJuice. But also this is not a recommended way to use the PiJuice to be honest - the long stretches of cable to connect. I would recommend testing all the functionality you need first on a PiJuice directly connected to this Pi before testing things with this additional cabling