PiSupply/PiJuice #636: Not waking up, while wake up is enabled (odd behaviour)

BoelD, 2021-02-04:
After leaving a installation with a PiJuice, camera and solar panels in the field for two weeks, I ran into odd behaviour. The settings where that the PiJuice wakes up the Pi every 30 minutes. Then, its calculated whether it is light or dark via the RTC. It it's light, a 30 second video is recorded. If it's dark, no recording takes place. In both instances, the Wake Up will be enabled and the Pi is shut down. When testing, everything worked as it should: the Pi woke up every 30 minutes and did what it had to do before shutdown.

In the field however, it worked fine the very first day of installing (Janari 21) but after the night, it only woke up three times during light, only to wake up again in the evening and all through the night. When I picked up the device, I noticed the blue led was   buring steady, although I found out there was no 100% charge (but eighty something) I suspect it has something to do with the solar panels. From the log files I made, I was able to derive that the problem occurs occurs when dark becomes light (and thus the solar panels potentially start giving energy). This seems to make the PiJuice NOT waking up the device anymore. And when it's getting dark again, the PiJuice will wake up again as planned: every 30 minutes steadily. The solar panels I am using are Voltaic 6V 3.5W panels, attached to JP4.

Is there anyone with a usefull clue to a solution?

BoelD, 2021-02-05:
The battery is a 12000mAh PiJuice LiPo. Firmware 1.4 (up to date)

Wake up alarm is set in the CLI: 
(X) weekday [x]
every day
[x] every hour
(x) minutes period
Minute: 2

Wakeup on charge is not active. No wake up scripts involved, no system events.

System task: [x] System Task enabled, [x] Software Halt Power off. Delay period: 20

And like I mentioned: it only happened after the first night in the field and continued the two weeks that followed. I have been  trying to reproduce the erroneous behaviour in the lab for the whole morning, but it's working now as intended. Tried applying various voltages with lower current limits on P4, mimicking the effect of the solar panel not being able to provide the neccesary current. But the PiJuice now keeps waking up as intended.

Also tried darkening the solar panels and allow sunlight on them again. Detected no problems.

Measured the solar output with a scope. I do not detect any voltage spikes or other unexpected behaviour.

mmilann, 2021-02-05:
If you are using 12000 battery from last batch there are issue with incompatible temperature sensor, that may disable charging and solution to set "Temperature sense" to "ON_BOARD". Also this battery has lower current limit and can be overloaded. 

What is estimated load current, Rpi plus peripherals? It may get overloaded if battery has no energy and power source is weak. PiJuice may power-off GPIO 5V if it gets overloaded and there is "Fault flag": forced_power_off when this happens. You may also make test by changing load current with resistors or similar.

Also if you have only one setup you can try new firmware update V1.5, currently in test phase, that solves some wake-up issues, just install new package and update firmware:
https://github.com/PiSupply/PiJuice/blob/v1.7/Software/Install/pijuice-base_1.7_all.deb

BoelD, 2021-02-05:
I already disabled the NTC reading. Charging is not the issue here, as I mentioned in my original post the battery level never drops below 80%. So the battery always has plenty energy.

The issue is waking up. Or better: not waking up.

I see this update in 1.5:

"   5V Regulator turn-on overload fault issue patch. DCDC 5V regulator goes to 
    fault state when turning on with high load current and large capacitance at GPIO 
    rail. turn-on procedure updated with 3 retires in short intervals ~1ms."

I hope this is the problem. Although there is noting else but the PiJuice attached to the GPIO rail, so one would wonder where the high capacitance would come from.

mmilann, 2021-02-05:
I am not sure how do you halt system after recording is finished. RTC wake-up configuration is saved inside mcu RAM so in some situations it might be lost like power reset. If you use any custom software you can use python API function "SetAlarm" to reset wake-up after every boot.

BoelD, 2021-02-05:
When the recording is finished, I enable the alarm with:

`pj = pijuice.PiJuice(1, 0x14)
pj.rtcAlarm.SetWakeupEnabled(True)
`
And then I shut the pi down with:

` os.system("sudo shutdown now")`

If the PiJuice would loose configuration, that does not explain why the PiJuice resumes its normal behaviour after the sun has set. When it's dark, it stars waking up again every half hour. And it stops waking up, when it's light. That's why I think the solar panels (connected to JP4) have something to do with it.

For the sake of completeness again: the PiJuice does charge during the day. But it does not wake up the Pi during the day.

mmilann, 2021-02-05:
Additionally you may re-set alarm config every time:

```
rtcConfig = { 
    'weekday': 'EVERY_DAY',
    'hour': 'EVERY_HOUR',
    'minute_period': 30
}
pj.rtcAlarm.SetAlarm(rtcConfig)
pj.rtcAlarm.SetWakeupEnabled(True)
```

I want to repeat similar conditions for test with my setup. What is Rpi version and do you have additional stuff like camera connected to USB or camera connector?

BoelD, 2021-02-05:
Thanks for diving into this!

Setup: Raspberri Pi 3B, PiCam v2, 2 3.5W Voltaic solar panels attached in parallel to J4 in a Naturebytes enclosure in the open field in The Netherlands, in Januari. Solar panels directed south.

Software configurations as described above. (Firmware 1.4)

Let me stress once more: de PiJuice does NOT seem to "forget" wake up settings. It just does not wake up during daytime, but resumes the intended wake up cycle after dark.

One strange thing that could be noted, is that when I came to pick it up, the Pi had did not wake up at the specified time and that after opening the enclosure, it appeared that the blue led was burning, indicating that the battery would be fully charged. After manually turning the Pi on, it appeared that the charge was not 100%, but somewhere in the eighties.

I just installed the whole seting in my garden, to see if I can reproduce the behaviour myself. Like I said earlier, I wasn't able to reproduce it under lab conditions.

mmilann, 2021-02-05:
Reported battery level may be inaccurate for this battery because of not accurate temperature measurement, but generally you may expect several percents below 100% when charger stops charging, LED is constant blue (charge current drops below threshold). 

Let me know if you catch it.

BoelD, 2021-02-05:
I will post an update as soon as I have it.

mmilann, 2021-02-06:
Hey @BoelD got something that looks like your scenario. 12000 battery and Power input connected PiJuice USB micro,, periodic alarm configured. And if power input is plugged, after first time it misses to trigger wake-up, and continues until power input is reconnected, than resumes with wake-ups. After running debugger what I see is that it misses to fire Alarm interrupts when in run mode (power input plugged), but it fires interrupts regularly when goes to standby (unplugged). This is likely firmware problem where stm32 rtc driver misses to properly trigger interrupts, so made patch to do polling as alternative that solved problem in my case. Attached is update that you can write with command:
`pijuiceboot 14 ./PiJuice-V1.5_2021_02_06.elf.binary`
[PiJuice-V1.5_2021_02_06.elf.binary.zip](https://github.com/PiSupply/PiJuice/files/5937159/PiJuice-V1.5_2021_02_06.elf.binary.zip)

BoelD, 2021-02-06:
This, indeed, looks very much like what i am experiencing. Another observation from today: I had the installation in my garden overnight. When I checked this morning, it didn't wake up. I disconnected the solar panels and then it woke up at the next scheduled time. I reconnected the solar panels and then it didn't wake up.

So I downloaded the firmware V1.5_2020_12_24 earlier today and flashed it to the PiJuice. So far, it does seem to wake up. But let's see what the night brings, my logs indicate that the problems started after the first night, so I will know more tomorrow. I am not sure whether the version you mention (PiJuice-V1.5_2021_02_06) addresses the issue in a different way.

A question that still remains is when the the unexpected behaviour exactly occurs (and why). Yesterday I applied various voltages in the range of 0-10V to JP4 with my lab supply and tried various max currents with it. I wasn't able to replicate the problem then. You mention that it only misses the trigger after the first time. I will do some specific testing with this in mind later.

mmilann, 2021-02-08:
@BoelD use latest PiJuice-V1.5_2021_02_06, it have this fix, earlier not.

BoelD, 2021-02-08:
To make things slightly more confusing: since I installed version V1.5_2020_12_24 the PiJuice didn't miss a single wake up call. The installation is outside, in the freezing cold and snow. It's been cloudy, so the panels probably haven't produced much (if any) energy. But in the two weeks where the installation worked on the older firmware, there were also many clouded days. But during those two weeks, the PiJuice never woke up during daytime.

I can install the newer firmware, but it may be interesting to wait for a sunny day, to see if there is a difference in behaviour.

BoelD, 2021-02-10:
Update: so far, with firmware V1.5_2020_12_24 the PiJuice has been waking up correctly now for for days already. The installation is outside, in the freezing cold. On sunny moments, the batteries do charge from the solar panels.

So @mmilann the issue you are addressing with firmware PiJuice-V1.5_2021_02_06 is not the same as the one that I was facing.

I'd love to know what the problem could be - or has been.

mmilann, 2021-02-10:
@BoelD if you have logging ability, you can log power presence, for example every minute and see if it is continuous or intermittent. For example if there is two consecutive wake-ups with power present from solar panel and without power present interruptions. If there is short period of sun disappearing it is enough that mcu goes to standby where it clears interrupt flag and is able to fire it on next rtc alarm timeout and hence no missing interrupt.

BoelD, 2021-02-11:
After almost a full week of not missing a single wake up alarm and charging in between, the device started to show the exact behaviour it was showing in the field: it didn't wake up this morning after the sun came out and only started to wake up when it was dark. Biggest difference with other days was, that it was quite a sunny day today.

I flashed the file PiJuice-V1.5_2021_02_06.elf.binary.zip now. To my surprise, the NTC reading - which was correct for a while now, suddenly shows 65305 degrees and disabling the temperature sensing with pijuice_cli does not make a difference. Also, the firmware path in the corresponding menu, still points to 1.4, although "Current version"  shows  1.5. Brings be to the the conclusion that this firmware is not stable yet and not to be used in the field. Correct?

Let's see what tomorrow brings. Let's hope for a sunny day, so that we can make some comparisons.

mmilann, 2021-02-11:
@BoelD Did you updated to latest base software package? You need it because some of later firmware updates was for temperature readings.

Link for base package: https://github.com/PiSupply/PiJuice/blob/v1.7/Software/Install/pijuice-base_1.7_all.deb

Also when you update software and latest firmware, you can write custom profile parameters for 12000mAh from this JSON file using pijuice_cli: https://github.com/PiSupply/PiJuice/blob/v1.7/Software/Configs/Batteries/LIPO_12000_PIS-1129.JSON
that should fix temperature readings, also option to disable NTC stays.

BoelD, 2021-02-16:
How can I use pijuice_cli to open import the settings of the JSON file? I may be overlooking it, but I don't see any mention of it in the documentation.

tvoverbeek, 2021-02-16:
You can use the values in the json file to create a custom battery profile for your 12000 mAh batteryvin pijuice_cli.

tvoverbeek, 2021-02-16:
No way to import it directly. Will think about how to do it.

BoelD, 2021-03-01:
The installation has been running, charging and waking up as intended for almost a week now. So I think the conclusion is that the new firmware does what it's supposed to do. Therefore: issue closed.

Thanks everybody, for your help!

closed 2021-03-01 by BoelD