PiSupply/PiJuice #637: Pijuice 12000mAh battery Fault: charging temperature suspend, with firmware 1.4

salimness, 2021-02-05:
Right out of the box I get this fault with temperatures over 200 degrees. Had this issue so far with 3 brand new 12000mAh batteries. The 1820 mAh battery works fine.  
I update first using this command and select the appropriate profile:
apt-get -y install pijuice-base pijuice-gui 

Do I just have a bad batch of batteries? 

[image]
[image]
[image]

BoelD, 2021-02-05:
This is a known issue. Search in the closed issues here, for extra details. For now, a workaround is to disable the NTC. You can choose to measure the onboard temperature, or measure no temperature at all.

BoelD, 2021-02-05:
[https://github.com/PiSupply/PiJuice/issues/617](url)
[https://github.com/PiSupply/PiJuice/issues?q=is%3Aissue+temperature](url)

salimness, 2021-02-05:
Thank you very much for the quick response.

closed 2021-02-05 by salimness

shawaj, 2021-02-05:
@salimness please change the battery NTC to NOT_USED in HAT battery menu

shawaj, 2021-02-05:
Oh thanks @BoelD 👌