PiSupply/PiJuice #638: pijuicetest.py no longer contains examples of setting LED2

jealousofyou, 2021-02-06:
The Hardware readme section on the LEDs suggests that pijuicetest.py shows examples of how LED2 can be configured.

New versions of the script have had the examples stripped out. Either the readme should be corrected or perhaps some documentation or examples of the relevant functions could be provided somewhere.

tvoverbeek, 2021-02-06:
See https://github.com/PiSupply/PiJuice/tree/master/Software#pijuice-status the SetLedBlink function

tvoverbeek, 2021-02-06:
See the description fo the SetLedBlink function in the Status section of the Software README (https://github.com/PiSupply/PiJuice/tree/master/Software#pijuice-status).
Also corrected in the hardware README.

closed 2021-02-06 by tvoverbeek