PiSupply/PiJuice #639: Ubuntu installation

LeeD76, 2021-02-08:
Hi, This may have been ask previously but is there a way to install the BASE drivers and GUI software onto Ubunti 20.10 64bit??  I'm working on a project and all other components work on Ubuntu but when I move the project to Raspberry Pi OS, as it is now, some of the components do not work in particular, SQLite Nuget packages for DotNet Core 5, sorry I'm a C# developer.

tvoverbeek, 2021-02-08:
@LeeD76 Sorry, the packages are not ready for 64-bit yet.
All the python programs should work. The problem is the 32-bit binaries for pijuice_cli and pijuice_gui and pijuiceboot.
The source is in Github, you would have to rebuild them on a 64-bit system.
For pijuice_cli and pijuice_gui you could work-around it by running them with this command:
`sudo -u pijuice python3 /usr/bin/pijuice_cli.py`. Similar for pijucie_gui.
Also look at the postinstall script for pijuice-base to create the pijuice user and set the right permissions on directories.

LeeD76, 2021-02-08:
Thanks for the info.  In an ideal world, I would want to build my project on a Server as I don't really need a desktop, so if I did want to use ubuntu I could build a 32 bit Ubuntu server, would this then allow me to install the PIJUICE_BASE and CLI normally?  Is there any plan to release an official 64bit version?

tvoverbeek, 2021-02-08:
@LeeD76 On a 320bit system it should work. See https://github.com/PiSupply/PiJuice/tree/master/Software/OS%20Support
There is also info for 64-bit systems, which is rebuilding the packages completely on a 64-bit system/

LeeD76, 2021-02-08:
Thanks for your help.  I'll take a look, hopefully, I'll get it working

LeeD76, 2021-02-08:
Sorry, one last thing as am not sure this is the correct place to ask but is the PIJUICE short case compatible with the Raspberry Pi 4?

tvoverbeek, 2021-02-08:
No, due to the changed positions of the ethernet and usb connections.

shawaj, 2021-02-08:
We have a new version of the case coming soon that will support it but the current one doesn't

LeeD76, 2021-02-09:
Sweet,  I keep an eye out for it.  Many thanks

LeeD76, 2021-02-10:
HI,  Well I couldn't get the Pijuice to work on ubuntu 64bit server, but I did manage to fix the dependencies I was having trouble with on the Raspberry Pi OS, so all is good.  Is there a way I can grab the battery status i.e. how much charge is left in the battery via a  bash or python script?  Sorry I don't really have any experience with Python as I normally dev in Dotnet / C#.  thanks.

tvoverbeek, 2021-02-10:
Here a small Python script to get the battery charge level:
```Python
#!/usr/bin/python3

from pijuice import PiJuice

pj = PiJuice(1,0x14)
charge_level = pj.status.GetChargeLevel()
print(charge_level)
```

LeeD76, 2021-02-10:
epic, many thanks