PiSupply/PiJuice #640: PiJuice wrongly turning off Raspberry Pi while on mains

JolleJolles, 2021-02-10:
Hi. Maybe I missed something, but I have the following issue.

I set up PiJuice to turn off when the battery charge is below 25%. However, it also turns off my Raspberry Pi when it is powered as well via the USB-C port and has a battery below 25% or it is powered via the USB-C port and has no batter at all. Also, when it is running on USB-C power and I remove the battery, it turns off my Raspberry pi immideately. And finally, when I turn on the Raspberry Pi via the SW1 button of the PiJuice with either of above scenarios, I can see the RPi starts-up, I get a couple pings, and then it is turned off immediately again.

Ofcourse I only want the Raspberry Pi to be turned off when it does not also receive power via the USB-C port, irrespective of the minimum charge or existence of a connected battery. How can this be fixed?

My PiJuice settings are:
```
System Task
[x] System task enabled
[x] Minimum change: 25%

System Events
[x] Low charge: SYS_FUNCT_HALT_POW_OFF
```

JolleJolles, 2021-02-10:
**Update**

When I change the Low Charge function to `USER_FUNC1` and have it link to a python script like in the PiJuice test, I get output `'low_charge', '0.0'` when the Raspberry Pi is powered via the USB-C and has no battery.

Does this mean that the System Task 'Minimum charge` does not take into account the availability of power from the USB-C? Is there a way to fix this? Or would I need to write a custom Python script that would only turn off when not on mains? 

Thanks

tvoverbeek, 2021-02-10:
The `low charge` event is intended to be used when running on only battery power.
So your observations are as expected.
If you want to continue on `low charge` when the RPi is still powered you have to use an user function and check the power input status.
See the pijuice.power.GetStatus() function (https://github.com/PiSupply/PiJuice/tree/master/Software#pijuice-status)
and do not shutdown when the powerInput5vIo is present.

JolleJolles, 2021-02-10:
Thanks. Does PiJuice have a setting to link a USER_FUNC based on power input status or would I have to start a monitoring script at boot or alternatively run a cronjob for a python file that checks power input status? I thought this is something implemented for pijuice.

Also, I tried to use the `No power` System Event, but linking that to a user function, such as to blink a led (confirmed to work by also linking to button press), does not do anything when I unplug the power cable. What I am doing wrong here..?

JolleJolles, 2021-02-10:
I discovered that the `No power` event does work with the usb port on the PiJuice itself but not when powered over the GPIO pins from the Raspberry Pi. Is there a 'No power' event for the Raspberry Pi USB power (giving power to PiJuice via GPIO pins) as well? Anyway this is not clear from the documentation. Finally, is there an opposite event from `No power` that could trigger a function when power is connected again?

tvoverbeek, 2021-02-10:
'No power' event is triggered if **both** power inputs (from the PiJuice USB and from the RPi via the 5V GPIO) are absent.

JolleJolles, 2021-02-10:
When my Raspberry Pi is on and powered via Battery power and I then connect power to the USB port of the Raspberry pi, I can see in the pijuice menu there is power over GPIO and when I disconnect again that it again shows no power, and my `No power` script is not executed. But when I do the same with connecting power to the PiJuice USB and removing it again the script ís executed. That is not in line with your statement somehow: 
> 'No power' event is triggered if **both** power inputs (from the PiJuice USB and from the RPi via the 5V GPIO) are absent.

tvoverbeek, 2021-02-10:
Here is the code of the PiJuice service (pijuice_sys.py) which checks for no power:
```
def _EvalPowerInputs(status):
    if (status['battery'] == 'NOT_PRESENT'): return
    global noPowCnt
    NO_POWER_STATUSES = ['NOT_PRESENT', 'BAD']
    if status['powerInput'] in NO_POWER_STATUSES and status['powerInput5vIo'] in NO_POWER_STATUSES:
        noPowCnt = noPowCnt + 1
    else:
        noPowCnt = 0
    if noPowCnt == 2:
        # unplugged
        ExecuteFunc(configData['system_events']['no_power']['function'],
                    'no_power', '')
```
Both inputs have to be BAD or NOT_PRESENT 2 consecutive times before the `no power` event is raised.
The sampling happens every 5 seconds, so ti can take up to 14 seconds before the event is triggered.

JolleJolles, 2021-02-11:
Thanks @tvoverbeek. I will do some more tests then, but that makes sense. 

> Does PiJuice have a setting to link a USER_FUNC based on power input status or would I have to start a monitoring script at boot or alternatively run a cronjob for a python file that checks power input status? I thought this is something implemented for pijuice.
Do you have any more feedback on this point?

Many thanks again

tvoverbeek, 2021-02-11:
USER-FUNCtions ca only be connected to the events on the system events tab in the cli/gui.
If you want to monitor only one of the power inputs you need to run your own periodic monitor job via cron or your own service.

JolleJolles, 2021-02-12:
Thanks for clarifying!

closed 2021-02-12 by JolleJolles