PiSupply/PiJuice #641: Can the Pijuice V2.2 handle 12v battery connected to J2?

ClarkeFL, 2021-02-12:
I want to find out if connecting a 12v battery to the J2 connection will cause any issues. I did read through the documentation and there was no mention under the J2 section how much voltage it can accept. however it does mention the the Charge IC (USB MICRO) is able to handle 4-10V does this mean I theoretically connect a 10v battery and the Pijuice hat will still work and not burn out? 

The reason I'm asking is because I'm developing a system that with a 3.7V battery connected with no other input power reboots when trying to complete its task. However if I attach power from solar or main power through the Charge IC port (USB MICRO) system runs fine as it supplies the gpis with 5v instead of 3.7 from the battery.

Hope someone can give clarification on this. Thank you

shawaj, 2021-02-12:
@ClarkeFL J2 on the PiJuice Zero can take a single cell lithium ion / lithium polymer battery only. These tend to be 3.7v "nominal" and 4.2v max.

The 2 pin J4 header on both the PiJuice HAT and PiJuice Zero can take 4.2 to 10v input. So this would be ok with a 10v battery as an input but it wouldn't be able to recharge the battery directly. It is just an input.

Having said all that, what battery are you using with the PiJuice that causes it to reboot when running a task? It sounds like you either have a faulty battery or you need a larger capacity battery that can handle higher current loads on discharge.

ClarkeFL, 2021-02-12:
@shawaj im currently running a 3.7V 20000mAh battery, I thought it was a faulty battery but replacing it with the new one has not helped. If the battery is fully charged the reboot seldom happens. Unfortunately due to the locations that we could be installing the device we might not have access to power 24/7 so we would have to rely on solar. That being said I need to have the system stable even if the battery is 50% which we hope will never run at that percentage but as soon as the battery lowers below 75% we have the issue.

shawaj, 2021-02-12:
@ClarkeFL do you have a link to the specific battery?

ClarkeFL, 2021-02-13:
@shawaj here is the link https://core-electronics.com.au/3-pin-lipo-battery-for-pijuice-20000mah.html?utm_source=google_shopping&gclid=CjwKCAiA65iBBhB-EiwAW253WyIqWbG6cavLAhHZco3baIqpsV2pzu0tmNWfj9yAUcJSSstP8APnUhoChFwQAvD_BwE

shawaj, 2021-02-13:
@ClarkeFL that looks like it might be a multi cell battery? Is it definitely single cell?

Also, have you set up a custom battery profile in the PiJuice battery settings?

ClarkeFL, 2021-02-13:
@shawaj 

Yes battery is a single cell. Just thicker due to carrying 20A.

Yes I set the battery to custom, where it automatically picked up it's Lipo, with 20A/h.

The battery runs runs okay when it it's fully charged in the system.

Probably something I should have led with is that I do have a dslr camera attached which runs at 7.2v. I step the voltage up double from the 5v5 gpio pins which provides enough for the camera. (Camera is needed for the tasks of the project) still looking if we can find a lower voltage with same specs dslr camera

This is why I think 3.7v battery is not sufficient especially if it hits 50%. Or even 75%

ClarkeFL, 2021-02-14:
Just another question. Does the Pijuice hat step-up voltage to the 5v5 GPIO? for example. The battery is sitting at 3.7V and the GPIO power is sitting at 5V is that correct? I would have thought that it would only sit at the battery voltage.

shawaj, 2021-02-14:
@ClarkeFL the PiJuice steps the voltage up to 5v0 from the battery voltage yes. 

To me that battery looks like it contains 2 cells attached together.

But in any case please try without the camera attached and see if that makes any difference to it. It is probably drawing too much.

Have you considered using the Raspberry Pi high quality camera for this instead of a DSLR?

ClarkeFL, 2021-02-14:
Thank you! 

Could be that there's not enough current going to the camera since there is two points of voltage step up. Ill order a single cell battery. I did look to see if I could find out if the current battery is two cells but all the information I could find points to one. 

I have looked at the Raspberry Pi high-quality camera, but the applications we are going to use this device requires a higher quality image. 

Thank you for all the help!

ClarkeFL, 2021-02-16:
I just want to open this issue again to get an understanding of something quickly.

if I attach two single-cell 10000mAh batteries in parallel to give me 20000mAh and voltage of 3.7 V. Will the Pijuice hat still work? I think that's what this battery is https://core-electronics.com.au/3-pin-lipo-battery-for-pijuice-20000mah.html?utm_source=google_shopping&gclid=CjwKCAiA65iBBhB-EiwAW253WyIqWbG6cavLAhHZco3baIqpsV2pzu0tmNWfj9yAUcJSSstP8APnUhoChFwQAvD_BwE

Or do I need to use just a single cell battery? From more testing, it's come to my attention the battery is at fault and so going to buy a new one and I don't know if I can use the above battery or if I need to use just a single 10000mAh battery? 

The above battery does work and I have been using it for my testing now. I think unfortunately one 10000mAh side of the battery has failed and so it's giving me issues. 

Would be nice to have a higher Amperage especially if I'm stepping up the voltage.

tvoverbeek, 2021-02-16:
Not a good idea.
Google on multi-cell LiPo charging.
The PiJuice does not habe the circuitry to do balanced multi-cell charging.
Also there are limits on the charge current the PiJuice can provide. For 2 large batteries in parallel it will certainly be over the limit resulting in much longer charge times.
Same goes for the output current which can be provided to the load.
You better look for a different solution.
See e.g. https://learn.adafruit.com/multi-cell-lipo-charging?view=all

ClarkeFL, 2021-02-17:
Thank you @tvoverbeek and @shawaj for all the help!

closed 2021-02-17 by ClarkeFL