PiSupply/PiJuice #642: Identifying potential conflicts between hats

bssackmann, 2021-02-15:
A while back I purchased two PiJuice HATs for my projects. My initial application will be headless and I simply need to log data from a NMEA0183 device (a Garmin GPS).  I would like to know whether the PiJuice HAT would be compatible with something like the PICAN-M Hat (https://copperhilltech.com/pican-m-nmea-0183-nmea-2000-hat-for-raspberry-pi/) or whether I should expect hardware conflicts if the boards were to be stacked and used simultaneously.  Thanks.

tvoverbeek, 2021-02-15:
PiJuice needs I2C and power and ground.
The HAT you link to uses SPI but makes the I2C bus available for extra sensors, but it does not use I2C itself as far as I can tell.
So no conflicts unless you connect an extra sensor  with an I2C address that conflicts with the PiJuice I2C addresses.
Note: In the next release it will be easy to change the I2C addresses the PiJuice uses.

bssackmann, 2021-02-15:
Perfect; thanks for the quick reply -- I really appreciate it!