PiSupply/PiJuice #645: Latest firmware stable and downgrade

rascalos, 2021-02-17:
Hello,
I Would like to know what is the lastest firmware stable version, please ?
Is it possible to downgrade a firmware version ? how to do for ?
Some PiJuice with V1.4 firmware seem have issue whereas not in V1.3.
In the first time, I would like upgrade with the latest firmware stable before to downgrade.
Thanks for help.

tvoverbeek, 2021-02-17:
V1.4 is the last stable firmware.

rascalos, 2021-02-17:
Should I use the firmware installation process to downgrade the firmware version as explained here https://github.com/PiSupply/PiJuice/tree/master/Firmware#update-pijuice-firmware ?
Thanks for your help.

tvoverbeek, 2021-02-17:
Yes. With pijuiceboot you can install any version of the firmware.
I guess you are aware that they are available at https://github.com/PiSupply/PiJuice/tree/master/Firmware

rascalos, 2021-02-17:
Thanks a lot.
Pascal

closed 2021-02-17 by rascalos