PiSupply/PiJuice #646: Loosing firmware

pluhacekp, 2021-02-18:
Hi, 

I would like to ask you for help with a strange behaviour I am observing on my PiJuice. I am using PiJuice zero with Rpi Zero WH and 12000mA battery on PiJuice_base CLI install. When checking status of the system by pijuice_cli I often get communication error and opening Battery profile menu often results in crash of the Python script. I have noticed that in the Firmware menu I get reported as "Current version 0.0", running an update is sucessfull and menu shows version 1.4. and in a minute or going back 0.0.

I would not mind this if the wakeup alarm works fine, but thats unfortunatelly not the case. Not sure if its caused by the FW issue, but does look strange.

Any suggestion what to do?

Thanks
Petr

[image]
[image]
[image]

mmilann, 2021-02-19:
Have you tried to reset configuration to default, via cli or dual button press? Also try to disable NTC in battery profile menu, regarding 12000 battery problem with wrong temperature sensor.

pluhacekp, 2021-02-19:
@mmilann  I have tried to load default config both ways, the NTC I have disabled by default always as the temperature reports are wrong for 12000mA battery. No difference, after update FW is fine on 1.4 and next moment its gone. Once its gone the battery profile for 12000mA battery is also gone as it was probably added in later version the FW only. Opening battery menu at times shows shorter list of batteries in other cases it results in script failure as per attached txt. What else to try ?

[trace.txt](https://github.com/PiSupply/PiJuice/files/6010313/trace.txt)

mmilann, 2021-02-19:
@pluhacekp it could be incompatible firmware/software. 1.6 debian package holds firmware V1.4.

Since you are using 12000mA I can suggest to install updated (still in test phase), software package V1.7 from here:
https://github.com/PiSupply/PiJuice/blob/v1.7/Software/Install/pijuice-base_1.7_all.deb
but before remove/purge current version.

After that update firmware to V1.5 from pijuice_cli.

You can than disable NTC same way, or set custom profile parameters found in this JSON file:
https://github.com/PiSupply/PiJuice/blob/v1.7/Software/Configs/Batteries/LIPO_12000_PIS-1129.JSON
this should fix temperature readings also if NTC is enabled.

pluhacekp, 2021-02-19:
ok, will try that, could you please point me to how to do the proper purging of the firmware to avoid that I brick the thing

tvoverbeek, 2021-02-19:
@pluhacekp mmilan is talking about the debian packages, not the firmware.
To purge: `sudo apt purge pijuice-cli pijuice-gui`
Then manually install the pijuice-base_1.7 package: `sudo dpkg -i <path-to>/pijuice-base_1.7_all.deb`.

pluhacekp, 2021-02-19:
@tvoverbeek  @mmilann   I have purged the sw and installed the 1.7 version with 1.5 FW and voila all is stable and working like a charm. I was basically immediatelly upgrading also the FW, but before I quickly checked the status and battery profile menu and all looked good even before upgrade of fw, so it might have been enought to solve  the problem

in any case big thanks for your support

closed 2021-02-19 by pluhacekp