PiSupply/PiJuice #647: COMMUNICATION_ERROR; "error receiving data -1" when trying to update firmware

johnpeart, 2021-02-19:
I've recently starting using the PiJuice HAT for a project. I am using Raspberry Pi OS on a Pi 4B. The only other hardware attached is a HQ Camera module. I'm using the default battery (1820mAh).

The Pi will not power on successfully when a micro-USB cable is connected via the PiJuice, but a green and blue LED is alternating when the cable is present.

The firmware reports that it is running v1.4 when using `pijuice_cli`. 

When trying to open the status in the CLI or via a Python print command, I see a `COMMUNICATION_ERROR`. 

I have attempted to update and flash the firmware, searching for similar errors in the issues log on this repo. I've attempted to do so through the `pijuice_cli` interface and directly through terminal commands. Unfortunately it hasn't worked. The former reports `STARTING_BOOTLADER_ERROR` (though I'm pretty confident I have pressed SW3 correctly!) and the latter produces an as per below:

```
Sending get command
error receiving data -1
```

Checking the I2C detect shows all entries `--`; though I2C interface is on.

Could you offer any advice please?

Thanks

mmilann, 2021-02-19:
Do you get any response from PiJuice, like status reported in pijuice_cli window, or it always shows "COMMUNICATION_ERROR"?

johnpeart, 2021-02-19:
Hi @mmilann 

In `pijuice_cli` in the `Status` menu I see the following information when the battery is plugged in and the power is provided through the Pi 4B's USB-C port:

```
HAT status

Battery: -1%
GPIO power input: 0.000V, 0.000A, N/A
USB Micro power input: 0
Fault: COMMUNICATION_ERROR
System switch: NonemA
```

It does that regardless of what I have tried to do to fix it.

Thanks for your help.

mmilann, 2021-02-19:
Can you run 
i2cdetect -y 1
?

johnpeart, 2021-02-19:
This is the output I get (everything is empty)!

```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- 
```

mmilann, 2021-02-19:
It may be firmware damaged so no response. You can try to activate bootloader manually before running pijuiceboot tool to write correct firmware. Disconnect power from Rpi and PiJuice and remove battery, than press and hold SW3 pressed, connect battery while holding SW3 pressed to power-up pijuice and than release SW3. Now you can power-up Rpi by plugging to power source via USB Type-C and than try to load firmware with command:

pijuiceboot 14 ./PiJuice-V1.4_2020_01_17.elf.binary

johnpeart, 2021-02-19:
Thanks. I've given that a go (three times!) and it produces the same error each time. Terminal output is:

```
using address 14
Input file ./PiJuice/Firmware/PiJuice-V1.4_2020_01_17.elf.binary
Starting bootloader 
Starting bootloader old firmware
Sending get command 
error receiving data -1
```

mmilann, 2021-02-19:
Ok, procedure with holding SW3 pressed while connecting power is important to launch bootloader manually, so make sure you have done it correctly before running pijuiceboot.

It may be that something is broken. You can do several checks:
- Measure 3V3 voltage at P2 connector, correct value is 3.3V +-2%.
- Measure voltage on SCL and SDA GPIO pins, should be around 3.3V.

mmilann, 2021-02-19:
Also you mention that  green and blue LED is alternating when the cable is present. So if firmware is running you can try to reset configuration, press and hold SW1 and SW2 simultaneously for 20 seconds, you will see all LEDs blinking when it resets.

johnpeart, 2021-02-19:
No dice on the SW1/SW2 reset unfortunately and I don't own a volt meter to be able to test the pins.

Out of curiosity if I'd broken through my wider activities, I've also tried starting from scratch with a clean OS install, and that hasn't resolved the errors. However, I have noticed a new error message that might make more sense to you than me (!):

During `sudo apt-get install pijuice-base` as part of the console output I saw the following error:

```
Enabling I2C
modprobe: ERROR:  ../libkmod/libkmod.c:586 mod_search_moddep() could not open mode file '/lib/modumodules/5.4.83-v71+modules.dep.bin'
modprobe: FATAL: Module i2c-dev not found in directory /lib/modules/5.4.83v71+
```

I'm guessing this might be why it isn't working? Any thoughts?

johnpeart, 2021-02-19:
Still not working. 

*smashes head on desk*

Terminal update fails with same error as before.
`pijuice_cli` update fails with same error as before. Selecting any options through most of the menus crashes the CLI and throws attribute errors (different for each menu item).
SW1/2 reset has no effect (does nothing if attempted when cable is in the Pi, and flashes but then no change if cable is in the PiJuice).

Any other thoughts or do you think it's a write off? (Have to say if it's this easily bricked I won't be buying another!)

mmilann, 2021-02-19:
May be os problem with i2c. pijuice-base package install should enable I2C module,  you can also try to enable via raspi_config menu, or reinstall pijuice-base.

johnpeart, 2021-02-19:
Thanks for your help. I'll order an extra Pi to see if it's an issue with that (I'll end up using it eventually anyway if it isn't!).

Will let you know how I get on.

tvoverbeek, 2021-02-19:
@johnpeart Are the i2c modules loaded? What is the output from `lsmod | grep i2c`?
Does `/dev/i2c-1` exist? (I guess it does, otherwise i2cdetect would not owrk).

johnpeart, 2021-02-19:
@tvoverbeek the output from `lsmod | grep i2c` is:

```
i2c_bcm2835    16384   1
i2c_dev        20480   2
```

I've pulled another Pi from a separate project and tried another fresh install to see if the problem is replicated: identical problems exhibited so it must be the PiJuice board or battery that must be the source of the fault, unless I'm missing something really, really obvious. 

The pins on the PiJuice do have power because I can trigger Python scripts using the GPIO Zero library with a 'normally open' button attached to physical pins 39 and 40. I fear the solution is far beyond my abilities! [*insert confused emoji*]

My only other thought is that I have soldered two separate wires to the terminals between SW1 and SW3, and loosely attached them to a separate N.O. button. The button functions exactly as it should, if I've understood the guidance correctly (replicating the functions of SW1). Could the soldering of those two wires have damaged the board to cause these errors? I can't see any obvious damage, but I don't know what's close to those terminals.

mmilann, 2021-02-20:
In this case something with I2C lines can be faulty, because as you described pijuice shows correct LED status blue blink when charging so it may be operational, except I2C communication.

Could be some handling damage, like ESD, or miss-connection when attached to Rpi, Also sometimes I2C GPIO2/3 on Rpi can be damaged. Anyway you get access to voltmeter you can check these points.

Jamos1988, 2021-03-20:
In my case I was running the newest pijuice software, everything working fine. When updating through the GUI from 1.4 to 1.5 it instantly said EEPROM_ERASE_ERROR

I then went on with the manual flashing process, and no luck. All my other i2c sensors are working fine so it is not the i2c bus. I've followed the sw3 instructions clearly, Pijuice LEDs are off so this should be good. Also 3.4 volts on the sda scl lines.

First time trying it was stuck for a long time on sending get command, but now it just instantly goes to error receiving data.

My setup:
compute module cm3+ (same internally as 3b+)
battery is pijuice 12000mah, this is the reason I needed 1.5 because it was not charging due to the temperature -9 celcius issue.

I hope sincerely this can be fixed. I have created this issue: https://github.com/PiSupply/PiJuice/issues/690

mmilann, 2021-03-22:
Can you measure voltage at P2-3V3 pin on pijuice while battery is connected and power source to pijuice USB micro?

Jamos1988, 2021-03-22:
Voltages are fine, 3.4v also on sdl sda likes.

mmilann, 2021-03-22:
Do you have photo of your setup?

Jamos1988, 2021-03-22:
Not at the moment, but it is a 3b+ and the pi directly on top of the gpio
header. Battery removed at the moment. It is feeded by J4, and a 3a psu.
Also tried with gpio power and regular USB cable.

Also tried the same on compute module.

tvoverbeek, 2021-03-22:
@mmilann @Jamos1988 If you see address 0x41 then the STM32 gets confused by the first commands sent by pijuiceboot (Starting bootloader and Starting bootloader old firmware).
I believe in your case you are already in the bootloader.
In the pijuiceboot source (https://github.com/PiSupply/PiJuice/blob/master/Firmware/pijuiceboot.c)
comment out the 2 write commands on lines 935 and 939 and rebuild pijuiceboot.
@mmilann Can you confirm that these 2 commands might cause problems if the firmware is (partly) erased?
I hope it works witth these 2 commands removed, so the first command sent is the GET command.