PiSupply/PiJuice #648: Failed to execute command "/usr/bin/pijuice_gui"

ro0TuX777, 2021-02-19:
**Complete error message when attempting to load it from the menu:** 
_Failed to execute child process "/usr/bin/pijuice_gui" (No such file or directory)_

**Complete error message when attempting to load it from the terminal window:**
_bash: /usr/bin/pijuice_gui: No such file or directory_

About the computer:
Kali version 2021.1
Kernel: Linux 4.19.118-Re4son-vl+
Architecture: arm
uname -m: armv7l

Within the /usr/bin folder the following pijuice files exist:
pijuice_cli, pijuice_gui, pijuice_sys.pi, pijuice_tray.py, pijuiceboot

Instructions followed: https://github.com/PiSupply/PiJuice/issues/405

Update sudo apt update then install dependancies sudo apt install i2c-tools sudo apt install lua5.3 Add user root to group pijuice run sudo visudo add root ALL=(pijuice) ALL save and exit. Clone the repository git clone https://github.com/PiSupply/PiJuice.git install pijuice base from cd PiJuice/Software/Install run dpkg -i pijuice-base_1.5_all.deb Fix broken packages with sudo apt --fix-broken install reboot and you should be able to run pijuice_cli from the terminal.

Once pijuice-base has been installed you can then install the GUI by running dpkg -i pijuice-gui_1.5_all.deb GUI requires Python3-tk install with sudo apt --fix-broken install reboot. PiJuice will be added to the application menu and you can run from there or from terminal with pijuice_gui.

tvoverbeek, 2021-02-19:
Something went wrong with the installation.
In /usr/bin I am missing pijuice_cli.py and pijuice_gui.py.
Also current version of the packages is 1.6 and not 1.5.

ro0TuX777, 2021-02-20:
I get the same error message with 1.6 as well.

tvoverbeek, 2021-02-20:
In a terminal window on the desktop what is the output from `ls -l /usr/bin/pijuice*?
Here is what I get:
```
-rwxr-xr-x 1 root    root     20344 Feb  6  2020 /usr/bin/pijuiceboot
-rwsr-xr-x 1 pijuice pijuice   9096 Feb  6  2020 /usr/bin/pijuice_cli
-rw-r--r-- 1 pijuice pijuice 101267 Feb  6  2020 /usr/bin/pijuice_cli.py
-rwsr-xr-x 1 pijuice pijuice   9096 Dec 17  2018 /usr/bin/pijuice_gui
-rw-r--r-- 1 pijuice pijuice 105156 Dec 24 01:00 /usr/bin/pijuice_gui.py
-rwxr-xr-x 1 root    root     13461 Feb  6  2020 /usr/bin/pijuice_sys.py
-rwxr-xr-x 1 root    root      6026 Dec 24 01:00 /usr/bin/pijuice_tray.py
```
What error message do you get when running from a terminal window `/usr/bin/pijuice_gui`?

ro0TuX777, 2021-02-20:
Um, mine looks a little different. 
-rwsr-xr-x 1 pijuice pijuice  9096 Dec 17  2018 pijuice_cli
-rw-r--r-- 1 pijuice pijuice 98358 Feb 15 15:14 pijuice_cli.py
-rwsr-xr-x 1 pijuice pijuice  9096 Dec 17  2018 pijuice_gui
[image]

-rw-r--r-- 1 pijuice pijuice 99437 Feb 15 15:13 pijuice_gui.py
-rwxr-xr-x 1 root    root    12051 Feb 15 15:13 pijuice_sys.py
-rwxr-xr-x 1 root    root     5133 Feb 15 15:13 pijuice_tray.py
-rwxr-xr-x 1 root    root    20344 Mar 16  2018 pijuiceboot

tvoverbeek, 2021-02-20:
The permissions look fine. No need to change anything there.
What error message do you get if you run `/usr/bin/pijuice-gui` in a terminal window?
Does running `/usr/bin/pijuice_cli` in a terminal window work?

ro0TuX777, 2021-02-20:
Complete error message when attempting to load it from the terminal window:
bash: /usr/bin/pijuice_gui: No such file or directory

Same for thr cli

tvoverbeek, 2021-02-20:
OK. Next test: In a terminal window: `sudo -u pijuice python3 /usr/bin/pijuice_gui.py`.
Should produce python3 error messages which might help us.

ro0TuX777, 2021-02-20:
kali@kali:/usr/bin$ sudo -u pijuice python3 /usr/bin/pijuice.py

python3: can't open file '/usr/bin/pijuice.py': [Errno 2] No such file or directory

tvoverbeek, 2021-02-20:
Thus helps.
What version of python3 are you running on Kali? (`python3 --version`)
Is it 3.9?

ro0TuX777, 2021-02-20:
Python 3.9.1+

tvoverbeek, 2021-02-20:
See my edited comment above.
In /usr/lib/python3/dist-packages/pijuice.py change the 3 occurrences of `self.t.isAlive()` in _DoTransfer (lines 70-84) to `self.t.is_alive()`.
That should solve it for both pijuice_gui and pijuice_cli.

I will update the source on github accordingly.
Thanks for catching this.

ro0TuX777, 2021-02-20:
[image]

ro0TuX777, 2021-02-20:
tvoverbeek, you are AWESOME!! cheers good sir! CHEERS!!

ro0TuX777, 2021-02-21:
Just to follow up on this, I able to launch the gui with the command: sudo -u pijuice python3 /usr/bin/pijuice_gui.py. The application will not launch from the menu bar. The error message when attempting will be 
_Failed to execute child process "/usr/bin/pijuice_gui" (No such file or directory)_

tvoverbeek, 2021-02-22:
@ro0TuX777 Are you by any chance running the 64-bit version of Kali?
The executables pijuice_gui and pijuice_cli are 32-bit only. They are setuid pijuice and drop all privileges before executing python3 /usr/bin/pijuice_gui.py or python3 /usr/bin/pijuice_cli.py..
In that case you have to rebuild the excutables with the build script in https://github.com/PiSupply/PiJuice/tree/master/Software/Source/Setuid-Wrapper.
See also https://github.com/PiSupply/PiJuice/blob/master/Software/OS%20Support/ubuntu_server_64.MD

ro0TuX777, 2021-02-22:
Thanks for the follow up, I am using 32bit. uname -m: armv7l

tvoverbeek, 2021-02-22:
When you start /usr/bin/pijuice_gui (or /usr/bin/pijuice_cli) from a terminal window, do you get the same error message?

ro0TuX777, 2021-02-22:
The erro message for both cli and gui show: "No such file or directory"

[image]

tvoverbeek, 2021-02-22:
I do not know what is  going on here.
In your post above (https://github.com/PiSupply/PiJuice/issues/648#issuecomment-782710560) the permissions and ownerships are all ok.
What is the output of `file /usr/bin/pijuice_gui`?
This is what I get:
```
/usr/bin/pijuice_gui: setuid ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV),
dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 3.2.0, 
BuildID[sha1]=eaac0a2132e4f8baa7c7a157a1e46d922d2e0d3f, not stripped
```

ro0TuX777, 2021-02-23:
kali@kali:/usr/bin$ file pijuice_gui
pijuice_gui: setuid ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 3.2.0, BuildID[sha1]=eaac0a2132e4f8baa7c7a157a1e46d922d2e0d3f, not stripped

tvoverbeek, 2021-02-23:
OK. That is as expected.
Next thing to check is if Kali uses the same dynamic loader.
What is the output from `ls -al /lib/ld*` and from `ldd /usr/bin/pijuice_gui`?
Here is my output:
```
pi@raspberrypi:~ $ ls -al /lib/ld*
lrwxrwxrwx 1 root root 30 May 14  2019 /lib/ld-linux-armhf.so.3 -> arm-linux-gnueabihf/ld-2.28.so
lrwxrwxrwx 1 root root 24 May 27  2020 /lib/ld-linux.so.3 -> /lib/ld-linux-armhf.so.3
pi@raspberrypi:~ $ ldd /usr/bin/pijuice_gui
	linux-vdso.so.1 (0xbef85000)
	/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so => /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so (0xb6f12000)
	libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0xb6dae000)
	/lib/ld-linux-armhf.so.3 (0xb6f27000)
pi@raspberrypi:~ $ 
```

ro0TuX777, 2021-02-26:
[image]

tvoverbeek, 2021-02-26:
Here you have your problem.
pijuice_gui is looking for /lib/ld-linux-armhf.so.3 which does not exist on your system
You can try to add a symbolic link in /lib:
`sudo ln -s arm-linux-gnueabi/ld-2.31.so ld_linux-armhf.so.3`
If that does not work you will have to rebuild the 2 wrappers pijuice_cli and pijuice_gui.
See https://github.com/PiSupply/PiJuice/tree/master/Software/Source/Setuid-Wrapper where you find the source and the build script. Resulting programs have to be installed in /usr/bin with setuid pijuice.

ro0TuX777, 2021-03-10:
thanks, attempting to add the symbolic link. Did not work and i'm not familiar with how to rebuild the two wrappers. But will tread carefully with following the instructions.

Further when trying to execute the command:sudo -u pijuice python3 /usr/bin/pijuice_gui.py
the new error is:

[image]

tvoverbeek, 2021-03-10:
It looks like something is missing for the python tkinter module.
It definitely requires that Tk/Tcl is installed.
See also https://docs.python.org/3.7/library/tkinter.html
On Raspbian Buster the current version of python3 is 3.7.3 not 3.5

BramWerbrouck, 2024-04-23:
Hi,

I have the same problem "failed to execute /usr/bin/pijuice_gui on a fresh installation on Bullseye V11, 32-bit

I read all the comments above, bit none were helping me out. I checked the /usr/lib/python3/dist-packages/pijuice.py file, but I have already the command self.t.is_alive().

Is there something else I can try to get this up and running?

tvoverbeek, 2024-04-24:
See issue #1000 for a solution.