PiSupply/PiJuice #649: need power-on restart

rjweber-UCB, 2021-02-20:
I could use some pointers for my PiJuice HAT:

My application in this project is to guard against SD corruption if power is pulled unexpectedly.  What I want is:
1) Upon loss of line power the Pi politely closes down to a shutdown state.
2) Upon line power restored the Pi restarts without pushing buttons on the HAT, etc.

Number 1 is easy enough.  PiJuice Settings->System Events->No Power[enabled]->SYS-FUNC_HALT_POW_OFF

But for number 2, I am not finding a way to reset or power-cycle the Pi on line power restored.  I'm not seeing any kind of line-power-restored event.

This seems like a really basic application.  What am I missing?

Thanks in advance!

tvoverbeek, 2021-02-20:
Where is line power connected? To the RPi or the PiJuice?
If connected to the RPi you get this automatically.
A different story when connected to the PiJuice.

rjweber-UCB, 2021-02-20:
@tvoverbeek:  Thank you sir, my problem is solved using the script you shared [here ](https://github.com/PiSupply/PiJuice/issues/357).  Line power is connected to the PiJuice.  Using your userscript and the settings you recommend, it all works just as I wanted.

I was not able to get a shutdown/restart with the power plugged into the Pi as the battery kicked in and kept the Pi running.  The real answer may be simpler as you suggest, but I've got the problem solved and this is a big project.

Again, thanks very much!

closed 2021-02-20 by rjweber-UCB