PiSupply/PiJuice #651: RTC pijuice late wakeup

mmilann, 2021-02-22:
Continue with #643. @tvoverbeek lets try module reload so we can test before V1.7 release.

tvoverbeek, 2021-02-22:
@mmilann Just committed the needed changes to the 1.7 branch.
Below some test results.
**Case 1: Pi off, PiJuice in low power, Apply power to Pi**
Excerpt from dmesg:
```
[    7.142146] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    7.154554] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Jan  4 2021 19:56:29 version 7.45.229 (617f1f5 CY) FWID 01-2dbd9d2e
[    7.710239] rtc-ds1307 1-0068: Unsupported ohm value 0 in dt                            <----
[    7.710928] rtc-ds1307: probe of 1-0068 failed with error -121                          <----
[    9.490025] uart-pl011 fe201000.serial: no DMA platform data
[    9.511394] random: crng init done
[    9.511416] random: 7 urandom warning(s) missed due to ratelimiting
[    9.823751] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[    9.863888] 8021q: 802.1Q VLAN Support v1.8
[   10.347462] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   10.643347] bcmgenet fd580000.ethernet: configuring instance for external RGMII (RX delay)
[   10.643779] bcmgenet fd580000.ethernet eth0: Link is Down
[   15.081373] rtc-ds1307 1-0068: Unsupported ohm value 0 in dt                            <----
[   15.083630] rtc-ds1307 1-0068: 'wakeup-source' is set, request for an IRQ is disabled!  <----
[   15.088475] rtc-ds1307 1-0068: registered as rtc0                                       <----
[   15.903826] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   16.359778] Bluetooth: Core ver 2.22
[   16.359872] NET: Registered protocol family 31
[   16.359883] Bluetooth: HCI device and connection manager initialized
[   16.360315] Bluetooth: HCI socket layer initialized
[   16.360331] Bluetooth: L2CAP socket layer initialized
[   16.360358] Bluetooth: SCO socket layer initialized
```
Note the failed original probe at 7.71 and the successful probe from pijuice.service at 15.08

Output from `systemctl status pijuice`:
```
pi@raspberrypi:~ $ systemctl status pijuice
● pijuice.service - PiJuice status service
   Loaded: loaded (/lib/systemd/system/pijuice.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2021-02-22 18:34:45 CET; 7min ago
 Main PID: 403 (pijuice_sys.py)
    Tasks: 1 (limit: 3860)
   CGroup: /system.slice/pijuice.service
           └─403 /usr/bin/python3 /usr/bin/pijuice_sys.py

Feb 22 18:34:45 raspberrypi systemd[1]: Started PiJuice status service.
Feb 22 18:34:50 raspberrypi sudo[555]:  pijuice : TTY=unknown ; PWD=/var/lib/pijuice ; USER=root ; COMMAND=/sbin/modprobe -r rtc_ds1307
Feb 22 18:34:50 raspberrypi sudo[555]: pam_unix(sudo:session): session opened for user root by (uid=0)
Feb 22 18:34:50 raspberrypi sudo[555]: pam_unix(sudo:session): session closed for user root
Feb 22 18:34:50 raspberrypi sudo[560]:  pijuice : TTY=unknown ; PWD=/var/lib/pijuice ; USER=root ; COMMAND=/sbin/modprobe rtc_ds1307
Feb 22 18:34:50 raspberrypi sudo[560]: pam_unix(sudo:session): session opened for user root by (uid=0)
Feb 22 18:34:50 raspberrypi sudo[560]: pam_unix(sudo:session): session closed for user root
Feb 22 18:34:50 raspberrypi pijuice_sys.py[403]: rtc_ds1307 mdule reloaded and RTC os-support OK
pi@raspberrypi:~ $ 
```
You see the rtc_ds1307 module removal and reload and the success message at the end.

**Case 2: Pi off, PiJuice in run mode, start boot** 
Excerpt from dmesg:
```
[    7.362845] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    7.372053] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Jan  4 2021 19:56:29 version 7.45.229 (617f1f5 CY) FWID 01-2dbd9d2e
[    7.783230] rtc-ds1307 1-0068: Unsupported ohm value 0 in dt                            <----
[    7.790714] rtc-ds1307 1-0068: 'wakeup-source' is set, request for an IRQ is disabled!  <----
[    7.796093] rtc-ds1307 1-0068: registered as rtc0                                       <----
[    9.325887] random: crng init done
[    9.325907] random: 7 urandom warning(s) missed due to ratelimiting
[    9.448134] uart-pl011 fe201000.serial: no DMA platform data
```
First probe succeeds.

Output from `systemctl status pijuice`:
```
pi@raspberrypi:~ $ systemctl status pijuice
● pijuice.service - PiJuice status service
   Loaded: loaded (/lib/systemd/system/pijuice.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2021-02-22 18:44:02 CET; 3min 45s ago
 Main PID: 442 (pijuice_sys.py)
    Tasks: 1 (limit: 3860)
   CGroup: /system.slice/pijuice.service
           └─442 /usr/bin/python3 /usr/bin/pijuice_sys.py

Feb 22 18:44:02 raspberrypi systemd[1]: Started PiJuice status service.
Feb 22 18:44:07 raspberrypi pijuice_sys.py[442]: RTC os-support OK
pi@raspberrypi:~ $ 
```
Note the success message

Next I will rebuild the packages and see if they install properly.
If so, I guess we could make an official release @shawaj

mmilann, 2021-02-22:
@tvoverbeek Looks like works very well, could be part of this release.

tvoverbeek, 2021-02-23:
Rebuild the packages after the last (?) commits to make changing the i2c address work on the fly (pijuice_sys and pijuice_tray informed about the address change).

mmilann, 2021-02-24:
Used to test with use case: 
 - no power connected initially
 - Rpi is in off state, GPIO 5V off
 - PiJuice in standby mode
Then connecting power to Rpi USB input:
 - Rpi boots
 - PiJuice stays few seconds  in standby then switches to run mode
When booted checking:
`hwclock -r`
and for Rpi 4 works perfectly, every time.
For older Rpi 3 it failes sometimes with dmesg shows:
```
[    8.850466] rtc-ds1307 1-0068: Unsupported ohm value 0 in dt
[    8.853291] rtc-ds1307: probe of 1-0068 failed with error -121
[   10.358216] uart-pl011 3f201000.serial: no DMA platform data
[   10.774563] 8021q: 802.1Q VLAN Support v1.8
[   11.606852] Adding 102396k swap on /var/swap.  Priority:-2 extents:2 across:1                                                                                                                                                             31072k SSFS
[   11.665717] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   12.688051] 8021q: adding VLAN 0 to HW filter on device eth0
[   12.691092] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   17.020009] rtc-ds1307 1-0068: Unsupported ohm value 0 in dt
[   18.076720] i2c-bcm2835 3f804000.i2c: i2c transfer timed out
[   18.076817] rtc-ds1307: probe of 1-0068 failed with error -110
```
I guess it is something correlated with core speed or order of modules loaded.

Anyway Rpi 4 is more important and this is great improvement for this release.

tvoverbeek, 2021-03-01:
@mmilann Tried to reproduce it on my Pi 3B+, but it works every time.
Were your tests on a 3B or a 3B+?

mmilann, 2021-03-01:
@tvoverbeek much more noticeable on older Rpi 3 using same sd card, kernel 5.4.61. Usually over 10 seconds delay of pijuice exiting standby.