PiSupply/PiJuice #652: PiJuice not working with Rpi4

geoffr98, 2021-02-23:
I have the PiJuice working as follows with the Rpi3, but when moving it over the Rpi4 it doesn't work.

- Power applied to PiJuice J4
- System task enabled
- Wakeup on Charge 10%
- Min Charge 5%
- Min battery voltage 3.3V
- No Power - USER_FUNC1
- User_FUNC1 - python script

With the Rpi4 if I allow the battery to charge the Rpi4 will not boot (it cycles)
If I turn off the battery charging it will boot, but when I remove power the script is not run
I did not change the firmware on the PiJuice when moving from the RPi3 to RPi4
(the script runs fine If I manually run it)

Any suggestions on where to look into this problem?

tvoverbeek, 2021-02-23:
Pi4 is more power hungry than a Pi3.
My guess is the power draw at boot makes the battery level drop below the 5% min charge level.
Try increasing the wakeup on charge and min charge to 30% and 20% respectively and see if that works on the Pi4

geoffr98, 2021-02-23:
I changed the wake up to 70% and the Min to 20%. With those and charging enabled it still doesn't boot, just cycles. If I turn off charging it works as it did before.

tvoverbeek, 2021-02-23:
I suppose you are still supplying power via J4 on the PiJuice.
So the external supply has to provide power for the PiJuice, the charging current (if enabled) and the power to the RPI trough the 5V converter on the PiJuice.
Smells like this sum is too high for the Pi4, but the Pi3 just makes it.
Is it possible in your setup to provide power directly to the RPi? Of course you need a USB-C plug for the Pi4.

geoffr98, 2021-02-23:
Yes I am supplying 5V (up to 3A) on J4 - when I hooked the same supply up to the GPIO it worked.

Have just had it all powered up and charging the battery for a couple of hours and I just unplugged and plugged back in (to J4) and it powered up correctly this time.. not sure what changed...

I do still have the problem that my power down script isn't getting run when I remove power.

geoffr98, 2021-02-23:
I could power the Pi via it's USBC, but then would the PiJuice still work?

tvoverbeek, 2021-02-23:
Yes, power to the PiJuice then comes from the Pi via the GPIO to the PiJuice.But your hook-up to the GPIO is almost equivalent (except you do not have the 5V regulation from the RPi)

geoffr98, 2021-03-13:
So I have tried both the GPIO power and RPi USB as the power source. In both cases when I remove the power the RPi shuts down immediately - so for some reason the PiJuice isn't powering it - any ideas?  Since it works on a RPi3 I know it's fine, so it's something to do with the RPi4.  Do I need the Run pin?
SW is 1.6
FW is 1.4
OS is Linux raspberrypi 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux

tvoverbeek, 2021-03-13:
Do you have extra peripherals? E.g. 7 inch CSI display, SSD.
Their poeer needs for these are probably too high to be supplied by the battery (assuming the default BP7X)

geoffr98, 2021-03-13:
No - Only "Extras" are a USB mouse dongle, Ethernet, HDMI monitor.

Geoff

tvoverbeek, 2021-03-13:
You wrote the same setup works on Pi3. Are you sure the settings are identical between Pi3 and Pi4?
Compare /var/lib/pijuice/pijuice_config.JSON on both systems. You need to be root due the file protections.

geoffr98, 2021-03-13:
Yes the settings are identical. I think the problem is that for the RPi4 I can't power it from the PiJuice header/usb.
Here are the details:

========
RPI3 Test:
 
 - plug in power on J4 - it powers up
     - delete old result file (run on shutdown)
 - remove power -
 - Wait until green LED on PiJuice stops flashing
 - plug in power on J4 - it powers up
      - result file is there :)

PiJuice
 - Fw 1.4
 - sw 1.6
 - OS Linux raspberrypi 5.4.79-v7+ #1373 SMP Mon Nov 23 13:22:33 GMT 2020 armv7l GNU/Linux
 
 PI4:
 - Will not power up correctly with J4 (with Battery Installed)
       - It will power up without the battery, but I need to press SW1
 - Will not power up correctly with PiJuice USB power input
 
- Power up using Rpi GPIO
- remove power, shuts down immediately
- Power up using Rpi GPIO
   - result file is not there 

PiJuice
 - FW 1.4
- SW 1.6
 - OS Linux raspberrypi 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux

Config:
pi@raspberrypi:~ $ sudo cat /var/lib/pijuice/pijuice_config.JSON
{
  "system_task": {
    "watchdog": {
      "period": ""
    },
    "wakeup_on_charge": {
      "trigger_level": "10",
      "enabled": true
    },
    "min_charge": {
      "threshold": "5",
      "enabled": true
    },
    "min_bat_voltage": {
      "threshold": "3.3",
      "enabled": true
    },
    "ext_halt_power_off": {
      "period": ""
    },
    "enabled": true
  },
  "user_functions": {
    "USER_FUNC1": "/home/pi/PiJuice_Shutdown.py",
    "USER_FUNC2": "",
    "USER_FUNC3": "",
    "USER_FUNC4": "",
    "USER_FUNC5": "",
    "USER_FUNC6": "",
    "USER_FUNC7": "",
    "USER_FUNC8": "",
    "USER_FUNC9": "",
    "USER_FUNC10": "",
    "USER_FUNC11": "",
    "USER_FUNC12": "",
    "USER_FUNC13": "",
    "USER_FUNC14": "",
    "USER_FUNC15": ""
  },
  "system_events": {
    "no_power": {
      "enabled": true,
      "function": "USER_FUNC1"
    },
    "forced_power_off": {
      "enabled": false
    },
    "button_power_off": {
      "enabled": true,
      "function": "USER_FUNC1"
    }
  }
}pi@raspberrypi:~ $ 

Shutdown Script:

pi@raspberrypi:~ $ cat PiJuice_Shutdown.py 
#!/usr/bin/python3

import os
import time
from pijuice import PiJuice

os.system("echo 'PiJuice Shutdown output '  > /home/pi/result")

pj = PiJuice(1, 0x14) # Instantiate PiJuice interface object

# Shutdown OpenCPN
ret1 = os.system("killall opencpn")
msg = "killall opencpn \n" + str(ret1) 
cmd = "echo '" + msg + "' >> /home/pi/result"
#os.system("rm result")
#os.system(cmd)
OCPN =  0
while(OCPN != 256 ):
        OCPN = os.system("pgrep -x opencpn") 
        time.sleep(1)
        cmd = "echo 'wait " + str(OCPN) + "'  >> /home/pi/result"
        os.system(cmd)

os.system("echo 'OPCN is not running now'  >> /home/pi/result")

time.sleep(10)
print("Shutting down")
os.system("echo 'Shutting Down '  >> /home/pi/result")
time.sleep(1)

#Remove Pi power after 30 seconds
pj.power.SetPowerOff(30)
os.system("echo 'pj(30) '  >> /home/pi/result")

# Shutdown
os.system("sudo halt")
pi@raspberrypi:~ $

tvoverbeek, 2021-03-13:
Is your HDMI monitor powered separately? I suppose it is a special one for your maritime charts.
Could still be that Pi3 is just above the voltage lower limit and Pi4 just below.
One thing you could try on the Pi4 is to install the 1.5 firmware by installing the 17 debian packages. This firmware accepts a little bit lower voltage when powered via the GPIO. After the 1.7 installation you can update the firmware to 1.5.
From the release notes:
```
- Fixed issue with wrongly marking GPIO input power as BAD when the power supply
      voltage is close to the threshold. DPM voltage is changed from 4.76V to 4.68V.
```
You can find the 1.7 packages here: https://github.com/PiSupply/PiJuice/tree/v1.7/Software/Install
Note: they will be uploaded to the Raspberry Pi OS repo in the near future

geoffr98, 2021-03-13:
I'm using a monitor which is powered. So the Pi shouldn't be powering it..  
            https://lilliputweb.net/a7s-full-hd-7-inch-monitor-with-4k-camera-assist/
            Not the same as on the boat as I'm testing this at home right now.
            
Just to clarify - on RPi3 I was not using the GPIO for power, but J4 on the PiJuice. I'll try the GPIO to confirm it works there as well.

I'll try sw 1.7 and let you know how it goes.

geoffr98, 2021-03-13:
Using the GPIO for the power on the RPi 3 is not working.
 - Powers up fine (as expected)
 - As soon as power is remove it shuts down (no result file is created)
 - It then seems to try to power up again, but is in a failed loop (I see no signal, then color gradient on the screen, and repeat)

geoffr98, 2021-03-13:
On the RPi 4 with (updated) SW to 1.7 and FW 1.5 there is no change.

I did try another experiment (no success thou):
 - apply power to RPi 4 USB (so it boots correctly)
 -  apply power to PiJuice J4
 - Remove RPi4 USB power --> Everything shuts down even though J4 on the PiJuice has power..

tvoverbeek, 2021-03-14:
The last experiments outcome is to be expected.
The firmware still sees power coming from the GPIO to the Pijuice and assumes it comes from the RPI.
Basically providing external power via the GPIO is not supported in the firmware.

tvoverbeek, 2021-03-14:
I am out of suggestions and wondering if there might be a faulty hardware after all.
You might ant to contact your supplier and see if you can get a replacement.

mgrey, 2021-03-14:
I have a Pi4 that works fine with PiJuice (no display)
I have power via PiJuice USB Micro which I guess is the same thing as J4.
My config. is 50% min charge level and 90% wake-up.
I've set the highest DPM value 4.76V and regulator mode to DCDC.
This will create the least 'competition' between supply and charge, at least I thought so.
I see no problem that the original BP7X battery can cope. Measured with oscilloscope I see no dips in voltage.
Halt and wakeup works fine and occurs at the set levels.
FW is 1.4

geoffr98, 2021-03-15:
mgrey - thanks, your settings do not help :)  I fear Ton van Overbeek may be correct and I have a hardware problem.

I have confirmed if I turn off the battery charging I can boot the Rpi4 using J4.. which points to something wrong with battery charging/discharge I think.. I'll ask UK PI Supply as they built it..