PiSupply/PiJuice #653: PiJuice Zero output current

jan94, 2021-02-23:
Hi :)
I have a question regarding the max current output of the PiJuice Zero, as I need a remote power source for a Zero W that makes use of a sixfab LTE Hat.

At pi-supply.com  [https://uk.pi-supply.com/products/pijuice-standard?lang=de](url)  the max current for the PiJuice is specified with 2.5 A and 2.1 A at VSYS (whatever VSYS stands for), but for the PIJuice Zero there is no such information provided.

Can you tell me what the max current output of the PiJuice Zero is ?

Best Regards
Jan

tvoverbeek, 2021-02-23:
PiJuice Zero has the same circuitry as the PiJuice HAT (minus the eeprom and the fuel-gauge IC).
So the figures should be the same.

jan94, 2021-02-24:
Thanks a lot!

jan94, 2021-02-24:
One question left:
Does this mean, the PI Juice Zero cannot determine the battery state, as is does not have a fuel gauge IC ?