PiSupply/PiJuice #654: Pi 4 and strange I2c or Battery issue

alpapak, 2021-02-24:
I just got my PiJuice for my Pi4.

My set up is:
- Pi 4 4GB
- Samsung Fit USB 32GB
- Raspberry Official Power Adaptor USB - C

I am using my Pi4 as my local web server, running the latest Rasbian OS, up today, Apache2, MariaDB, PHPMyAdmin, PHP, on power 24/7, headless. 

I installed PiJuice(power to Pi 4), checked GPIO position, boot, install software and got the following message:

    Adding user pijuice to group i2c
    Adding user pi to group pijuice 
    settings file created
    Setting ownership and permissions
    Enabling I2C
    pijuice.service is not active running, attempting to enable
    Created symlink /etc/systemd/system/multi-user.target.wants/pijuice.service → /lib/systemd/system/pijuice.service.
    *************************************************************
    Until you update the firmware to V1.4 you still need the
    'dtoverlay=i2c-bcm2708' work around in /boot/config.txt.
    After the firmware update to V1.4 the 'dtoverlay=i2c-bcm2708'
    can be removed from /boot/config.txt
    *************************************************************

    Press any key to continue
    Processing triggers for man-db (2.8.5-2) ...
    Processing triggers for libc-bin (2.28-10+rpi1) ...

I ran the terminal software and noticed firmware version 1.4.
While i was checking the software i got errors.

	Traceback (most recent call last):
	  File "/usr/bin/pijuice_cli.py", line 2013, in <module>
		loop.run()
	  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 286, in run
		self._run()
	  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 384, in _run
		self.event_loop.run()
	  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 788, in run
		self._loop()
	  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 821, in _loop
		alarm_callback()
	  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 172, in cb
		callback(self, user_data)
	  File "/usr/bin/pijuice_cli.py", line 241, in update_status
		status_args = self.get_status()
	  File "/usr/bin/pijuice_cli.py", line 206, in get_status
		general_info += ", %3d°C, %s" % (temp, status['battery'])
	KeyError: 'battery'

	Traceback (most recent call last):
	  File "/usr/bin/pijuice_cli.py", line 2013, in <module>
		loop.run()
	  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 286, in run
		self._run()
	  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 384, in _run
		self.event_loop.run()
	  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 788, in run
		self._loop()
	  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 825, in _loop
		self._watch_files[fd]()
	  File "/usr/lib/python3/dist-packages/urwid/raw_display.py", line 404, in <lambda>
		event_loop, callback, self.get_available_raw_input())
	  File "/usr/lib/python3/dist-packages/urwid/raw_display.py", line 502, in parse_input
		callback(processed, processed_codes)
	  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 411, in _update
		self.process_input(keys)
	  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 511, in process_input
		k = self._topmost_widget.keypress(self.screen_size, k)
	  File "/usr/lib/python3/dist-packages/urwid/container.py", line 595, in keypress
		*self.calculate_padding_filler(size, True)), key)
	  File "/usr/lib/python3/dist-packages/urwid/container.py", line 1590, in keypress
		key = self.focus.keypress(tsize, key)
	  File "/usr/lib/python3/dist-packages/urwid/container.py", line 2271, in keypress
		key = w.keypress((mc,) + size[1:], key)
	  File "/usr/lib/python3/dist-packages/urwid/decoration.py", line 622, in keypress
		return self._original_widget.keypress(maxvals, key)
	  File "/usr/lib/python3/dist-packages/urwid/listbox.py", line 999, in keypress
		key = focus_widget.keypress((maxcol,),key)
	  File "/usr/lib/python3/dist-packages/urwid/decoration.py", line 622, in keypress
		return self._original_widget.keypress(maxvals, key)
	  File "/usr/lib/python3/dist-packages/urwid/wimp.py", line 540, in keypress
		self._emit('click')
	  File "/usr/lib/python3/dist-packages/urwid/widget.py", line 460, in _emit
		signals.emit_signal(self, name, self, *args)
	  File "/usr/lib/python3/dist-packages/urwid/signals.py", line 265, in emit
		result |= self._call_callback(callback, user_arg, user_args, args)
	  File "/usr/lib/python3/dist-packages/urwid/signals.py", line 295, in _call_callback
		return bool(callback(*args_to_pass))
	  File "/usr/bin/pijuice_cli.py", line 1913, in item_chosen
		callback()
	  File "/usr/bin/pijuice_cli.py", line 1009, in __init__
		self.refresh()
	  File "/usr/bin/pijuice_cli.py", line 1130, in refresh
		self.main()
	  File "/usr/bin/pijuice_cli.py", line 1102, in main
		self.TEMP_SENSE_OPTIONS[self.temp_sense_profile_idx]), on_press=self.select_sense)),
	AttributeError: 'BatteryProfileTab' object has no attribute 'temp_sense_profile_idx'

At that point, i decided to add the line in boot/config.txt under [pi4] section and reboot.

The software crashes disappeared but i noticed on status page the following error:

[image]

My lsmod output:

	pi@raspberrypi:/ $ lsmod|grep i2c
	regmap_i2c             16384  1 rtc_ds1307
	i2c_bcm2708            16384  0
	i2c_dev                20480  2

My i2cdetect output:

	pi@raspberrypi:/ $ i2cdetect -y 1
		 0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
	00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
	10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- -- 
	20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
	30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
	40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
	50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
	60: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- -- 
	70: -- -- -- -- -- -- -- --

I tried to measure discharge time of my setup and got around 20min.

On my status page, on charging, i show jumping from 12% to 35%. I was showing charging then error communication and then 35%.

Also, my current status page is the following:

[image]

As you can see, my battery is stack on 96% to 97% . 

Is there any error log file?
How can i fix this issue?
My hardware seems to work fine, as if i unplug the power source the Pi is working fine.

Thank you in advance.

tvoverbeek, 2021-02-24:
i2cdetect output looks fine. All other output smells like I2C communication between RPi and PiJuice is not working correctly.
What version of the packages do you have installed (`apt policy pijuice-base`)?
What is the output of `systemctl status pijuice` and `journalctl -u pijuice`? (probably identical).

alpapak, 2021-02-24:
pi@raspberrypi:/ $ apt policy pijuice-base
	pijuice-base:
	  Installed: 1.6
	  Candidate: 1.6
	  Version table:
	 *** 1.6 500
			500 http://archive.raspberrypi.org/debian buster/main armhf Packages
			100 /var/lib/dpkg/status

	pi@raspberrypi:/ $ systemctl status pijuice
	● pijuice.service - PiJuice status service
	   Loaded: loaded (/lib/systemd/system/pijuice.service; enabled; vendor preset: enabled)
	   Active: active (running) since Wed 2021-02-24 08:13:48 EET; 7h ago
	 Main PID: 749 (pijuice_sys.py)
		Tasks: 1 (limit: 4915)
	   CGroup: /system.slice/pijuice.service
			   └─749 /usr/bin/python3 /usr/bin/pijuice_sys.py

	Feb 24 08:13:48 iphonepi systemd[1]: Started PiJuice status service.

	pi@raspberrypi:/ $ journalctl -u pijuice
	-- Logs begin at Wed 2021-02-24 08:13:38 EET, end at Wed 2021-02-24 15:29:01 EET. --
	Feb 24 08:13:48 iphonepi systemd[1]: Started PiJuice status service.

tvoverbeek, 2021-02-24:
That all looks the way it should be.
Since you have quite a few applications running the Pi4 is probably changing its clock speed depending on the load.
These clock frequency changes are then also happening to the I2C clock which could cause failed I2C transactions.
Could you try the following:
In /boot/config.txt add the lines:
```
core_freq=500
core_freq_min=500
```
This forces the clock frquency to be constant.
Reboot and see if it behaves better.

alpapak, 2021-02-25:
Thank you.

I remove `'dtoverlay=i2c-bcm2708' from boot/config.txt and add your lines.
After 12 hours it is working fine.

I want to ask:
- Can i use this line with Pi Zero? As i use my Pi zero as back up system.
- Which function to use, as i want in low charge in system event to do `sudo shutdown`. (`SYS_FUNC_HALT` or `SYS_FUNC_HALT_POWER_OFF` or `SYS_FUNC_SYS_OFF_HALT`)?
- If `pijuice_cli` software is running and not close by user, i can't load it as i am getting an message that it is used by another. How can i force to close in order to load it?

Thank you in advance.

tvoverbeek, 2021-02-25:
You wrote:

`Can i use this line with Pi Zero? As i use my Pi zero as back up system.`
For the Pi Zero use `force_turbo-1` in /boot/config.txt instead to guarantee a fixed clock speed..

`Which function to use, as i want in low charge in system event to do sudo shutdown. (SYS_FUNC_HALT or SYS_FUNC_HALT_POWER_OFF or SYS_FUNC_SYS_OFF_HALT)?`
SYS_FUNC_HALT_POWER_OFF. This shuts down the Pi (provided the pijuice service is running) and removes the 5V on the GPIO after 30 seconds.

`If pijuice_cli software is running and not close by user, i can't load it as i am getting an message that it is used by another. How can i force to close in order to load it?`
You can only run one instance of either pkjuice_cli or pijuice_gui (for obvious reasons: only one captain on the ship).
If you have an other instance running in a different terminal window or ssh session you will have to kill it using the os kill command.

alpapak, 2021-02-25:
Thank you.

One final question. Can i use one boot/config.txt for Pi 4 and Pi Zero?

`core_freq=500`
`core_freq_min=500`

and 

`force_turbo-1`

in one file?

Or i have to change it every time?
If yes, how it will look like?

Thank you very much for your help.

tvoverbeek, 2021-02-25:
You can have a common config.txt file using conditional filters.
See https://www.raspberrypi.org/documentation/configuration/config-txt/conditional.md
Use [pi0] for Pi Zero and [pi4] for ... you guessed it!.

closed 2021-02-25 by alpapak