PiSupply/PiJuice #655: Possibly Defective PiJuice HAT?

speedeysurfer, 2021-02-24:
Hi, I recently purchased the PiJuice HAT and a 12000mAh battery from Pi Supply. I followed all the instructions to install the battery exactly but was having all sorts of issues getting it to work with my Raspberry Pi 4 4GB. 

First, the software gui kept throwing communication error messages at random. And when it didn't, it was displaying all sorts of different values for voltages and battery level. It also would not ever indicate that it was charging (blinking blue LED), no matter if I was using the microUSB on the HAT or the USB-C power input on the Pi4. The LED was always just red. Eventually, my Pi4 wouldn't even boot up if the PiJuice HAT was attached. 

I detached the HAT because I was afraid it would damage the Pi4 if it hadn't already, removed the 12000 mAh battery, reattached the 1820 mAh battery it came with, and tried to charge the smaller battery without the PiJuice attached to the Pi4. The LED is constantly red and will not blink blue to indicate charging. Holding down the switch to power it off, even for more than 20 seconds, doesn't seem to do anything- the light just remains red.  I tried holding down switch 2 & 3 for 20 seconds. The lights flashed, but nothing changed after that. When not plugged in, the LED is simply off, and pressing switch 1 doesn't do anything. I'm afraid I have a defective HAT, or the 12000mAh battery was defective and ruined the HAT. I have no clue. 

I tried to initiate a return and potential replacement but I was told to open an issue here. I tried reading through some previous issues and tried a few things I read but to no avail. 

Any help would be greatly appreciated!

mmilann, 2021-02-24:
Can you post screenshot of pijuice_cli or gui status window?

If you have voltmeter nearby, can you check:
 - battery voltage, pins 1,4 on terminal block J1
 - 3V3 voltage on connector P2
 - voltage at GPIO2 and GPIO3 Rpi header while Rpi is on?

speedeysurfer, 2021-02-24:
I realize this is not helpful, but I can't provide screenshots because the Pi4 won't even boot if the PiJuice HAT is attached. It just loops in an attempt to boot but never actually does, which I'm afraid will damage my Pi4 or SD card if I keep trying. The Pi4 boots fine on it's own. 

Unrelated, but I wasn't able to view pijuice_cli when I was able to boot with the PiJuice initially (I read to try that in another thread). The prompt said access denied or something along those lines. Is that because I installed the gui version? 

I don't have a voltmeter.

mmilann, 2021-02-24:
Only in case you can do wiring, try to connect only I2C lines GPIO2 and GPIO3 plus ground, without attaching to Rpi, it would be safe. This way you can get access over I2C and read status with pijuice cli. Also if pijuice_cli does not run try adding user to "sudo" group.

speedeysurfer, 2021-02-24:
I unfortunately don't have any wires available to test with, I'd have to purchase some for the effort. I was hoping the PiJuice HAT UPS would be more of a plug and play solution.

mmilann, 2021-02-24:
It may be non-functional firmware, like when firmware update process is interrupted. In this case it could be rewritten:
`pijuiceboot 14 ./PiJuice/Firmware/PiJuice-V1.4_2020_01_17.elf.binary`
however you need I2C connection for this, 5V GPIO is not necessary between Rpi and PiJuice if both sides have own power.

PiJuice is tested in production before shipped, however  it is open frame board, can be damaged by ESD, or miss aligned connection when attached to Rpi, or any other reason that may may stress circuit when it is found in unfamiliar position. In that case you may request return.

speedeysurfer, 2021-02-24:
Thank you for the help. Without the ability to diagnose the issue I will initiate the return. I will order replacements of the HAT and 12000mAh battery because I don't see why this solution wouldn't work giving the success others have had.

closed 2021-02-24 by speedeysurfer

speedeysurfer, 2021-03-07:
I just wanted to update this as I have received replacements for both the HAT and 12000mAh battery. Both function perfectly with no issues! Not sure what was wrong with the original parts I received.