PiSupply/PiJuice #656: External Switch to Power On RPi4 in Custom Case

speedeysurfer, 2021-02-25:
I'm designing a case that I plan to 3D print to house my components, so the switches on the board won't be accessable. I need to then install external switches. Reading through the documentation it sounds like there are a couple ways to accomplish this. I have no soldering experience, and no iron, so I'd prefer not to mess with J5. It sounds like if I were to short pins 5 & 6 with some sort of wired up switch that would do the trick? Ideally I want to use a toggle switch because I like them, but it sounds like I need a normally on (NO) switch? Can I use a toggle instead? I suppose I'd have to toggle it on then off to stimulate an NO switch? Any guidance here would be greatly appreciated!

speedeysurfer, 2021-02-25:
Also, why pin 6? I realize the proximity is convenient for a header or something, but could I use any other ground pin? Like pin 9?

tvoverbeek, 2021-02-25:
You are talking about the GPIO3 (pin 5) wake from halt.
This way you can start the Pi when conecting GPIO3 to ground momentarily.
But it requires the Pi boot loader is in Halt state **with the 5V already present**.
Not a good situation for an UPS, since it requires the 5V to remain present -> draining the battery.

speedeysurfer, 2021-02-25:
Ok I think I understand. Let me back up then, is there a way to have external switches on a custom case that safely boot and shutdown the RPi4? I'd also like the ability to do the same with the PiJuice. Meaning (I think) there would be two external switches, one controlling on/off of the RPi4, and the other controlling on/off of the PiJuice. 

I guess that I'd really prefer to turn on/off (safely) the whole system with one switch. Is that possible?

tvoverbeek, 2021-02-26:
How do you power your setup? Why do you need the PiJuice if you only need turn on/turnoff of th RPi?

speedeysurfer, 2021-02-26:
I was looking for a portable, battery powered solution so that my Rpi4 could be powered for a period of time when not plugged in. The PiJuice would also provide a UPS so that if it were to run out of battery or power went out it would safely shut down (via PiJuice software). Being able to unplug the system, move it somewhere else, and plug it back in while it remaining powered up is desired. 

Is there something I'm missing? It sounded to me like this was a perfect product for this application. And all I simply want to do is control power to my system (PiJuice and RPi4), safely, with an external switch located on a custom case.

tvoverbeek, 2021-02-26:
The default BP7X PiJuice battery will not last very long to keep your Pi4+SSD going. If you need a decent period you need a bigger battery.

Here is something which might work.
External power connected to the Pi4 USB-C.
Use  2 momentary switches (push buttons), one for RPI on and one for RPi off.
The on switch connected between GPIO3 and Ground (as you proposed above).
The off switch to a GPIO (configured as input) of your choice but not to GPIO2 or GPIO3 (needed for I2C when the RPi has booted0 and Ground.
Use the gpio-shutdown overlay. See /boot/overlays/README
This causes the Pi to do an orderly shutdown.
Program the PiJuice on the system task tab to enable 'Software Halt Power Off' with a timeout long enough to do a proper shutdown (e.g. 20 seconds). This will remove the 5V from the PiJuice to the RPi after shutdown to prevent battery drain when the external power disappears after the RPi has shut down.
Also on the PiJuice program on the system task tab enable the low charge with a threshold of say 15%.
On the system events tab enable low charge and connect it to SYS_FUNC-HALT_POW_OFF. This will shut down the Pi when the battery goes below 15%
Note that with this setup you can only switch on by connecting the external power to the RPi again.

speedeysurfer, 2021-02-26:
Thank you that is very helpful. Forgive my ignorance, but may I ask why in your proposed layout that the RPi needs to be connected to external power to be switched on? If the battery has full charge, wouldn't the short of pins 5&6 turn the RPi on?

speedeysurfer, 2021-02-26:
Also FYI, I plan on using the 12000 mAh battery.

tvoverbeek, 2021-02-26:
For the on switch to work the Pi must already be powered with 5V, which then has to come from the PiJuice.
You can only get 5V to the Pi by pressing SW1 (or a pushbutton connected to J5, which requires soldering).
12000mAh battery would be OK for your application.

speedeysurfer, 2021-02-26:
Understood, thank you. I may have found a friend who has offered to solder a 2 pin header onto J5 for me. In this scenario, would I control the entire system (safely) using one switch connected to J5? 

What I'm proposing is: external power connected to RPi to charge the battery. One push button switch located on the external case connected to J5. The entire setup (PiJuice, RPi, Display (powered by RPi/PiJuice pins 2, 4, & 6) is controlled (on/off safely) by that pushbutton. Would this work? In this setup I would still like the same programmable options to safely (automatically) shut down at a certain battery level.

I guess in my proposed scenario I would need a separate way of switching on/off the display itself so that it stops draining PiJuice battery when I want the system powered down? Or does J5 being switched off cut power to the whole system?

tvoverbeek, 2021-02-27:
With a pushbutton connected to J5 and the external power to the PiJuice it should work.
A short press will turn it on and a long press will initiate a shutdown and remove the 5V power.
On the buttons tab you have to specify HARD POWER On for short press (= the default) and for long press 1
SYS_FUNC_HALT_POW_OFF. I would remove the default assignment for long press 2 and change it to NO FUNCTION.
The default hard power off will remove the 5V abruptly without an orderly shutdown (danger of sd-card corruption).
Keep the low charge setting as described above.

speedeysurfer, 2021-03-07:
Thank you very much for all the help, I did pretty much exactly as you said (with a few tweaks) and everything is working how I hoped it would.

closed 2021-03-07 by speedeysurfer