PiSupply/PiJuice #657: PiJuice Settings shows a loop of normal battery info, data_corrupted and communication_error

Paul760, 2021-02-25:
I have two issues with my new PiJuice and my new Raspberry Pi 4 with 8gb. 

1.) When I have my PiJuice Settings window open, I can see the battery information with the GPIO power input. It shows normal data for sometime, then DATA_CORRUPTED and later COMMUNICATION_ERROR. Then everything is fine again and another loop with DATA_CORRUPTED and COMMUNICATION_ERRORs occur. What can be the case? I2C is enabled.

2.) I charged the battery to 100% last evening. And because I am testing my new board and it is not operational for 24h a day I  did shutdown the rpi last night. Now when I started the rpi again the battery was empty... Is that normal? Only a samsung T7 SSD was attached to the Rpi because my future plan is that it will be a mysql server with the PiJuice as UPS. But the led turned off when the Pi was shutdown. Is there a problem with the battery?

tvoverbeek, 2021-02-26:
Regarding 1). If your Pi4 has a varying load it will vary its clock speed. This includes the I2C clock (SCL). When this happens during an I2C transaction you get COMMUNICATION ERROR. Try to put the following in /boot/config.txt:
```
core_freq=500
min_core_freq=500
```
and reboot.
See if this improves te I2C communication.

Regarding 2)
If you just shut down the Pi, the PiJuice continues to deliver 5V to the Pi and its peripherals (your SSD) still draining the battery. You need to program the PiJuice to remove the 5V after RPi shutdown.

Regarding 3)
I do not know. Personlaly I cannot distinguish between orange and red, only between below and above 50%

Paul760, 2021-02-26:
Thank you very much

closed 2021-02-26 by Paul760