PiSupply/PiJuice #658: Communication Error

dshookowsky, 2021-02-26:
I got a PiJuice from the Kickstarter. After the long delay, I ended up scrapping the project that I had planned using the PiJuice for. It's sat on a shelf for a while. I was looking to use it as a UPS for my PiHole and tried setting it up tonight. 

I'm getting nothing. I installed the pijuice gui, but there's not tray icon and I'm seeing "COMMUNICATION ERROR" everywhere.
i2cdetect -y 1 shows nothing.

The pijuice status service wasn't running, but I was able to enable it with sudo systemctl enable pijuice.  Still nothing working

What should I look at next?

dshookowsky, 2021-02-26:
More information - the battery has a charge and I can power the Raspberry Pi after hitting one of the buttons,

```
pi@raspberrypi:~ $ lsmod | grep i2c
regmap_i2c             16384  1 rtc_ds1307
i2c_bcm2835            16384  2
i2c_dev                20480  4
```

i2c seems to be working otherwise 
```
pi@raspberrypi:~ $ ls -l /dev/i2c-1 
crw-rw---- 1 root i2c 89, 1 Feb 25 22:45 /dev/i2c-1
```

pijuice_cli also returns the COMMUNICATION_ERROR. Is this a case of me running early firmware?

mmilann, 2021-02-26:
What is your Rpi version?

Is there visable  blue/green or blue/red status LED blinking when power source is connected to PiJuice?

dshookowsky, 2021-02-26:
This is a Raspberry Pi 3B (v1.2).

When I connect power to the PiJuice on its own, there are no indicator lights.

mmilann, 2021-02-26:
If pijuice is not damaged on some way, you may try to write firmware manually with command:
`pijuiceboot 14 ./PiJuice-V1.4_2020_01_17.elf.binary
`

dshookowsky, 2021-02-26:
Unfortunately, no luck there
```
pi@raspberrypi:~ $ pijuiceboot 14 ./PiJuice-V1.4_2020_01_17.elf.binary
using address 14
Input file ./PiJuice-V1.4_2020_01_17.elf.binary
Starting bootloader
Starting bootloader old firmware
Sending get command
error receiving data -1
```

mmilann, 2021-02-26:
One thing you can check is i2c is enabled, is there line:
dtparam=i2c_arm=on
in /boot/config.txt?
if not add it and reboot.

If i2c on Rpi side is working, you can then try to manualy start pijuice bootloader before writing firmware. To start bootloader
first completely remove power and battery, then press and hold SW3 while connecting battery, after that, release SW3 and plugin Rpi to power source to boot and then try above pijuiceboot command to write firmware.

dshookowsky, 2021-02-26:
- Confirmed i2c is enabled in config.txt
- Re-imaged a brand new SD card
- Attempted the bootloader instructions above (reconnect battery while holding SW3 as opposed to below where the battery is out until after flashing)
- Removed battery and followed bootloader instructions at https://github.com/PiSupply/PiJuice/tree/master/Firmware under "PiJuice bricked after successful firmware update". 

Unfortunately, the result is the same, "error receiving data -1"

mmilann, 2021-02-26:
Thanks for doing procedure. Now I can just guess battery or something on board may be faulty. 

If you own voltmeter you can test battery voltage, if totally discharged, voltage may be below 3V. also you can measure voltage at I2C pins, GPIO2 and GPIO3 when Rpi is on pijuice on top, it should be around 3.3V normaly.

Also if you can to post photo of your setup.

dshookowsky, 2021-02-26:
Measuring GPIO2->GND 3.1V
Measuring GPIO3->GND 0.3V

Battery is actually reading 3.7V

There really isn't much to my setup right now. Just a Pi3B and the PiJuice on top of it.

[image]

mmilann, 2021-02-26:
Thanks for the photo. No LED is lighting on pijuice, like mcu is unpowered, you can check pijuice mcu voltage, 3V3 labeled pin at P3 header.

SCL line is held low, in order to check if it is caused by pijuice or something is wrong on Rpi side, you can detach pijuice and then measure voltage on Rpi GPIO3.

dshookowsky, 2021-02-26:
Hmm...still only getting .2V on SCL without the PiJuice, but getting 3.3V on the header

dshookowsky, 2021-02-26:
Yanked a different Raspberry Pi 3 B from my retropie setup and booted it up with the pijuice and the same SD card and now I'm seeing stuff on the i2cdetect call. Guessing there's something wrong with the Pi itself. Thank you for the prompt and very involved assistance diagnosing the issue.

mmilann, 2021-02-26:
That's interesting, normaly there are 1K8 pullups on Rpi side to keep is 3V3 when I2C bus is idling. You mentioned that you used fresh image, so there may not be other software or driver enabled that takes over GPIO3 and sets it low. Anyway if you have any other Rpi then try it, supposing that GPIO3 pin is damaged.

closed 2021-02-26 by dshookowsky