PiSupply/PiJuice #659: 12000mAh battery.

SebAcou, 2021-02-26:
Hello, 
I have the same kind of problem with a pi 4, pijuice and a 12000mAh battery. With the original battery everything is fine. But with the big one, impossible to charge. All the options are checked in the software (GUI): charging enable, battery
i have tried long press on SW1 and SW3, not working. That red light don't want to go blue.
Any clue before concluding a hardware issue?
Best,

SebAcou, 2021-02-26:
I found the answer here: #625 
Is it safe not to monitoring the temperature. I would like put the all thing outside in a box powered by a solar panel. Can I have safety issue ? 60° is kind of a limit?

BoelD, 2021-03-01:
There is a new fir new firmware, as mentioned in #636, that apart from other issues also repairs the wrong temperature reading. If you install this firmware, you should also update the software package on the pi.

SebAcou, 2021-03-02:
Thank you very much! I will try that new firmware to see if I can have a bit more control

SebAcou, 2021-03-02:
I installed the new firmware and the new software pijuice-base  v1.7. 
The software displays well the firmaware version. However the temperature is minus 9 and it is not charging in the AutoDetect Mode. I didn't quite understand the json file since it is already in the list of available batteries.
Anyclue for the mode?

tvoverbeek, 2021-03-02:
You seem to have one of the 12000mAh batteries with a deviating NTC resistor.
Create a CUSTOM battery profile with the values specified in
https://github.com/PiSupply/PiJuice/blob/v1.7/Software/Configs/Batteries/LIPO_12000_PIS-1129.JSON
and see if this produces more reasonable temperatures and charging.
You also need to install the 1.7 deb packages which contain the 1.5 firmware which can deal with non-standard NTC resistor values. You can find them in https://github.com/PiSupply/PiJuice/tree/v1.7/Software/Install

SebAcou, 2021-03-03:
ok, I understood. Thanks. I added the custom battery specs and I have now a temperature of 22°c with the NTC resistor.

shawaj, 2021-03-15:
this is fixed by #612 and #678 and #679 but will leave open until debs are pulled into raspi repo

shawaj, 2021-03-16:
See #679 these are now pulled into the repo so there is a new battery profile and also the ability to override the settings for NTC sensor resistance

closed 2021-03-16 by shawaj

SebAcou, 2021-06-14:
Hello, 
I am  reopening that issue because even with the new battery profile, i have got much trouble ( https://github.com/PiSupply/PiJuice/blob/master/Software/Configs/Batteries/LIPO_12000_PIS-1129.JSON). The temperature is moving in a binary mode between 44 and 88.  Is there a really safety problem if i put the temperature sensing on "ON BOARD"?  
[image]
[image]