PiSupply/PiJuice #662: RPi4 vs PiJuice : battery communication error

LGM29, 2021-03-02:
Hi,

I'm trying to use a PiJuice Hat on my RPi4.
I've ordered one, turned it back, got a replacement one, but both of them have the same issues.

1- The DIP switches were incorrectly set (both on OFF position when they should be ON|OFF) as stated here https://github.com/PiSupply/PiJuice/blob/master/Hardware/README.md

2- When the PiJuice is fitted on the Rpi4, software installed, everything seems to be ok but I get "battery communication errors" in the settings window, battery percentage varies randomly, GPIO power input stated as "BAD" ..

Is this thing really compatible with RPi4 ?
Firmware is presumably up to date (1.4), nothing fancy on my pi, running stock raspberry pi OS without any other hardware connected ..

Thanks for your help, pardon my english I'm french.

LGM29, 2021-03-02:
I've juste noticed that the LED is steady green .. Pi is on and connected via USB-C, PiJuice plugged on the Pi. Micro USB is not connected.
This LED status is not listed in https://github.com/PiSupply/PiJuice

LGM29, 2021-03-04:
Hi,
Is there any log I could provide to help ?
Is there any kind of support ?
I bought this device because RPi4 was supposed to be supported .. can anyone confirm this compatibilty ?
Or should I just ask the seller for a refund .. I've already paid shipping twice ..
Thanks

mmilann, 2021-03-04:
What shows status GUI window, Battery, voltage, GPIO voltage, battery voltage?

Do you see communication errors intermittently or you never seen correct response?

Also what battery you are using? You can set profile from Battery window, pijuice_cli or pijuice config GUI.

Llewv123, 2021-03-04:
Thanks, the errors are intermittent. When data is corrupted/comms error (and batt charge was say 40%), when comms resumes, batt charge =0%...
I'll do pics of the settings and voltage levels soon as I can thanks

The battery type is correctly selected in PiJuice GUI.

LGM29, 2021-03-09:
Hi,

The **core_freq=500 core_freq_min=500** workaround did make the PiJuice work here.
I also upgraded firmware from 1.4 to 1.5 from the v1.7 package.
I still have some problems, which might be due to my power adapter, which could be too weak. Will try with the official power adapter. (My Pi uses a SSD drive, which might eat a lot of energy).

Bye