PiSupply/PiJuice #663: Raspberry shuts down directly after boot

Jamos1988, 2021-03-03:
Hi guys, which one of these do I need to select?

[image]

I run on gpio main power, and the battery is fallback.
If gpio power is lost, I need battery fallback and safe shutdown.

At the moment when I configure like so, system will shutdown immediately:

[image]

Hope someone can help find me the proper configuration.

tvoverbeek, 2021-03-03:
GPIO power PRESENT means the PiJuice gets powered via the 5V GPIO pins by the Raspberry Pi.
In other words externak

Jamos1988, 2021-03-04:
Also, the backup battery is not always taking over when primary power fails. This used to be the case.
My presumption was that the Pijuice is automatically switched on when the Pi is switched on, and that pressing SW buttons is not really necessary right? Is this all dependent on the software configuration?

Llewv123, 2021-03-04:
I have the same issue. I have a script that's supposed to run on NoPower  but it just shuts down Also sometimes it looses comms (I2C), the appears to shut down and fucking the GPIO power does nothing - nothing starts again, he Rpi us Off, the PiJuice reports NO GPIO power, and ONLY way out of it is to momentarilyREMOVE the battery, then return it...THEN it sees the GPIO power (4A Traco DCDC power supply).

tvoverbeek, 2021-03-04:
@Llewv123 See my email. 
Suggestion: connect your 4A DCDC supply to J5 of the PiJuice. Then you get power conditioning form the power converters on the PiJuice.

Llewv123, 2021-03-04:
Thanks Ton will check mail. I will tommorow try the J5 power option and post.

Jamos1988, 2021-03-04:
Hi Ton,

I think I received this mail from you but not sure:

GPIO power PRESENT means the PiJuice gets powered via the 5V GPIO pins by the Raspberry Pi.
In other words externak

Yes, I think I understand. However, can you elaborate on how I can reach the desired situation?
So gpio (external) power is primary, if this is lost the battery takes over and does a safe shutdown?

Llewv123, 2021-03-04:
Hi Ton,

 

I tried powering up via J5 with no success (i.e. power not seen via J5 – current limited when trying both polarities as there appears to be no info on J5). 

 

I then powered PiJuice via the 4 to 10V input, and that works (battery charges, power is reported seen in GUI).

 

BUT, the Rpi does not switch on when I power up PiJuice. If I connect USB C power to the Rpi, it starts up.  I can then remove the C power, and Rpi keeps going.

 

When I switch of power to PiJuice, the system now runs the user script and shuts down after tasks completed.

 

BUT the only way I can get the Rpi to turn ON again, is a setting in the GUI that says turn on “No Battery Turn on “ enabled.

 

I then discovered that I had GPIO Input Enabled. I turned this feature OFF and presto, the Rpi now starts up when power is supplied to the PiJuice power input.

 

It appears that I am out of the woods.

 

Thanks very much for your guidance.

 

Groetenis,

Llewellyn

Jamos1988, 2021-03-05:
Hi, what about my issue. It does seem to be a little different.
I've done some testing:

- Remove gpio power source: pijuice battery does indeed take over
- After only a few seconds, pijuice will completely turn off

What am I doing wrong? Battery is fully charged, so it can't be a battery failure of some sort.

[image]
[image]

Jamos1988, 2021-03-05:
I've done some more testing, no matter which option I select at NO_POWER it will not run any function.

SYS_FUNC_HALT_POW_OFF instantly kills the power
SYS_FUNC_REBOOT will stay on for a few seconds, then kills the power

I am really having trouble configuring this in my idea simple scenario:

- gpio power lost
- battery takes over
- safe shutdown
- after 60 seconds kill power

How can I configure this.

tvoverbeek, 2021-03-05:
One thing: Both SYS_FUNC_HALT_POW_OFF and SYS_FUNC_REBOOT need the pijuice service running.
When the Pi is on what does `systemctl status pijuice` output?
The 60 seconds you specify for Software Halt Pow off only are valid when you shutdown yourself from e.g. an own script or the gui desktop menu.
SYS_FUNC_HALT_POW_OFF has its own time delay (30 seconds) and is not affected by the Software Halt Pow Off setting.
How do you provide the 5V GPIO power?

Jamos1988, 2021-03-05:
Hi, service is running fine:

```
● pijuice.service - PiJuice status service
   Loaded: loaded (/lib/systemd/system/pijuice.service; enabled; vendor preset: 
   Active: active (running) since Fri 2021-03-05 11:46:03 CET; 7h ago
 Main PID: 608 (pijuice_sys.py)
    Tasks: 1 (limit: 2065)
   CGroup: /system.slice/pijuice.service
           └─608 /usr/bin/python3 /usr/bin/pijuice_sys.py

Warning: Journal has been rotated since unit was started. Log output is incomplete
```
I have a custom PCB and in there is a DC/DC power supply. It takes between 9-39 volts, and produces stable 5v 3a.
This is then directly routed to the Pi's GPIO header.

Llewv123, 2021-03-05:
Hi,
I have a custom pcb with 4A Traco DCDC routed directly to GPIO. could not get it to work unless I powered either the Pi or PiJuice. In the end I broke the GPIO connection and powered the PiJuice via it's 4-10V input.  It appeared to be working, but now the Pi seems to be an endless loop trying to start up (observing the PSU power supply, the current draw goes up, then down, then up, then down, but never boots up....

Jamos1988, 2021-03-08:
@Llewv123 @tvoverbeek Should we split the issues? My issue is not as the description anymore, but I can't honestly say how the issue disappeared in my case as I don't have the shutdown issue anymore. It was temporary, but now my issue is something else totally different as my original description. Shall I create a new issue more in the line of "scripts not running when on gpio power" ?

tvoverbeek, 2021-03-08:
@Jamos1988 I saw you already opened a new issue.
@Llewv123 The case where external power is delivered directly to the GPIO pins is not foreseen in the firmware.
That may be related to the problems you are seeing.

Jamos1988, 2021-03-10:
For the firmware, would in the future be included that it is also detected from GPIO rather than J4? Or is this more of a hardware thing. This way I know if I permanently need to include a header to J4 on our PCB. Thanks in advance. In the new issue I created I posted my test results by the way, hope you can support on that also. Final step into deciding if Pijuice will be permanently integrated into our asssembly line!

tvoverbeek, 2021-03-10:
I cannot answer for Pi-Supply, but my guess is this would require hardware changes. So not very likely in the near future.