PiSupply/PiJuice #665: PiJuice Firmware Version

Llewv123, 2021-03-08:
I would like to know if it is possible to upgrade the firmware on the PiJuice to whatever is later than 1.4?
I have tried many options/connections/scripts to get stability in behavior to no avail. 

I have seen posts where version 1.5 has been mentioned, and even seen a post and links to download two .deb files for v1.7

Can anyone help with this please - it would be appreciated.

Llewellyn

tvoverbeek, 2021-03-08:
Firmware version and deb package version are 2 different things.
Current deb packages version is 1.6 which contains firmware 1.5.
The beta 1.7 packages contain firmware 1.5.
You can always upgrade/downgrade the firmware directly from the terminal using pijuiceboot.
`pijuiceboot <i2c-address-in-hex> <path-to-firmware-file)` e.g.
`pijuiceboot 14 <path-to>/PiJuice-V1.5_2021_02_06.elf.binary`

Llewv123, 2021-03-08:
Powering up via J4 only:
System does not shut down immediately so it appears that user function script runs (although I don't see the message thats supposed to be in the events list "powerfail")

wait for at 30 seconds before power back on - Pi does not start up.
IF I leave the battery OUT, with the option of "no battery turn on" , the Pi always starts up - put the battery in - Pi never starts.

battery charge = 99%
Wakeup on charge enabled, with a trigger level of 0%, 15%, 30% - Pi does not start irrespective of trigger level.

What am I missing?
.

tvoverbeek, 2021-03-09:
The pijuice service polls for 'no power' every 5 seconds. So it can take up to 5 seconds before the script connected to the no power event starts.
How do you shut down the pi on the 'no power event, a user function or SYS_FUNC_HALT_POW_OFF?

Llewv123, 2021-03-10:
Thanks Ton,

Shut down via user script:

===================
#!/usr/bin/python3
import os
os.system("/home/pi/iMGuard -POWERFAIL")
os.system("sudo poweroff")
===================