PiSupply/PiJuice #666: Scripts not running - battery not taking over when on GPIO power

Jamos1988, 2021-03-08:
I have a custom PCB with a DC/DC PSU. It powers a stable 5v with 3a to the Pijuice GPIO header.
The status of the hat is:

[image]

Two things are happening:

1. when no event is configured and the power to the PSU is disconnected, the battery takes over for about 10 seconds, then the power is killed.
2. when an event is configured on NO_POWER and the power to the PSU is disconnected, the internal battery does not take over and the power is killed instantly. There is no safe shutdown. This happens with any function I try to connect to the event NO_POWER, it just kills the power. Example:

[image]

How can I configure the following:

1. GPIO power is killed and status goes to NOT_PRESENT
2. Internal battery takes over 
3. Safe shutdown is initiated
4. Power is killed
5. When power is returned to the PSU, raspberry will boot again

tvoverbeek, 2021-03-08:
The firmware cannot handle the case where external power is directly delivered to the GPIO pins.
When it sees power to the PiJuice coming via the GPIO it assumes it is coming from the RPi (-> external power delivered to the RPi).
Your best bet is to connect your external power to J4 on the PiJuice (https://github.com/PiSupply/PiJuice/tree/master/Hardware#top-of-the-board).
You probably ant to solder a header to J4.

Jamos1988, 2021-03-10:
Yes! Soldering + to J4 helps. Didn't need the - because there is a common GND. At least I think?
The problem now is, when testing the POWER_OFF, following happens:

- I remove the PSU, so no charge and POWER_OFF (seems) initiated.
- Next, the Pi does seem to have a bit of power, and the Pijuice steps in
- But, the red power LED from the Pi will blink, when it does not blink it is still burning a bit but very faint

I didn't get the idea that the safe shutdown is succeeding, so I tested with a multimeter what happens with the 5v:

- So, when system fully booted and charging, there is 4.98 V
- When I kill the PSU, voltage drops to 2.9 V
- When the red LED of the Pi blinks, the voltage is also 4.98 V and then when it is not blinking it is 2.9 V
- This is repeating again and again, power led is blinking and the voltage is jumping between 2.9 and 4.9 V.

Do I need a bigger battery? I have the standard 1820mha
Or, do I also need to solder the - of J4?

Jamos1988, 2021-03-10:
I see this is probably a new issue, I'll close this one because the battery now takes over.

closed 2021-03-10 by Jamos1988