PiSupply/PiJuice #667: PihJuice running on RPi 3b with Kali Linux 2012.1 32bit

Z3r01mPact, 2021-03-08:
I've had this running today using the sudo -u pijuice python3 /usr/bin/pijuice_gui.py after hitting the isAlive()  issue.
i was experimenting with the voltage and battery system tasks when it was rebooting endlessly due to some mistake i made.

so i blew away the image, reloaded to the sd card, 
sudo apt-get update && sudo apt-get upgrade -y
sudo apt install i2c-tools
sudo apt install lua5.3
rebooted
Cloned the git repo
sudo dpkg -i pijuice-base_1.6_all.deb
sudo dpkg -i pijuice-gui_1.6_all.deb
sudo nano /usr/lib/python3/dist-packages/pijuice.py    ( made relevant  isAlive() to is_alive() changes )
sudo visudo 
 Added  "  root ALL=(pijuice) ALL "
Rebooted

now i run the gui command and i get 

`No protocol specified
No protocol specified
Traceback (most recent call last):
  File "/usr/bin/pijuice_gui.py", line 2049, in <module>
    start_app()
  File "/usr/bin/pijuice_gui.py", line 2008, in start_app
    root = Tk()
  File "/usr/lib/python3.9/tkinter/__init__.py", line 2262, in __init__
    self.tk = _tkinter.create(screenName, baseName, className, interactive, wantobjects, useTk, sync, use)
_tkinter.TclError: couldn't connect to display ":0.0"
`
Blew it away again and followed the exact same steps and get the exact same thing. yet the same steps earlier allowed be to access the gui and make changes to system tasks etc as well as see the hat status page.

Additional System info:

s -al /lib/ld*
lrwxrwxrwx 1 root root 30 Jan  5 05:47 /lib/ld-linux-armhf.so.3 -> arm-linux-gnueabihf/ld-2.31.so
lrwxrwxrwx 1 root root 24 Feb 24 02:48 /lib/ld-linux.so.3 -> /lib/ld-linux-armhf.so.3

ldd /usr/bin/pijuice_gui
        linux-vdso.so.1 (0x7efb0000)
        libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x76e00000)
        /lib/ld-linux-armhf.so.3 (0x76f14000)

sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         

python3 --version
Python 3.9.1+

Z3r01mPact, 2021-03-08:
btw,, the image im using is kali-linux-2012.1-rpi4-nexmon.img

tvoverbeek, 2021-03-08:
The main error is `couldn't connect to display ":0.0"`.
pijuice_gui.py could not contact the X-server (maybe because not running under the same userid as the desktop)
Can you run utilities like xdpyinfo which also needs the X-server?

ro0TuX777, 2021-03-08:
Following this, as I am in the same boat.