PiSupply/PiJuice #668: Cannot Reset Wakeup Alarm on Boot Up

kneiser, 2021-03-08:
Upon boot or reboot I cannot reset the wakeup alarm. I am running headless. I want to boot the pi at a specific time everyday. From the "pijuice_cli" I can manually set the alarm, the system boots up at the correct time. Using the "wakeup_enable.py" script I can set the alarm from the command line. But when setting up a pi user crontab entry the system does not boot the next scheduled chrony period.

Wakeup Alarm post boot:
[image]

pi user crontab is:
@reboot /usr/bin/python3 /home/pi/.scripts/misc/wakeup_enable.py 3m
"wakeup_enable.py" script is:
import os
import pijuice, time

while not os.path.exists('/dev/i2c-1'):
    time.sleep(0.1)

pj = pijuice.PiJuice(1, 0x14)

pj.rtcAlarm.SetWakeupEnabled(True)

What am i doing wrong? Or are there additional settings needed?
Otherwise the pijuice is operating as expected.

kneiser, 2021-03-08:
Using the following firmware version:
[image]

kneiser, 2021-03-09:
@tvoverbeek
I tried that too:
**Python script:**
#!/usr/bin/python3
import pijuice

print("Reset Wakeup Alarm...")

pj = pijuice.PiJuice(1, 0x14)

pj.power.SetPowerOff(30)

pj.rtcAlarm.SetWakeupEnabled(True)

print("Exiting Reset Wakeup Alarm...")
**Pi user crontab:**
@reboot /usr/bin/python3 /home/pi/.scripts/misc//home/pi/.scripts/misc/resetWakeupAlarm.py

I can run resetWakeupAlarm.py from the command line and it resets the alarm but not from the pi user crontab.

tvoverbeek, 2021-03-10:
You are still running the script at boot. @reboot scripts do not execute at shutdown.

kneiser, 2021-03-15:
@tvoverbeek
I redid all of my pijuice settings:

1. From issue #309, I setup entries to allow the pijuice to for manual boot and power off.
2. I not successfully run the resetWakeupAlarm.py script via a cron scheduled job.
3. resetWakeupAlarm.py code:

`#!/usr/bin/python3

import subprocess
import pijuice

print("Reset Wakeup Alarm...")

pj = pijuice.PiJuice(1, 0x14)

pj.power.SetPowerOff(60)

pj.rtcAlarm.SetWakeupEnabled(True)

print("Exiting Reset Wakeup Alarm..."
)
print("Powering off..")

subprocess.call(['sudo', 'poweroff'])
`

closed 2021-03-15 by kneiser