PiSupply/PiJuice #669: GPIO pins still supplied with power after using SetPowerOff() function

patman141414, 2021-03-09:
I am using the SetPowerOff() function with a 30 second delay to perform a clean shutdown and subsequently stop the PiJuice from powering the GPIO pins. I have noticed that if main power is reconnected to the Raspberry Pi 4 USB-C port at any time during the 30 second delay between the start of the clean shutdown and the execution of the SetPowerOff() function, the GPIO pins will actually still be powered by the PiJuice even if the main power is disconnected again.

In this scenario, the only way to power the Pi back on is to disconnect main power, disconnect the UPS battery, and then reconnect main power. It seems as though reconnecting main power during the delay causes the SetPowerOff() function to not execute and for the GPIO pins to remain powered by the UPS.

Llewv123, 2021-03-09:
I have seen this also. It's a problem when the deployment is remote, as the only way out of this is to visit the site and remove the battery then power up. And if the power is restored within the 30 seconds shutdown..the battery will not go flat and the Pi will never start up as long as there is power. This could be very problematic. Perhaps there is a workaround that someone knows of?

tvoverbeek, 2021-03-10:
I do not know if this helps:
Did a test with my Pi4/PiJuice HAT combo.
Power connected to the PIs USB, no-power event linked to SYS_FUNC_HALT_POW_OFF, wakeup-on-charge enabled with low level (1%).
Removed power to the running Pi. Shutdown started after ca 5 seconds as expected.
Reconnected the power to the Pi about 10 seconds after the 10 flashes (shutdown finished) of the green led.
Of course the Pi remained powered, but it started booting again after ca 20 seconds (due to the wakeup-on-charge setting).
To get this scenario working you need to enable the wakeup-on-charge setting again just before shutdown.
The pijuice service does this for you when you use the SYS_FUNC_HALT_POW_OFF system function.
Otherwise you need to do it in your user function.

Note. The 1.5 firmware in combination with the gui/cli from the 1.7 deb packages gives you the option to store the wakeup-on-charge setting in non-volatile-memory. Then it will work also with user scripts without enabling wakeup-on-charge before shutdown.

patman141414, 2021-03-19:
This solution did the trick. Setting wakeup-on-charge to 1% takes precedence over other conditions and will automatically reboot the Pi when the power supply is reconnected under all the circumstances I tested. I added the set_wakeup_on_charge() function to my python code to accomplish this but it also works by adjusting the value in the PiJuice GUI.

closed 2021-03-19 by patman141414